evcc-io/evcc #17243: Login fehlgeschlagen: Passwort ist ungültig.

SmartVolker, 2024-11-13:
### Describe the bug

Nachdem ein Passwort gesetzt wurde, kann dieses verwendet werden. Nach einem Aus/Einschalten werde ich aufgefordert das Passwort einzugeben, jedoch wird nun ein Fehler angezeigt: Passwort ungültig.

### Steps to reproduce

1. Setze ein Passwort
2. Stromversorgung ausschalten
3. Stromversorgung einschalten
4. Konfiguration auswählen
5. Passwort eingeben
6. Meldung: Passwort ungültig
...

### Configuration details

```yaml
Nur Ladepunkte in der evcc.yaml
```

### Log details

```text
Kann ich nachliefern wenn erforderlich
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

naltatis, 2024-11-13:
Du schreibst, dass du den Strom ausschaltest. Aktuell werden Settings (dazu gehöhrt auch das Admin Password und JWT Secret) max. einmal pro Minute geschrieben. Funktioniert es wie gewünscht, wenn du eine Minute wartest bevor du den Stecker ziehst?

andig, 2024-11-13:
Ich möchte das egtl. Nicht beheben. Linux sollte man immer runter fahren, nicht ausschalten da jederzeit das Dateisystem zerstört werden kann. Dann hi,ft auch kein Workaound in evcc mehr.

SmartVolker, 2024-11-14:
Ich habe folgende Kommandos eingegeben:
1.  sudo systemctl stop evcc
2. evcc password reset
3. sudo reboot
4. Konfiguration Menu

Ergebnis: Login fehlgeschlagen 

1. sudo systemctl stop evcc
2. evcc password set - Neues Passwort eingegeben - 2 Minuten gewartet.
3. sudo reboot
5. Konfiguration Menu

Ergebnis: Login fehlgeschlagen

SmartVolker, 2024-11-14:
> Du schreibst, dass du den Strom ausschaltest. Aktuell werden Settings (dazu gehöhrt auch das Admin Password und JWT Secret) max. einmal pro Minute geschrieben. Funktioniert es wie gewünscht, wenn du eine Minute wartest bevor du den Stecker ziehst?
> 
> see #15145

Ja, ich habe gewartet. Es kommt immer die Meldung "Login fehlgeschlagen. Passwort ungültig"

closed 2024-11-26 by github-actions[bot]