evcc-io/evcc #17247: "broken pipe" zu E3DC WR mit RSCP Protokoll

Dirk77777, 2024-11-13:
### Describe the bug

Ich bekomme sporadisch den Fehler
[site  ] ERROR 2024/11/13 17:39:02 pv 1 power: write tcp 172.18.0.2:60170->192.168.1.16:5033: write: broken pipe
Die 172... IP ist vermutlich eine interne IP vom Docker-Container. Die 192...16 ist die IP vom E3DC WR.

Wenn der Fehler mal da ist, geht er nicht wieder weg. 
evcc steuert dann den Ladevorgang nicht mehr, d.h. z.B. ein laufender Ladevorgang wird nicht beendet, wenn z.B. Ziel SoC erreicht ist, oder der Ladevorgang wird nach anstecken des Autos nicht gestoppt, falls nicht genug PV Leistung im PV-Modus vorhanden ist. 

Ich habe den Eindruck, ein Neustart von evcc über das UI löst das Problem, aber nicht immer. 

### Steps to reproduce

Kann ich nicht sagen. Gefühlt tritt es einfach sporadisch irgendwann auf.

### Configuration details

```yaml
E3DC WR wurde über das UI konfiguriert, d.h. RSCP Protokoll mit Port 5033.
evcc läuft auf einem Raspberry Pi im Docker-Container, Pi und E3DC sind im gleichen Subnet und über LAN verbunden.
```

### Log details

```text
[site  ] DEBUG 2024/11/13 17:25:31 grid power: -47W
time="2024-11-13T17:25:31+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site  ] DEBUG 2024/11/13 17:25:31 pv power: 0W
[site  ] DEBUG 2024/11/13 17:25:31 battery soc: 20%
[site  ] DEBUG 2024/11/13 17:25:31 battery power: 2025W
[site  ] DEBUG 2024/11/13 17:25:31 site power: 2078W
[lp-1  ] DEBUG 2024/11/13 17:25:31 charge voltages: [0 0 0.9]V
[lp-1  ] DEBUG 2024/11/13 17:25:31 charger status: A
[site  ] DEBUG 2024/11/13 17:25:41 ----
[lp-1  ] DEBUG 2024/11/13 17:25:41 charge power: 0W
[lp-1  ] DEBUG 2024/11/13 17:25:41 charge currents: [0.1 0 0.1]A
time="2024-11-13T17:25:41+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-13T17:25:41+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-13T17:25:41+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] DEBUG 2024/11/13 17:25:41 pv power: 0W
[site  ] DEBUG 2024/11/13 17:25:41 grid power: 6W
time="2024-11-13T17:25:41+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site  ] DEBUG 2024/11/13 17:25:41 battery soc: 20%
[site  ] DEBUG 2024/11/13 17:25:41 battery power: 2046W
[site  ] DEBUG 2024/11/13 17:25:41 site power: 2152W
[lp-1  ] DEBUG 2024/11/13 17:25:41 charge voltages: [0 0 0.9]V
[lp-1  ] DEBUG 2024/11/13 17:25:41 charger status: A
[site  ] DEBUG 2024/11/13 17:25:51 ----
[lp-1  ] DEBUG 2024/11/13 17:25:51 charge power: 0W
[lp-1  ] DEBUG 2024/11/13 17:25:51 charge currents: [0.1 0 0.1]A
time="2024-11-13T17:25:51+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-13T17:25:51+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-13T17:25:51+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
[site  ] DEBUG 2024/11/13 17:25:51 grid power: 10W
[site  ] DEBUG 2024/11/13 17:25:51 pv power: 0W
time="2024-11-13T17:25:51+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site  ] DEBUG 2024/11/13 17:25:51 battery soc: 20%
[site  ] DEBUG 2024/11/13 17:25:51 battery power: 2016W
[site  ] DEBUG 2024/11/13 17:25:51 site power: 2126W
[lp-1  ] DEBUG 2024/11/13 17:25:51 charge voltages: [0 0 0.9]V
[lp-1  ] DEBUG 2024/11/13 17:25:51 charger status: A
[site  ] DEBUG 2024/11/13 17:26:01 ----
[lp-1  ] DEBUG 2024/11/13 17:26:01 charge power: 0W
[lp-1  ] DEBUG 2024/11/13 17:26:01 charge currents: [0.1 0 0.1]A
time="2024-11-13T17:26:01+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-13T17:26:01+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-13T17:26:01+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] DEBUG 2024/11/13 17:26:01 pv power: 0W
time="2024-11-13T17:26:01+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[site  ] DEBUG 2024/11/13 17:26:01 grid power: -16W
[... 714 lines cut ...]
[lp-1  ] DEBUG 2024/11/13 17:34:51 charge currents: [0.1 0 0.1]A
time="2024-11-13T17:34:51+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-13T17:34:51+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-13T17:34:51+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-13T17:34:52+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-13T17:34:52+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/13 17:34:52 pv 1 power: write tcp 172.18.0.2:60170->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/13 17:34:52 pv power: 0W
time="2024-11-13T17:34:52+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-13T17:34:52+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-13T17:34:52+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/13 17:34:52 battery 1 power: write tcp 172.18.0.2:60164->192.168.1.16:5033: write: broken pipe
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.5

andig, 2024-11-13:
> Docker container

See issue template

closed 2024-11-13 by andig

Dirk77777, 2024-11-26:
Sorry, den Satz hatte ich übersehen.
Hab jetzt auch evcc als Service in Linux umgestellt, also ohne Docker.
Ist wieder aufgetreten

Hier ein Ausschnitt aus dem Log, die Einträge wiederholen sich.
192.168.1.18 = Raspberry mit mit evcc Service
192.168.1.16 = E3DC
192.168.1.30 = Wallbox (Tesla Wall Connector)

> [site  ] ERROR 2024/11/25 19:59:38 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:38 pv power: 0W
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:38 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:38 ----
[lp-1  ] DEBUG 2024/11/25 19:59:38 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 19:59:38 charge currents: [0.1 0 0.1]A
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:39 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:39 pv power: 0W
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:39 grid power: write tcp 192.168.1.18:32970->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:46 ----
[lp-1  ] DEBUG 2024/11/25 19:59:46 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 19:59:46 charge currents: [0.1 0 0.1]A
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:46 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:46 pv power: 0W
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:46 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:56 ----
[lp-1  ] DEBUG 2024/11/25 19:59:56 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 19:59:56 charge currents: [0.1 0 0.1]A
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:56 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:56 pv power: 0W
time="2024-11-25T19:59:57+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:57 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 20:00:06 ----
[lp-1  ] DEBUG 2024/11/25 20:00:06 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 20:00:06 charge currents: [0.1 0 0.1]A
time="2024-11-25T20:00:06+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T20:00:06+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T20:00:06+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T20:00:06+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T20:00:06+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 20:00:06 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 20:00:06 pv power: 0W
time="2024-11-25T20:00:06+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T20:00:07+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 20:00:07 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 20:00:16 ----
[lp-1  ] DEBUG 2024/11/25 20:00:16 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 20:00:16 charge currents: [0.1 0 0.1]A
[... cut ...]
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T20:02:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"

Dirk77777, 2024-11-26:
Habe neues Issue #17446 aufgemacht.