evcc-io/evcc #17251: New server not functioning, get "connect: no route to host" error messages

b394373, 2024-11-14:
### Describe the bug

I have a working evcc server setup on an old Mac mini.

When I setup a new server on a new Mac mini M4 on the same network with the same configuration, it doesn't work even though the original server has been stopped.

I get "connect: no route to host" error messages, see log details below.

### Steps to reproduce

1. configured evcc
2. started evcc
3. not communicating with charger or meter
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ea0538807bbd6a1f1895f7aecaad897d39d117ab3b4be64bd539c4e01e4c3b7a

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.188.176  
  user: customer  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.188.176  
  user: customer  
  name: pv2

chargers:
- type: template
  template: twc3 
  host: 192.168.188.163  
  name: wallbox3

loadpoints:
- title: Garage
  charger: wallbox3
  mode: off

site:
  title: My Home
  meters:
    grid: grid1
    pv:
    - pv2
```

### Log details

```text
[main ] INFO 2024/11/14 09:37:13 evcc 0.131.5
[main ] INFO 2024/11/14 09:37:13 using config file: /etc/evcc.yaml
[db ] INFO 2024/11/14 09:37:13 using sqlite database: /Users/Eugene/.evcc/evcc.db
[main ] INFO 2024/11/14 09:37:14 listening at :7070
[site ] INFO 2024/11/14 09:37:15 site config:
[site ] INFO 2024/11/14 09:37:15 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2024/11/14 09:37:15 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/11/14 09:37:15 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/11/14 09:37:15 vehicles:
[site ] INFO 2024/11/14 09:37:15 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/11/14 09:37:15 loadpoint 1:
[lp-1 ] INFO 2024/11/14 09:37:15 mode: off
[lp-1 ] INFO 2024/11/14 09:37:15 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/11/14 09:37:15 meters: charge ✓
[lp-1 ] INFO 2024/11/14 09:37:15 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2024/11/14 09:37:15 phase timer inactive
[lp-1 ] DEBUG 2024/11/14 09:37:15 pv timer inactive
[lp-1 ] ERROR 2024/11/14 09:37:15 charger enabled: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] DEBUG 2024/11/14 09:37:15 ----
[lp-1 ] ERROR 2024/11/14 09:37:15 charge power: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[lp-1 ] ERROR 2024/11/14 09:37:16 charge currents: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] ERROR 2024/11/14 09:37:17 pv 1 power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/14 09:37:17 pv power: 0W
[site ] ERROR 2024/11/14 09:37:17 grid power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/14 09:37:45 ----
[lp-1 ] ERROR 2024/11/14 09:37:45 charge power: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[lp-1 ] ERROR 2024/11/14 09:37:46 charge currents: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] ERROR 2024/11/14 09:37:46 pv 1 power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/14 09:37:46 pv power: 0W
[site ] ERROR 2024/11/14 09:37:46 grid power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

Mac-mini-M4 /etc % evcc -v evcc version 0.131.5

b394373, 2024-11-14:
When I input http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi/ with a browser on the Mac mini M4 I get the following reply;

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 102,
               "E_Day" : 8503,
               "E_Total" : 46116400,
               "E_Year" : 11705433,
               "P" : 4979
            }
         },
         "Site" : {
            "E_Day" : 8503,
            "E_Total" : 46116400,
            "E_Year" : 11705433,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -564.28999999999996,
            "P_Load" : -4414.71,
            "P_PV" : 4979,
            "rel_Autonomy" : 100,
            "rel_SelfConsumption" : 88.666599718819043
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2024-11-14T08:17:51+10:00"
   }
}

b394373, 2024-11-14:
Sorry, pressed the wrong button

b394373, 2024-11-14:
And yet when I run the server from the command line, I don't get any errors and it works

Mac-mini-M4 ~ % evcc -c evcc.yaml       
[main  ] INFO 2024/11/14 13:51:21 evcc 0.131.5
[main  ] INFO 2024/11/14 13:51:21 using config file: evcc.yaml
[db    ] INFO 2024/11/14 13:51:21 using sqlite database: /Users/Eugene/.evcc/evcc.db
[main  ] INFO 2024/11/14 13:51:21 listening at :7070
[site  ] INFO 2024/11/14 13:51:24 site config:
[site  ] INFO 2024/11/14 13:51:24   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/11/14 13:51:24     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/11/14 13:51:24     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/11/14 13:51:24   vehicles:
[site  ] INFO 2024/11/14 13:51:24     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/11/14 13:51:24 loadpoint 1:
[lp-1  ] INFO 2024/11/14 13:51:24   mode:        off
[lp-1  ] INFO 2024/11/14 13:51:24   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/11/14 13:51:24   meters:      charge ✓
[lp-1  ] INFO 2024/11/14 13:51:24     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2024/11/14 13:51:24 phase timer inactive
[lp-1  ] DEBUG 2024/11/14 13:51:24 pv timer inactive
[site  ] DEBUG 2024/11/14 13:51:24 ----
[lp-1  ] DEBUG 2024/11/14 13:51:24 charge power: 0W
[lp-1  ] DEBUG 2024/11/14 13:51:24 charge currents: [0.3 0.2 0.1]A
[site  ] DEBUG 2024/11/14 13:51:24 pv power: 6700W
[site  ] DEBUG 2024/11/14 13:51:24 grid power: -4886W
[site  ] DEBUG 2024/11/14 13:51:24 site power: -4886W
[lp-1  ] DEBUG 2024/11/14 13:51:24 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2024/11/14 13:51:24 charger status: B
[lp-1  ] INFO 2024/11/14 13:51:24 car connected
[lp-1  ] DEBUG 2024/11/14 13:51:24 vehicle api refresh
[lp-1  ] DEBUG 2024/11/14 13:51:24 pv timer elapse
[lp-1  ] DEBUG 2024/11/14 13:51:24 pv timer inactive
[main  ] DEBUG 2024/11/14 13:51:24 vehicle status: B (BlueY)
[lp-1  ] INFO 2024/11/14 13:51:24 vehicle updated: unknown -> BlueY
[lp-1  ] DEBUG 2024/11/14 13:51:24 vehicle status: B
[lp-1  ] DEBUG 2024/11/14 13:51:24 vehicle soc: 100%
[lp-1  ] DEBUG 2024/11/14 13:51:24 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/11/14 13:51:24 vehicle range: 414km
[site  ] DEBUG 2024/11/14 13:51:54 ----
[lp-1  ] DEBUG 2024/11/14 13:51:54 charge power: 0W
[lp-1  ] DEBUG 2024/11/14 13:51:54 charge currents: [0.3 0.2 0.1]A
[site  ] DEBUG 2024/11/14 13:51:54 grid power: -4853W
[site  ] DEBUG 2024/11/14 13:51:54 pv power: 6678W
[site  ] DEBUG 2024/11/14 13:51:54 site power: -4853W
[lp-1  ] DEBUG 2024/11/14 13:51:54 vehicle odometer: 47639km
[lp-1  ] DEBUG 2024/11/14 13:51:54 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2024/11/14 13:51:54 charger status: B
[site  ] DEBUG 2024/11/14 13:52:24 ----
[lp-1  ] DEBUG 2024/11/14 13:52:25 charge power: 0W
[lp-1  ] DEBUG 2024/11/14 13:52:25 charge currents: [0.3 0.2 0.1]A
[site  ] DEBUG 2024/11/14 13:52:25 pv power: 6676W
[site  ] DEBUG 2024/11/14 13:52:25 grid power: -4879W
[site  ] DEBUG 2024/11/14 13:52:25 site power: -4879W
[lp-1  ] DEBUG 2024/11/14 13:52:25 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2024/11/14 13:52:25 charger status: B

StefanSchoof, 2024-11-14:
Ist das https://github.com/evcc-io/evcc/discussions/16165#discussioncomment-10702537?

b394373, 2024-11-14:
So, the answer is that the problem is with Apple, awaiting a fix and the workaround is to run evcc in a shell, is that correct?

andig, 2024-11-14:
Das wird mit der nächsten Go Version gelöst. Wir upgraden fast immer sofort.

andig, 2024-11-14:
Fixed in https://github.com/evcc-io/evcc/commit/039cbf5631bf70476e46eda8c66a557a8cb1a814

b394373, 2024-11-17:
or simply run the following command;

Mac-mini-M4 ~ % nohup evcc -c evcc.yaml 2>/dev/null &
[1] 81564
Mac-mini-M4 ~ %

b394373, 2024-11-17:
"039cbf5 MacOS Sequoia: fix network errors" in 0.131.6 still results in "connect: no route to host" errors

[main ] INFO 2024/11/17 15:45:00 evcc 0.131.6
[main ] INFO 2024/11/17 15:45:00 using config file: /Users/Eugene/evcc.yaml
[db ] INFO 2024/11/17 15:45:00 using sqlite database: /Users/Eugene/.evcc/evcc.db
[main ] INFO 2024/11/17 15:45:02 listening at :7070
[site ] INFO 2024/11/17 15:45:04 site config:
[site ] INFO 2024/11/17 15:45:04 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2024/11/17 15:45:04 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/11/17 15:45:04 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/11/17 15:45:04 vehicles:
[site ] INFO 2024/11/17 15:45:04 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/11/17 15:45:04 loadpoint 1:
[lp-1 ] INFO 2024/11/17 15:45:04 mode: pv
[lp-1 ] INFO 2024/11/17 15:45:04 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/11/17 15:45:04 meters: charge ✓
[lp-1 ] INFO 2024/11/17 15:45:04 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2024/11/17 15:45:04 phase timer inactive
[lp-1 ] DEBUG 2024/11/17 15:45:04 pv timer inactive
[lp-1 ] ERROR 2024/11/17 15:45:04 charger enabled: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] DEBUG 2024/11/17 15:45:04 ----
[lp-1 ] ERROR 2024/11/17 15:45:04 charge power: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[lp-1 ] ERROR 2024/11/17 15:45:05 charge currents: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] ERROR 2024/11/17 15:45:06 pv 1 power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/17 15:45:06 pv power: 0W
[site ] ERROR 2024/11/17 15:45:06 grid power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/17 15:45:34 ----

andig, 2024-11-17:
Looking it https://github.com/evcc-io/evcc/actions/runs/11871342079/job/33083793728 shows that the release version has been compiled with Go 1.23.3 which seems correct. I'm not sure why this wouldn't work.

Does anybody have an idea? Can we check if the generated binary has the expected properties?

StefanSchoof, 2024-11-17:
From 
https://github.com/golang/go/issues/68678#issuecomment-2447965386

> For previous releases (Go 1.22 and 1.23), we did a simpler backport, which requires a Go linker flag (-B gobuildid) to generate the UUID.

Is this linker flag set?

andig, 2024-11-17:
I wouldn‘t even know how 😮

andig, 2024-11-18:
Should hopefully be fixed in b8b7135028298d03e2f267a1688b2abba8e7c3a9

b394373, 2024-11-25:
"b8b7135 MacOS: add gobuildid" in 0.131.7 still results in "connect: no route to host" errors

[main ] INFO 2024/11/26 06:33:39 evcc 0.131.7
[main ] INFO 2024/11/26 06:33:39 using config file: /Users/Eugene/evcc.yaml
[db ] INFO 2024/11/26 06:33:39 using sqlite database: /Users/Eugene/.evcc/evcc.db
[main ] INFO 2024/11/26 06:33:40 listening at :7070
[site ] INFO 2024/11/26 06:33:41 site config:
[site ] INFO 2024/11/26 06:33:41 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2024/11/26 06:33:41 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/11/26 06:33:41 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/11/26 06:33:41 vehicles:
[site ] INFO 2024/11/26 06:33:41 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/11/26 06:33:41 loadpoint 1:
[lp-1 ] INFO 2024/11/26 06:33:41 mode: pv
[lp-1 ] INFO 2024/11/26 06:33:41 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/11/26 06:33:41 meters: charge ✓
[lp-1 ] INFO 2024/11/26 06:33:41 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2024/11/26 06:33:41 phase timer inactive
[lp-1 ] DEBUG 2024/11/26 06:33:41 pv timer inactive
[lp-1 ] ERROR 2024/11/26 06:33:41 charger enabled: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] DEBUG 2024/11/26 06:33:41 ----
[lp-1 ] ERROR 2024/11/26 06:33:42 charge power: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[lp-1 ] ERROR 2024/11/26 06:33:42 charge currents: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] ERROR 2024/11/26 06:33:43 pv 1 power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/26 06:33:43 pv power: 0W
[site ] ERROR 2024/11/26 06:33:43 grid power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/26 06:34:11 ----
[lp-1 ] ERROR 2024/11/26 06:34:12 charge power: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[lp-1 ] ERROR 2024/11/26 06:34:13 charge currents: Get "http://192.168.188.163/api/1/vitals": dial tcp 192.168.188.163:80: connect: no route to host
[site ] ERROR 2024/11/26 06:34:13 pv 1 power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host
[site ] DEBUG 2024/11/26 06:34:13 pv power: 0W
[site ] ERROR 2024/11/26 06:34:13 grid power: Get "http://192.168.188.176/solar_api/v1/GetPowerFlowRealtimeData.fcgi": dial tcp 192.168.188.176:80: connect: no route to host

b394373, 2024-11-27:
Still not fixed in v0.131.7...Sent from my Galaxy

  no route to host" error messages (Issue #17251)

andig, 2024-11-29:
See https://github.com/orgs/goreleaser/discussions/5321. I'm not sure why this doesn't work.

andig, 2024-11-29:
Closed in https://github.com/evcc-io/evcc/commit/877a1752eded25b8954002e87c22ebd6cad3335f

b394373, 2024-12-03:
FIXED. Thanks for all your efforts and attention to this issue, it is much appreciated.

[main ] INFO 2024/12/03 10:26:39 evcc 0.131.8
[main ] INFO 2024/12/03 10:26:39 using config file: /Users/Eugene/evcc.yaml
[db ] INFO 2024/12/03 10:26:39 using sqlite database: /Users/Eugene/.evcc/evcc.db
[main ] INFO 2024/12/03 10:26:40 listening at :7070
[site ] INFO 2024/12/03 10:26:41 site config:
[site ] INFO 2024/12/03 10:26:41 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2024/12/03 10:26:41 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/12/03 10:26:41 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2024/12/03 10:26:41 vehicles:
[site ] INFO 2024/12/03 10:26:41 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/12/03 10:26:41 loadpoint 1:
[lp-1 ] INFO 2024/12/03 10:26:41 mode: pv
[lp-1 ] INFO 2024/12/03 10:26:41 charger: power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/12/03 10:26:41 meters: charge ✓
[lp-1 ] INFO 2024/12/03 10:26:41 charge: power ✓ energy ✗ currents ✓
[lp-1 ] DEBUG 2024/12/03 10:26:41 phase timer inactive
[lp-1 ] DEBUG 2024/12/03 10:26:41 pv timer inactive
[site ] DEBUG 2024/12/03 10:26:41 ----
[lp-1 ] DEBUG 2024/12/03 10:26:41 charge power: 5695W
[lp-1 ] DEBUG 2024/12/03 10:26:41 charge currents: [24.1 0.2 0.1]A
[site ] DEBUG 2024/12/03 10:26:41 pv power: 8281W
[site ] DEBUG 2024/12/03 10:26:41 grid power: -834W
[site ] DEBUG 2024/12/03 10:26:41 site power: -834W
[lp-1 ] DEBUG 2024/12/03 10:26:41 charge voltages: [236 0 0]V
[lp-1 ] DEBUG 2024/12/03 10:26:41 detected connected phases: 1p
[lp-1 ] DEBUG 2024/12/03 10:26:41 charger status: C
[lp-1 ] INFO 2024/12/03 10:26:41 car connected
[lp-1 ] DEBUG 2024/12/03 10:26:41 vehicle api refresh
[lp-1 ] DEBUG 2024/12/03 10:26:41 pv timer elapse
[lp-1 ] DEBUG 2024/12/03 10:26:41 pv timer inactive
[lp-1 ] INFO 2024/12/03 10:26:41 start charging ->
[lp-1 ] DEBUG 2024/12/03 10:26:41 wake-up timer: stop
[main ] DEBUG 2024/12/03 10:26:41 vehicle status: C (BlueY)
[lp-1 ] INFO 2024/12/03 10:26:41 vehicle updated: unknown -> BlueY
[lp-1 ] DEBUG 2024/12/03 10:26:41 vehicle status: C
[lp-1 ] DEBUG 2024/12/03 10:26:41 vehicle soc: 66%
[lp-1 ] DEBUG 2024/12/03 10:26:41 vehicle soc limit: 100%
[lp-1 ] DEBUG 2024/12/03 10:26:41 vehicle range: 267km
[lp-1 ] WARN 2024/12/03 10:26:41 charger logic error: current mismatch (got 24.1A measured, expected 0A)
[lp-1 ] DEBUG 2024/12/03 10:26:41 pv charge current: 27.7A = 24.1A + 3.63A (-834W @ 1p)
[lp-1 ] DEBUG 2024/12/03 10:26:41 pv timer reset
[lp-1 ] DEBUG 2024/12/03 10:26:41 pv timer inactive

stonemaan, 2024-12-23:
same issue after updating from 0.131.11 to .12 on Mac OS 15.2 MacMini M1.
Any suggestions?

[main ] INFO 2024/12/23 21:05:24 evcc 0.131.12
[main ] INFO 2024/12/23 21:05:24 using config file: /etc/evcc.yaml
[db ] INFO 2024/12/23 21:05:24 using sqlite database: /Users/stonemaan/.evcc/evcc.db
[mqtt ] INFO 2024/12/23 21:05:27 connecting evcc-194437939 at tcp://192.168.188.122:1883
[mqtt ] DEBUG 2024/12/23 21:05:27 tcp://192.168.188.122:1883 connected
[main ] INFO 2024/12/23 21:05:27 listening at :7070
[main ] FATAL 2024/12/23 21:05:27 cannot create meter 'grid': cannot create meter type 'template': cannot create meter type 'shelly-energymeter': Get "http://192.168.188.88/shelly": dial tcp 192.168.188.88:80: connect: no route to host
[main ] FATAL 2024/12/23 21:05:27 will attempt restart in: 15m0s

andig, 2024-12-23:
Seems mqtt connected just fine? So likely unrelated.

stonemaan, 2024-12-23:
mqtt seems to be connected.
is it possible to downgrade to 0.131.11? as this version worked fine.

stonemaan, 2024-12-23:
just downgraded to 0.131.11 - works fine now. no errors - no changes in any configuration

stonemaan, 2024-12-31:
okay -  added my grid (shelly) in database - works fine now with 0.132.0

andig, 2025-02-20:
@rjpease please open new issue, this has been resolved.

closed 2025-02-20 by andig