evcc-io/evcc #17253: Warning: Planner no matching rate

Ziltoidium, 2024-11-14:
### Describe the bug

Seit Version 0.131.5 erscheint die Warnung "planner no matching rate".
Ich habe keinen Plan aktiviert und nutze die Funktion auch nie.

Wenn man in der Konfigurationsmaske ein paar Tage in die Zukunft konfiguriert und alle möglichen Einträge anpasst, dann werden diese jedoch nicht übernommen.

### Steps to reproduce

1. Ladeplaner öffnen
[image]

2. Ladeplan anpassen
[image]

3. Wechsel in Tab "Ankunft"
[image]

4. Wechsel zurück nach "Abfahrt".  Einträge wurden überschrieben
[image]

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070
log: debug
levels:
  cache: error
plant: xxxxxxxxxxxxxxxxxxxxxxx
interval: 30s
sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxx
telemetry: true
meters:
  - name: my_grid
    type: template
    template: e3dc-rscp
    usage: grid
    host: 10.0.0.201
    port: 5033
    user: xxxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxxxx
    key: xxxxxxxxxxxxxxxxx
  - name: my_pv
    type: template
    template: e3dc-rscp
    usage: pv
    host: 10.0.0.201
    port: 5033
    user: xxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxx
    key: xxxxxxxxxxxxxxxxx
  - name: my_battery
    type: template
    template: e3dc-rscp
    usage: battery
    host: xxxxxxxxxxxxx
    port: 5033
    user: xxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxx
    key: xxxxxxxxxxxxxxxxx
chargers:
  - type: template
    template: heidelberg
    id: 1
    device: /dev/ttyUSB0
    baudrate: 19200
    comset: 80
    modbus: rs485serial
    name: wallbox5
vehicles:
  - type: template
    icon: car
    template: dacia
    title: Dacia Spring
    user: xxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxx
    vin: xxxxxxxxxxxxxxxx
    capacity: 27
    name: ev4
loadpoints:
  - title: Wallbox Garage
    charger: wallbox5
    mode: minpv
site:
  title: Zuhause 🏡
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3367
  feedin:
    type: fixed
    price: 0.0997
  co2:
    type: grünstromindex
    zip: xxxxxxxxxxxxxxxxxx
```

### Log details

```text
[site ] WARN 2024/11/14 04:21:19 planner: no matching rate
[site ] WARN 2024/11/14 04:28:08 planner: no matching rate
[site ] WARN 2024/11/14 04:28:09 planner: no matching rate
[site ] WARN 2024/11/14 04:28:28 planner: no matching rate
[site ] WARN 2024/11/14 04:28:38 planner: no matching rate
[site ] WARN 2024/11/14 04:29:08 planner: no matching rate
[site ] WARN 2024/11/14 04:29:38 planner: no matching rate
[site ] WARN 2024/11/14 04:30:08 planner: no matching rate
[site ] WARN 2024/11/14 04:30:38 planner: no matching rate
[site ] WARN 2024/11/14 04:31:08 planner: no matching rate
[site ] WARN 2024/11/14 04:31:38 planner: no matching rate
[site ] WARN 2024/11/14 04:32:08 planner: no matching rate
[site ] WARN 2024/11/14 04:32:38 planner: no matching rate
[site ] WARN 2024/11/14 04:33:08 planner: no matching rate
[site ] WARN 2024/11/14 04:33:38 planner: no matching rate
[site ] WARN 2024/11/14 04:34:08 planner: no matching rate
[site ] WARN 2024/11/14 04:34:38 planner: no matching rate
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.5

naltatis, 2024-11-14:
Hi @Ziltoidium, diese Meldung kommt, wenn wir in den Grünstromdaten keinen Zeitslot für "jetzt" finden. Deinen Screenshots nach zu urteilen bekommst du ja generell Daten. Ist die Uhrzeit auf deiner Maschine korrekt eingestellt? Tritt das Problem durchgehend auf oder nur für einen bestimmten Zeitraum?

andig, 2024-11-14:
Es braucht bitte

    evcc tariff --log trace,db:error

Ziltoidium, 2024-11-14:
@naltatis auf meinem dietpi war tatsächlich noch timezone London eingestellt und eine Stund Zeitversatz. Das kann natürlich schon die Fehlerquelle sein. Das habe ich soeben korrigiert und den raspi neu gebootet.Jetzt passt die Uhrzeit. Das Problem ist heute morgen zum ersten Mal aufgetaucht. Es scheint, die Pläne werden nun nach Änderungen auch korrekt übernommen, vielleicht ist der Fehler jetzt schon verschwunden, ich werde das mal beobachten.

In der evcc.yaml hier hatte ich die PLZ entfernt.  In meiner config ist sie natürlich eingetragen.

@andig 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v9.8.0 : 08:33 - Thu 11/14/24
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (aarch64)
 - CPU temp : 37 °C / 98 °F : Cool runnings
 - LAN IP : 10.0.0.6 (eth0)
 - MOTD : DietPi v9.8 is here. Check out all changes:
          https://dietpi.com/docs/releases/v9_8/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@Raspi-1-DietPi:~# date
Thu Nov 14 08:33:22 CET 2024
root@Raspi-1-DietPi:~# evcc tariff --log trace,db:error
[main  ] INFO 2024/11/14 08:34:14 evcc 0.131.5
[main  ] INFO 2024/11/14 08:34:14 using config file: /etc/evcc.yaml
grid:
From                   To                     Price/Cost
2024-11-14 00:00:00    2024-11-14 01:00:00    0.337
2024-11-14 01:00:00    2024-11-14 02:00:00    0.337
2024-11-14 02:00:00    2024-11-14 03:00:00    0.337
2024-11-14 03:00:00    2024-11-14 04:00:00    0.337
2024-11-14 04:00:00    2024-11-14 05:00:00    0.337
2024-11-14 05:00:00    2024-11-14 06:00:00    0.337
2024-11-14 06:00:00    2024-11-14 07:00:00    0.337
2024-11-14 07:00:00    2024-11-14 08:00:00    0.337
2024-11-14 08:00:00    2024-11-14 09:00:00    0.337
2024-11-14 09:00:00    2024-11-14 10:00:00    0.337
2024-11-14 10:00:00    2024-11-14 11:00:00    0.337
2024-11-14 11:00:00    2024-11-14 12:00:00    0.337
2024-11-14 12:00:00    2024-11-14 13:00:00    0.337
2024-11-14 13:00:00    2024-11-14 14:00:00    0.337
2024-11-14 14:00:00    2024-11-14 15:00:00    0.337
2024-11-14 15:00:00    2024-11-14 16:00:00    0.337
2024-11-14 16:00:00    2024-11-14 17:00:00    0.337
2024-11-14 17:00:00    2024-11-14 18:00:00    0.337
2024-11-14 18:00:00    2024-11-14 19:00:00    0.337
2024-11-14 19:00:00    2024-11-14 20:00:00    0.337
2024-11-14 20:00:00    2024-11-14 21:00:00    0.337
2024-11-14 21:00:00    2024-11-14 22:00:00    0.337
2024-11-14 22:00:00    2024-11-14 23:00:00    0.337
2024-11-14 23:00:00    2024-11-15 00:00:00    0.337
2024-11-15 00:00:00    2024-11-15 01:00:00    0.337
2024-11-15 01:00:00    2024-11-15 02:00:00    0.337
2024-11-15 02:00:00    2024-11-15 03:00:00    0.337
2024-11-15 03:00:00    2024-11-15 04:00:00    0.337
2024-11-15 04:00:00    2024-11-15 05:00:00    0.337
[... 380 lines cut ...]
2024-11-17 01:00:00    2024-11-17 02:00:00    194.000
2024-11-17 02:00:00    2024-11-17 03:00:00    178.000
2024-11-17 03:00:00    2024-11-17 04:00:00    171.000
2024-11-17 04:00:00    2024-11-17 05:00:00    171.000
2024-11-17 05:00:00    2024-11-17 06:00:00    171.000
2024-11-17 06:00:00    2024-11-17 07:00:00    175.000
2024-11-17 07:00:00    2024-11-17 08:00:00    171.000
2024-11-17 08:00:00    2024-11-17 09:00:00    130.000
2024-11-17 09:00:00    2024-11-17 10:00:00    130.000
2024-11-17 10:00:00    2024-11-17 11:00:00    130.000
2024-11-17 11:00:00    2024-11-17 12:00:00    140.000
2024-11-17 12:00:00    2024-11-17 13:00:00    140.000
2024-11-17 13:00:00    2024-11-17 14:00:00    133.000
2024-11-17 14:00:00    2024-11-17 15:00:00    130.000
2024-11-17 15:00:00    2024-11-17 16:00:00    127.000
2024-11-17 16:00:00    2024-11-17 17:00:00    121.000
2024-11-17 17:00:00    2024-11-17 18:00:00    111.000
2024-11-17 18:00:00    2024-11-17 19:00:00    133.000
2024-11-17 19:00:00    2024-11-17 20:00:00    178.000
2024-11-17 20:00:00    2024-11-17 21:00:00    194.000
2024-11-17 21:00:00    2024-11-17 22:00:00    194.000
2024-11-17 22:00:00    2024-11-17 23:00:00    200.000
2024-11-17 23:00:00    2024-11-18 00:00:00    194.000

root@Raspi-1-DietPi:~#

naltatis, 2024-11-14:
Ok, klingt doch gut. In https://github.com/evcc-io/evcc/pull/17255 hab ich mehr Debug-Info in die Fehlermeldung mit aufgenommen, sodass wir beim nächsten mal einfachern analysieren können. Ich mach hier erst einmal zu. Wenn du wieder komische Effekte beobachtest kommentier gern.

closed 2024-11-14 by naltatis

phrozen77, 2024-12-18:
Kann man hier wieder aufmachen? :)

Hier gibt es ähnlich merkwürdige Phänomene seit 0.131.9, kein Grünstromindex, dafür Tibber - derzeit läuft die 0.131.11

<details><summary>no matching rate</summary>
<p>
Dec 18 10:30:30 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:30:30 planner: outdated
Dec 18 10:30:30 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:30:30 planner: no matching rate for: 2024-12-18T10:30:30+01:00, 24 rates (2024-12-17T00:00:00+01:00 to 2024-12-18T00:00:00+01:00)
Dec 18 10:30:30 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:30 pv power: 884W
Dec 18 10:30:31 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:31 grid power: 1290W
Dec 18 10:30:31 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:31 site power: 1290W
Dec 18 10:30:31 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:31 GET http://192.168.xxx.yyy/api/status?filter=alw,car,eto,nrg,wh,trx,cards
Dec 18 10:30:31 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:31 {"alw":false,"eto":5569574,"car":1,"wh":123.9746,"trx":null,"nrg":[227.85,227.23,227.54,2.48,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":1,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"ca>
Dec 18 10:30:31 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:31 charge total import: 5569.574kWh
Dec 18 10:30:31 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:31 charger status: A
Dec 18 10:30:40 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:40 ----
Dec 18 10:30:40 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:40 GET http://192.168.xxx.yyy/api/status?filter=alw,car,eto,nrg,wh,trx,cards
Dec 18 10:30:40 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:40 {"alw":false,"eto":5569574,"car":1,"wh":123.9746,"trx":null,"nrg":[227.85,227.23,227.54,2.48,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":1,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"ca>
Dec 18 10:30:40 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:40 charge power: 0W
Dec 18 10:30:40 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:40 charge currents: [0 0 0]A
Dec 18 10:30:40 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:30:40 planner: outdated
Dec 18 10:30:40 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:30:40 planner: no matching rate for: 2024-12-18T10:30:40+01:00, 24 rates (2024-12-17T00:00:00+01:00 to 2024-12-18T00:00:00+01:00)
Dec 18 10:30:41 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:41 pv power: 901W
Dec 18 10:30:42 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:42 grid power: 1223W
Dec 18 10:30:42 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:42 site power: 1223W
Dec 18 10:30:42 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:42 GET http://192.168.xxx.yyy/api/status?filter=alw,car,eto,nrg,wh,trx,cards
Dec 18 10:30:42 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:42 {"alw":false,"eto":5569574,"car":1,"wh":123.9746,"trx":null,"nrg":[227.54,227.23,227.54,2.48,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":1,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"ca>
Dec 18 10:30:42 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:42 charge total import: 5569.574kWh
Dec 18 10:30:42 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:42 charger status: A
Dec 18 10:30:50 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:50 ----
Dec 18 10:30:50 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:50 GET http://192.168.xxx.yyy/api/status?filter=alw,car,eto,nrg,wh,trx,cards
Dec 18 10:30:50 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:50 {"alw":false,"eto":5569574,"car":1,"wh":123.9746,"trx":null,"nrg":[227.54,227.23,228.16,2.48,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":1,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"ca>
Dec 18 10:30:50 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:50 charge power: 0W
Dec 18 10:30:50 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:50 charge currents: [0 0 0]A
Dec 18 10:30:50 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:30:50 planner: outdated
Dec 18 10:30:50 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:30:50 planner: no matching rate for: 2024-12-18T10:30:50+01:00, 24 rates (2024-12-17T00:00:00+01:00 to 2024-12-18T00:00:00+01:00)
Dec 18 10:30:51 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:51 grid power: 1268W
Dec 18 10:30:52 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:52 pv power: 907W
Dec 18 10:30:52 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:30:52 site power: 1268W
Dec 18 10:30:52 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:52 GET http://192.168.xxx.yyy/api/status?filter=alw,car,eto,nrg,wh,trx,cards
Dec 18 10:30:52 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:30:52 {"alw":false,"eto":5569574,"car":1,"wh":123.9746,"trx":null,"nrg":[227.54,227.23,227.54,2.48,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":1,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"ca>
Dec 18 10:30:52 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:52 charge total import: 5569.574kWh
Dec 18 10:30:52 evcc evcc[4555]: [lp-1  ] DEBUG 2024/12/18 10:30:52 charger status: A
Dec 18 10:31:00 evcc evcc[4555]: [site  ] DEBUG 2024/12/18 10:31:00 ----
Dec 18 10:31:00 evcc evcc[4555]: [go-e  ] TRACE 2024/12/18 10:31:00 GET http://192.168.xxx.yyy/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[... cut ...]
</details> 

Die IP ist seit mehreren Tagen dieselbe, ich hab also nicht zufälligerweise nach Zwangstrennung o.ä. eine geblacklistete bekommen.

Nichts außer evcc (gut, die originale App noch, aber die nutze ich kaum) im lokalen Netzwerk frägt sonst bei Tibber an - es ist also eine selbst verursachte Wunde...

Beim letzten Problem kamen von Tibber auch HTTP 429er zurück - ähnliche wie im obigen Log ist es vermutlich nicht sonderlich hilfreich dann trotzdem beharrlich im x-Sekunden-Takt anzufragen?

andig, 2024-12-18:
Ich sehe hier keinen Diskussionsbedarf:

    Dec 18 10:31:00 evcc evcc[4555]: [site  ] WARN 2024/12/18 10:31:00 planner: no matching rate for: 2024-12-18T10:31:00+01:00, 24 rates (2024-12-17T00:00:00+01:00 to 2024-12-18T00:00:00+01:00)

es gibt keine Rate für den aktuellen Zeitpunkt, daher die Warnung. Alles korrekt!

Tibber API wäre ein separates Thema- da hatte wohl der Server Schluckauf und das war hier ursächlich...

phrozen77, 2024-12-18:
Ok, dann ist dieses Verhalten inklusive Warnungen in der GUI relativ neu?

Und: schön finde ich das deswegen trotzdem nicht. Was soll man als Enduser mit der Meldung anfangen? Was kann ich als Enduser konkret dagegen tun?