evcc-io/evcc #17254: Enphase multiple batteries - jq: too many results

scholy, 2024-11-14:
### Describe the bug

After installing 3 enphase solar batteries and reconfiguring evcc with the new controller, I see all the solar production and consumption data, but the battery SOC is not available. Checking through the logs, I can see evcc polls and receives the battery data from the IQ controller (GET https://192.168.3.126/ivp/ensemble/inventory), but it looks like evcc is expecting to see only 1 battery, producing the following error logs: 
[site  ] ERROR 2024/11/14 16:41:29 battery 1 soc: jq: too many results
[site  ] DEBUG 2024/11/14 16:41:29 battery soc: 0%
[site  ] DEBUG 2024/11/14 16:41:29 battery power: 0W

### Steps to reproduce

1. configure enphase batter meter
2.  configure site meters
3. restart evcc
4. 
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 60s # control cycle interval

sponsortoken: <snip>

log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - name: enphase_pv
    type: template
    template: enphase
    usage: pv
    host: 192.168.3.126
    token: <snip>
  - name: enphase_grid
    type: template
    template: enphase
    usage: grid
    host: 192.168.3.126
    token: <snip>
  - name: enphase_battery
    type: template
    template: enphase
    usage: battery
    host: 192.168.3.126
    token: <snip>
    capacity: 15

chargers:
  - name: TeslaCharger
    type: template # Wallbe charger
    template: twc3
    host: 192.168.3.200

vehicles:
  - name: ZROGAS
    type: template
    template: tesla
    icon: car
    commandProxy: http://192.168.1.78:8080
    accessToken: <snip>
    refreshToken: <snip>
[... 288 lines cut ...]
[enphase_grid] TRACE 2024/11/14 17:14:06 {"production":[{"type":"inverters","activeCount":25,"readingTime":1731564828,"wNow":1092,"whLifetime":3259},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1731564846,"wNow":977.991,"whLifetime":2716.431,"varhLeadLifetime":378.572,"varhLagLifetime":159.077,"vahLifetime":2901.147,"rmsCurrent":4.649,"rmsVoltage":721.078,"reactPwr":-338.643,"apprntPwr":1111.89,"pwrFactor":0.88,"whToday":2716.431,"whLastSevenDays":71.431,"vahToday":2901.147,"varhLeadToday":378.572,"varhLagToday":159.077}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1731564846,"wNow":3817.776,"whLifetime":4375.72,"varhLeadLifetime":996.672,"varhLagLifetime":340.675,"vahLifetime":4227.765,"rmsCurrent":17.949,"rmsVoltage":721.294,"reactPwr":-342.813,"apprntPwr":4315.922,"pwrFactor":0.88,"whToday":4375.72,"whLastSevenDays":256.72,"vahToday":4227.765,"varhLeadToday":996.672,"varhLagToday":340.675},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1731564846,"wNow":2839.785,"whLifetime":2128.285,"varhLeadLifetime":1375.244,"varhLagLifetime":499.752,"vahLifetime":4227.765,"rmsCurrent":13.301,"rmsVoltage":721.294,"reactPwr":-4.17,"apprntPwr":3197.011,"pwrFactor":0.89,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
[site  ] DEBUG 2024/11/14 17:14:06 grid power: 2840W
[enphase_pv] TRACE 2024/11/14 17:14:08 {"production":[{"type":"inverters","activeCount":25,"readingTime":1731564828,"wNow":1092,"whLifetime":3259},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1731564848,"wNow":975.066,"whLifetime":2717.0,"varhLeadLifetime":378.784,"varhLagLifetime":159.08,"vahLifetime":2901.799,"rmsCurrent":4.643,"rmsVoltage":721.165,"reactPwr":-360.793,"apprntPwr":1118.314,"pwrFactor":0.87,"whToday":2717.0,"whLastSevenDays":72.0,"vahToday":2901.799,"varhLeadToday":378.784,"varhLagToday":159.08}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1731564848,"wNow":3817.053,"whLifetime":4377.946,"varhLeadLifetime":996.732,"varhLagLifetime":340.967,"vahLifetime":4229.629,"rmsCurrent":17.943,"rmsVoltage":721.379,"reactPwr":-306.994,"apprntPwr":4314.775,"pwrFactor":0.89,"whToday":4377.946,"whLastSevenDays":258.946,"vahToday":4229.629,"varhLeadToday":996.732,"varhLagToday":340.967},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1731564848,"wNow":2841.988,"whLifetime":2129.942,"varhLeadLifetime":1375.516,"varhLagLifetime":500.047,"vahLifetime":4229.629,"rmsCurrent":13.3,"rmsVoltage":721.379,"reactPwr":53.798,"apprntPwr":3197.198,"pwrFactor":0.89,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
[site  ] DEBUG 2024/11/14 17:14:08 pv power: 1092W
[site  ] DEBUG 2024/11/14 17:14:08 site power: 2840W
[lp-1  ] DEBUG 2024/11/14 17:14:09 charge voltages: [4.7 1.1 0]V
[lp-1  ] DEBUG 2024/11/14 17:14:09 charger status: A
[lp-1  ] INFO 2024/11/14 17:14:09 car disconnected
[lp-1  ] DEBUG 2024/11/14 17:14:09 set charge mode: pv
[lp-1  ] DEBUG 2024/11/14 17:14:09 set battery boost: false
[lp-1  ] DEBUG 2024/11/14 17:14:09 set session soc limit: 0
[lp-1  ] DEBUG 2024/11/14 17:14:09 set session energy limit: 0
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.5 (6836749c)

andig, 2024-11-14:
The template's jq query needs be updated and a battery number added. Happy to take a PR.

scholy, 2024-11-14:
Could the issue be the 2 top level arrays of ENPOWER and ENCHARGE? Both return devices, only ENCHARGE is will return percentFull values.

Is the intended bahaviour to average the percentFull values? I'm curious as the log files seems to indicate battery numbers"
`ERROR 2024/11/14 17:14:05 **battery 1 soc**: jq: too many results
[site  ] DEBUG 2024/11/14 17:14:05 battery soc: 0%
[site  ] DEBUG 2024/11/14 17:14:05 battery power: 0W
`

I'll try playing around with a custom meter and the `jq`

scholy, 2024-11-14:
Fixed in a custom meter with:
`jq: '[.[].devices[] | select(.percentFull != null) | .percentFull] | add / length'`

premultiply, 2024-11-14:
Every battery unit should be added as own device for configuration. evcc will sum up the measurands internally where required.

andig, 2024-11-15:
Bitte nicht komplizierter machen als notwendig. Hier gehts erstmal überhaupt um die richtige Formel, ees fehlt nur der PR.

premultiply, 2024-11-15:
Eben, daher ja die Bitte keine komplizierten und ggf. falschen Formeln zu verwenden sondern einfach mit dem jeweiligen Index auf die jeweilige Speicher-Einheit einzeln zuzugreifen.
Hier fehlt nämlich derzeit die Kapazität der einzelnen Module, so dass der SoC bei unterschiedlichen Speichereinheiten falsch wäre.

andig, 2024-11-15:
Darum geht es hier aber nicht. Bitte nicht alles durcheinander mischen. Bei Stapelspeichern passt das wunderbar.

premultiply, 2024-11-15:
Das sind mwn. alles einzelne Speichereinheiten mit AC-Anschluss, keine gestapelten DC-Module.

andig, 2024-11-15:
Yep, hat mit diesem Issue aber trotzdem nix zu tun 😉

scholy, 2024-11-16:
Struggling to follow, maybe there is 2 things here:
#1 is the bug/enhancement I've seen and the solution to the problem - working on a PR for this.

#2 is a feature request to record, track and report on each battery individually. I see this as a benefit too given I have 3 batteries connected to 3 different phases that all discharge at separate rates based on which household circuits are connected to each phase. 

Happy to support and contribute to the latter, in the mean time I believe I've identified a solution for the first issues and will request PR.

andig, 2024-11-16:
Closed in a9d9565e62c05842c097a983bf8901423fc4d55f

closed 2024-11-16 by andig