evcc-io/evcc #17268: Easee Home lädt im PV Modus ohne Überschuss

Janjek84, 2024-11-14:
### Describe the bug

Seit einigen Wochen ist mir aufgefallen, dass die Easee Home WB lädt, obwohl kein Überschuss vorhanden ist. 
Also habe ich mal auf die aktuellsten  EVCC Version upgedatet. 
Dr Fehler ist aber weiterhin da. Heute wurde geladen, obwohl es dunkel war.

Hier der Log. 
[evcc-20241114-195506-debug.log.txt](https://github.com/user-attachments/files/17757079/evcc-20241114-195506-debug.log.txt)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Wird nachgereicht, wenn benötigt.
```

### Log details

```text
Ist als Datei angehängt.
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-11-16:
Ist im Log eine falsche Stromvorgabe durch evcc zu erkennen?

andig, 2024-11-16:
Siehe auch https://github.com/evcc-io/evcc/issues/17280, Fehler bei Easee?

Janjek84, 2024-11-16:
Hallo, 

> Ist im Log eine falsche Stromvorgabe durch evcc zu erkennen?
Es sieht mir nicht danach aus...
Bei available Power steht ein Minuszeichen...

          [easee ] WARN 2024/11/14 19:53:28 smart charging: timeout
          [lp-1  ] DEBUG 2024/11/14 19:53:28 charge power: 0W
          [lp-1  ] DEBUG 2024/11/14 19:53:28 charge currents: [0.009 0.008 0.008]A
          [site  ] DEBUG 2024/11/14 19:53:28 grid power: 1700W
          [site  ] DEBUG 2024/11/14 19:53:28 pv power: 0W
          [site  ] DEBUG 2024/11/14 19:53:28 site power: 1700W
          [lp-1  ] DEBUG 2024/11/14 19:53:28 charge total import: 2877.745kWh
          [easee ] WARN 2024/11/14 19:53:48 smart charging: timeout
          [lp-1  ] DEBUG 2024/11/14 19:53:48 charger status: B
          [lp-1  ] DEBUG 2024/11/14 19:53:48 !! pvScalePhases DOWN activePhases: 1, available power: -1700W, scalable: false
          [lp-1  ] DEBUG 2024/11/14 19:53:48 !! pvScalePhases UP maxPhases: 3, available power: -1700W, scalable: false
          [lp-1  ] DEBUG 2024/11/14 19:53:48 !! pvScalePhases EXIT
          [lp-1  ] DEBUG 2024/11/14 19:53:48 pv charge current: 0A = 0A + -7.39A (1700W @ 1p)
          [site  ] DEBUG 2024/11/14 19:53:48 ----
          [lp-2  ] DEBUG 2024/11/14 19:53:48 charge power: 0W
          [easee ] WARN 2024/11/14 19:54:08 smart charging: timeout
          [lp-1  ] DEBUG 2024/11/14 19:54:08 charge power: 0W
          [lp-1  ] DEBUG 2024/11/14 19:54:08 charge currents: [0.009 0.008 0.008]A
          [site  ] DEBUG 2024/11/14 19:54:08 grid power: 1707W
          [site  ] DEBUG 2024/11/14 19:54:08 pv power: 0W
          [site  ] DEBUG 2024/11/14 19:54:08 site power: 1707W
          [lp-2  ] DEBUG 2024/11/14 19:54:08 charge total import: 0.000kWh
          [lp-2  ] DEBUG 2024/11/14 19:54:09 charger status: B
          [lp-2  ] DEBUG 2024/11/14 19:54:09 pv charge current: 0A = 0A + -7.42A (1707W @ 1p)
          [site  ] DEBUG 2024/11/14 19:54:09 ----
          [lp-2  ] DEBUG 2024/11/14 19:54:09 charge power: 0W
          [easee ] WARN 2024/11/14 19:54:29 smart charging: timeout
          [lp-1  ] DEBUG 2024/11/14 19:54:29 charge power: 0W
          [lp-1  ] DEBUG 2024/11/14 19:54:29 charge currents: [0.009 0.008 0.008]A
          [site  ] DEBUG 2024/11/14 19:54:29 grid power: 1693W
          [site  ] DEBUG 2024/11/14 19:54:29 pv power: 0W
          [site  ] DEBUG 2024/11/14 19:54:29 site power: 1693W
          [lp-1  ] DEBUG 2024/11/14 19:54:29 charge total import: 2877.745kWh
          [easee ] WARN 2024/11/14 19:54:49 smart charging: timeout
          [lp-1  ] DEBUG 2024/11/14 19:54:49 charger status: B
          [lp-1  ] DEBUG 2024/11/14 19:54:49 !! pvScalePhases DOWN activePhases: 1, available power: -1693W, scalable: false
          [lp-1  ] DEBUG 2024/11/14 19:54:49 !! pvScalePhases UP maxPhases: 3, available power: -1693W, scalable: false
          [lp-1  ] DEBUG 2024/11/14 19:54:49 !! pvScalePhases EXIT
          [lp-1  ] DEBUG 2024/11/14 19:54:49 pv charge current: 0A = 0A + -7.36A (1693W @ 1p)
          [site  ] DEBUG 2024/11/14 19:54:49 ----
          [lp-2  ] DEBUG 2024/11/14 19:54:49 charge power: 0W

Heute wurde nochmal geladen: 
     
[image]
[image]
[image]

Den kompletten Log vom 14.11. und von heute 16.11 habe ich als txt. File angehängt.

Liegt der Fehler vielleicht bei der Easee Home? Aber da wurde nichts geändert an den Einstellungen. 

[evcc-20241114-195506-debug.log (1).txt](https://github.com/user-attachments/files/17787858/evcc-20241114-195506-debug.log.1.txt)
[evcc-20241116-132229-debug.log.txt](https://github.com/user-attachments/files/17787859/evcc-20241116-132229-debug.log.txt)

Vielen Dank und beste Grüße 
Janjek84

GrimmiMeloni, 2024-11-17:
Hier sieht man bereits in den Screenshot die Timeouts. Wenn es sonst funktioniert hat, würde ich hier erstmal auf ein Easee seitiges Problem tippen. Falls es andauert kann es ggf. noch einen Versuch wert sein, daß Timeout für die Easee nochmal von 20s auf 30s zu erhöhen.