evcc-io/evcc #17277: evcc.db disappears after update to 0.131.5

floppybl, 2024-11-15:
### Describe the bug

Hi all,

after updating evcc running in docker on unraid to current version the evcc.db was deleted during the update process. Is this a known problem? And will this happen again on future updates?

### Steps to reproduce

1. update docker container from unraid gui
2. after update all things are gone in evcc gui except of settings in yaml (admin password, cars etc.

### Configuration details

```yaml
network:
  schema: http
  host: XXX.XXX.XXX.XXX
  port: 7070

interval: 30s # control cycle interval

sponsortoken: XXX

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug

loadpoints

meters:
  - name: my_pv
    type: template
    template: kostal-piko
    usage: pv      
    host: XXX.XXX.XXX.XXX # Hostname

  - name: my_grid
    type: template
    template: kostal-ksem
    usage: grid      
    
    # Modbus TCP
    modbus: tcpip
    id: 71
    host: XXX.XXX.XXX.XXX # Hostname
    port: 502 # Port 

chargers:
  - name: my_charger_right
    type: template
    template: easee
    user: mail@mail # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: secretpw # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    charger: EXXXXXXL  
  - name: my_charger_left
    type: template
    template: easee
    user: mail@mail # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: secretpw # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    charger: EXXXXXXT

site:
  title: Home # display name for UI
  meters:
    grid: my_grid # grid meter
    pv:
      - my_pv # list of pv inverters/ meters

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Wallbox links # display name for UI
    charger: my_charger_left # charger
    mode: pv # set default charge mode, use "off" to disable by default if charger is publicly available
    # vehicle: car1 # set default vehicle (disables vehicle detection)
    # resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
#    minCurrent: 6 # minimum charge current (default 6A)
#    maxCurrent: 16 # maximum charge current (default 16A)
  - title: Wallbox rechts # display name for UI
    charger: my_charger_right # charger
#    vehicle: my_modely
    mode: pv # set default charge mode, use "off" to disable by default if charger is publicly available
    # vehicle: car1 # set default vehicle (disables vehicle detection)
    # resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
#    minCurrent: 6 # minimum charge current (default 6A)
#    maxCurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    # type: fixed
    # price: 0.294 # EUR/kWh

    # or variable via tibber
    type: tibber
    # cheap: 0.2 # EUR/kWh
    token: token # access token
    # homeid: "xxx" # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # cheap: 0.2 # EUR/kWh
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07 # EUR/kWh
```

### Log details

```text
no warnings or errors in log
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

StefanSchoof, 2024-11-15:
You need to mount it, see https://docs.evcc.io/en/docs/installation/docker#volumes

floppybl, 2024-11-15:
It is all correctly mounted like unraid let set the mounting points. It worked for weeks with updating from 0.131.3 to 0.131.4 without touching the database.

[image]