evcc-io/evcc #17280: easee: evcc constantly switches between pause and resume

MaTr74, 2024-11-15:
### Describe the bug

As a car was connected evcc switched constantly between pause and resume. The log shows logic errors on phases and currents.

I don't know where the 32 A is coming from. The circuit is configured to 32 A but the charger to 25 A. The car was loading at 1 phase.
[image]
[image]

### Steps to reproduce

1. plug-in car

### Configuration details

```yaml
site:
  - title: Hotel
    meters:
      grid: grid-meter
      pv:
      - sungrow1
      - sungrow2

loadpoints:
  - title: Wallbox-links
    charger: easee-master
    mode: minpv
    enable: 
      threshold: 0
      delay: 1m
    disable:
      threshold: 6000
      delay: 3m
      #phases: 3
    priority: 1
      #    mincurrent: 6
      #    maxcurrent: 32
  - title: Wallbox-rechts
    charger: easee-slave
    mode: minpv
    enable: 
      threshold: 0
      delay: 1m
    disable:
      threshold: 6000
      delay: 3m
        #phases: 3
    priority: 1
      #    mincurrent: 6
      #    maxcurrent: 32
  - title: Klimatruhe
    charger: klimatruhe
    mode: off
    enable:
      threshold: 0
      delay: 1m
    disable:
      threshold: 100
      delay: 3m
    guardduration: 5m
      #phases: 1
      #    mincurrent: 4 
      #    maxcurrent: 4 
    priority: 2 
    meter: sonoff-powr2-01

chargers:
  - name: easee-master
    type: template
    template: easee
    user: **** 
    password: ****
    charger: ABCD
  - name: easee-slave
    type: template
    template: easee
    user: ****
    password: ****
    charger: EFGH

meters:
  - name: grid-meter
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.2.64
  - name: sungrow1
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.2.75
    port: 502
  - name: sungrow2
    type: template
    template: sungrow-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.2.76

interval: 30s
log: debug
levels:
#  site: info
#  Terasse: info 
#  easee-master: trace 
#  easee-slave: trace
  easee: trace
#  grid-meter: info
  cache: error
  influx: info

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: "****"
    homeid: "****"
  feedin: 
    type: fixed
    price: 0.0754

mqtt:
  broker: mosquitto:1883
  topic: evcc

influx:
  url: http://influxdb:8086
  database: evcc/default
  token: ****
  org: waldperle

plant: ****
sponsortoken: ****
telemetry: true
```

### Log details

```text
attached
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

andig, 2024-11-16:
Does the log show any wrong current setting from evcc side?

andig, 2024-11-16:
Siehe auch https://github.com/evcc-io/evcc/issues/17268. Fehler bei Easee?

GrimmiMeloni, 2024-11-17:
At first glance this reminds me of https://github.com/evcc-io/evcc/pull/16116.
When the Easee is requested more Amps than physcially configured, it does not ignore the request, but instead crashes/resets. When it comes back to life, it will go full power. 

Based no the config you have not configured the max amp limit of 25A.
While the above PR added some checks to adhere to this automatically, there could still be corner cases.
I suggest you explicitly configure the Loadpoint to only have 25A to match your installation, this should prevent you from hitting this problem.

GrimmiMeloni, 2024-11-17:
Some more remarks. "Something" is clearly starting Loadpoint-1 without evcc knowing about it. Depending on your configuration the box might automatically start on connection (if authentication on the charger itself is off). That would be expected.
However, the following charger logic error is very suspicious:

> evcc    | [lp-1  ] WARN 2024/11/15 13:13:30 charger logic error: disabled but charging

The reason is, that unless you reconnected the car in that timeframe (which I doubt), it shows that evcc pauses the charge due to the mismatch in phases. The box follows that command just fine, however the next time we look at the box, it has "magically" started the charge.

I highly recommend you dumb the setup down first. Also, I see you have a tibber tariff. Have you also given Tibber access to your Easee's? It is known to cause issues.

MaTr74, 2024-11-17:
@GrimmiMeloni Thanks for your investigation!

Since the last successful loading session the easee charge berry was replaced after a damage, but the configuration was completely overtaken and never changed after that: https://github.com/evcc-io/evcc/issues/15556#issuecomment-2365107179

The evcc configuration is still set to max 25 A 
[image]

After observing this unwanted behaviour of pausing and resuming, I restarted evcc so that the charge session was already going.

Tibber hasen't any access to the chargers.

From my point of view I didn't change anything except the charge berry replacement. As the charge session was smoothly ongoing after I shutted down evcc I would assume that the hardware is ok.

My question would be what exactly was the reason for pausing and resuming the session? Realy the phase mismatch or  the max current mismatch? What is if the car decided to still keep 1 phase even 3 phases were possible?

GrimmiMeloni, 2024-11-18:
> My question would be what exactly was the reason for pausing and resuming the session? Realy the phase mismatch or the max current mismatch? 

Definitely the phase mismatch. That is because in setups like yours (multiple Easee chargers on a single circuit) we have to control the phases on charger (not on circuit) level. Switching phases like this requires a charge pause to become effective. 

> What is if the car decided to still keep 1 phase even 3 phases were possible?

Currently it would behave like above, and detect this as a phase mismatch. Probably this is not what we want.

@andig what is the contract for `api.PhaseGetter` ? Shall it return the phases allowed/possible to use according to what evcc configured, or shall it return the number of phases actually in use? Or specifically in this case (3p active on charger, but car only uses 1p) what is the expected result from the PhaseGetter: 1 or 3 ?

premultiply, 2024-11-18:
> what is the contract for api.PhaseGetter ? Shall it return the phases allowed/possible to use according to what evcc configured

Yes.

Detection what phases are in use from vehicle side is automatically done on the currents.

GrimmiMeloni, 2024-11-18:
Thanks @premultiply.

@MaTr74 congrats, you found a bug. ;) I'll prepare a PR in a sec to fix this.

GrimmiMeloni, 2024-11-18:
@MaTr74 I need to wait for the car to return to test, before I can ask for the merge. If you are interested, you can grab a current master build including #17326 via the following container image`grimmimeloni/evcc:nightly`

PR #17326 (2024-11-19): Easee: fix PhaseGetter returning used, not configured, phases 
fix #17280

marking as draft until I can test it locally later today
closed 2024-11-19 by andig