evcc-io/evcc #17298: Tibber Tarif wird nicht mehr verwendet

Staschewsk, 2024-11-17:
### Describe the bug

Seit gestern werden die dynamischen Strompreise von Tibber nicht mehr verwendet. Ich habe gestern Abend auch das neue Update installiert. Kann aber nicht sagen, ob es davor schon nicht mehr ging. Folgende Warnung kommt jetzt immer 
[image] 

Danke im Voraus 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
evcc.yaml
```

### Log details

```text
[site ] WARN 2024/11/17 10:43:40 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'tibber': Message: Post "https://api.tibber.com/v1-beta/gql": context deadline exceeded, Locations: [], Extensions: map[code:request_error], Path: []
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

naltatis, 2024-11-17:
@Staschewsk Ist das ein anhaltendes Problem bei dir? Bei mir macht die Tibber API was sie soll, und ich bekomme den Fehler nicht. Sonst ggf. noch mal deine Tibber Zugangsdaten kontrollieren (Token, Homeid)

Staschewsk, 2024-11-17:
Das Problem besteht seit gestern Abend. Wie schon gesagt, habe ich gestern Abend auch die Version 0.131.6 von EVCC installiert. Tibber Token habe ich kontrolliert. Tibber läuft auch auf HA komplett Problemlos.

StefanSchoof, 2024-11-17:
Dann probiere mal ein Downgrade, um zu prüfen ob es an evcc oder doch an was anderem liegt.

Staschewsk, 2024-11-17:
Ich habe jetzt ein Downgrade durchgeführt. Bei Version 0.131.4 geht es wieder.

andig, 2024-11-18:
> Ich habe jetzt ein Downgrade durchgeführt. Bei Version 0.131.4 geht es wieder.

Schwer zu glauben. Da hat sich nix geändert.

andig, 2024-11-18:
Ich mache hier mal zu. Falls noch relevant bitte Tibber Zugangsdaten an info@evcc.io.

andig, 2024-11-18:
Was zu tun ist steht oben. Alles andere ist Zeitverschwendung.

Staschewsk, 2024-11-18:
Also muss ich meine vertraulichen Zugangsdaten (auch Passwort?) an Dritte weitergeben, die ich nicht mal kenne? Nein Danke. Keine zufriedenstellende Lösung. Danke für nichts. Ich verwende erstmal die funktionierende Version.

andig, 2024-11-18:
Wenn du dich wieder beruhigt hast darfst du gerne einen sachdienlichen Kommentar hinterlassen. Anderenfalls bitte ich dich es dabei bewenden zu lassen. Danke.

Staschewsk, 2024-11-18:
Würdest du deine Zugangsdaten an dritte schicken? Ernstgemeinte Frage.

andig, 2024-11-19:
Was schlägst Du zur Fehlersuche denn _alternativ_ vor? Überraschenderweise hat das Entwicklerteam nicht jeden Tarif und jedes Gerät selbst installiert. Ansonsten: selbstverständlich nach persönlichem Kontakt.

Ich mache hier zu- es gibt keine neuen Erkenntnisse.