evcc-io/evcc #17299: Options dialog flickering under Android 15 on Pixel 7

jahoooo, 2024-11-17:
### Describe the bug

When evcc is charging and I open the dialog to switch between charging duration, solar usage, costs, ... (see video

https://github.com/user-attachments/assets/643fdd0f-6dd9-4ab7-aca9-567223299474

), the dialog starts flickering.
I can still use the dialog and select the options, but it does not look so nice.

### Steps to reproduce

On Pixel 7 with Android 15 (I think it also happened before under Android 14 as well) when accessing the evcc webpage.
1. evcc in charging state
2. open the dialog to switch between charging duration, solar usage, costs, ...
3. screen flickers constantly

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  # Homeassistant lokale IP Adresse ist:
  host: 192.168.0.81
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

plant: ***

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features
sponsortoken: ***

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  polestar: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
# modbusproxy:
[... 305 lines cut ...]
[site  ] DEBUG 2024/11/17 09:17:29 grid power: -333W
[site  ] DEBUG 2024/11/17 09:17:29 grid powers: [360 -565 -127]W
[site  ] DEBUG 2024/11/17 09:17:29 pv power: 1768W
[site  ] DEBUG 2024/11/17 09:17:29 grid currents: [2.41 -2.79 -0.833]A
[site  ] DEBUG 2024/11/17 09:17:29 site power: -333W
[lp-1  ] DEBUG 2024/11/17 09:17:29 charge total import: 1382.009kWh
[lp-1  ] DEBUG 2024/11/17 09:17:29 charger status: B
[lp-1  ] DEBUG 2024/11/17 09:17:29 !! pvScalePhases DOWN activePhases: 1, available power: 333W, scalable: false
[lp-1  ] DEBUG 2024/11/17 09:17:29 !! pvScalePhases UP maxPhases: 3, available power: 333W, scalable: false
[lp-1  ] DEBUG 2024/11/17 09:17:29 !! pvScalePhases EXIT
[lp-1  ] DEBUG 2024/11/17 09:17:29 pv charge current: 1.45A = 0A + 1.45A (-333W @ 1p)
[site  ] DEBUG 2024/11/17 09:17:59 ----
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.6

naltatis, 2024-11-17:
Does this also happen if you access evcc without Home Assistant context?
What browser are you using?
Does this also happen, when accessing https://demo.evcc.io/ ?

Can't reproduce it on my Pixel 6a, Android 15, Chrome.

jahoooo, 2024-11-17:
Your are right, it is a browser issue!
So for me this issue can be closed.
Does not happen when accessing from home assistant or from chrome.
Only happens when accessing through Firefox. I had a shortcut link to evcc with Firefox on my main android screen. Solved now by using Chrome.

closed 2024-11-17 by naltatis