evcc-io/evcc #17303: not starting after update

hallerdavid, 2024-11-17:
### Describe the bug

I updated to the actual version, first the sponsored token was not valid, i changed it with the new one, but then there is an error in loading my charger and evcc is not starting.

How can i check with nightly build? have to tick the box^^

### Steps to reproduce

1. update
2. start evcc

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 3s

javascript:
  - vm: shared
    script: |
      state = {
        residualpower: 500,
        pvpower: 5000,
        batterypower: -750,
        batterySoc: 55,
        gridpower: -1000,
        loadpoints: [
          { enabled: true, vehicleSoc: 62, maxcurrent: 6, phases: 1, chargepower: 0 },
          { enabled: false, vehicleSoc: 22, maxcurrent: 0, phases: 3, chargepower: 0 }
        ]
      };
      function logState() {
        console.log("state:", JSON.stringify(state));
      }

meters:
  - name: grid
    type: template
    template: alpha-ess-smile
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 85
    host: 10.10.30.62
    port: 502

  - name: pv
    type: template
    template: alpha-ess-smile
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 85
    host: 10.10.30.62
    port: 502

  - name: battery
    type: template
    template: alpha-ess-smile
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 85
    host: 10.10.30.62
    port: 502

chargers:
  - name: keba
    type: template
    template: keba
    host: 10.10.30.25

vehicles:
  - name: vehicle_id4gtx
    title: ID.4 GTX
    type: template
    template: vw
    user: david@hallerdavid.de
    password: T2iPAXmayes48X

site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
    battery: battery

loadpoints:
  - title: Garage
    charger: keba
    mode: "off"
    #meter: grid
    vehicle: vehicle_id4gtx
    enable:
      delay: 1m
      threshold: -4500
    disable:
      delay: 15m
      threshold: 1000

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.4219
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh

sponsortoken: TOKEN
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/11/17 12:58:26 evcc 0.131.6
[main  ] INFO 2024/11/17 12:58:26 using config file: /config/evcc.yaml
[db    ] INFO 2024/11/17 12:58:26 using sqlite database: /data/evcc.db
[main  ] INFO 2024/11/17 12:58:27 listening at :7070
[main  ] FATAL 2024/11/17 12:58:28 cannot create charger 'keba': cannot create charger type 'template': cannot create charger type 'keba-udp': recv timeout
[main  ] FATAL 2024/11/17 12:58:28 will attempt restart in: 15m0s
 Live
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

0x3d13f, 2024-11-17:
The wallbox does not answer. I suggest restarting the wallbox (disconnect from power and reconnect).

closed 2024-11-18 by andig

hallerdavid, 2024-11-18:
you where right, sorry for disturbing. I just wondered why it is not reachable. Every other Smarthome app was able to connect to, also ping and the keba app worked successfull.

But, i restarted it via the fuse in the electric cabinet and now its connecting successfully to evcc again. Thank you!