evcc-io/evcc #17307: Vehicle Data: Different values are displayed in Home Screen and Configuration

mho21git, 2024-11-17:
### Describe the bug

Correct values are shown in Configuration:
[image]
Correct: lädt - 68 km - 27 %

Outdated and wrong values are displayed in Home Screen:
[image]
Wrong: verbunden - 74 km - 30 %

Also the values trasferred to MQTT-Server are not correct.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Configuration is done in UI, not in evcc.yaml.
```

### Log details

```text
Log file for vehicle is empty.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.6

naltatis, 2024-11-17:
Hast du `estimate` am Ladepunkt aktiviert?
https://docs.evcc.io/docs/reference/configuration/loadpoints#estimate

mho21git, 2024-11-18:
Ja, in evcc.yaml ist estimate auf true gesetzt.
[image]

naltatis, 2024-11-18:
Ok, das erklärt den Unterschied. Der Estimator probiert anhand von Ladeleistung und konfigurierter Akkugröße des Fahrzeugs auch zwischen den Abfrageintervallen des Fahrzeugs den Akkustand zu interpolieren. Abhängig von deiner Umgebung (Verluste, Konditionierung, Brutto/Netto, ...) kann der geschätzte Wert dann durchaus abweichen. Wenn du `estimator: false` machst hast du immer den Wert von der API.

closed 2024-11-18 by naltatis

mho21git, 2024-11-22:
OK, ich habe estimate: false gesetzt und mehrfach neu gestartet.
Es bleibt dabei. Im Home und Configuration werden unterschiedliche Werte angezeigt.
Im MQTT Explorer werden auch die falschen Werte aus dem Home angezeigt.

naltatis, 2024-11-23:
@mho21git Kannst du konkreter werden und mal ein Log posten.

mho21git, 2024-11-24:
Auch mit Release v0,131.7 besteht der Fehler weiterhin.

Ich habe ein kleines Video aufgenommen: 
https://github.com/user-attachments/assets/f146ba35-25d7-4dc4-bfba-7905f128948a
Wenn man im Home das Fahrzeug wechselt und wieder zurück, dann kommen die korrekten Werte.

Anbei auch ein Trace.
[evcc-20241124-174240-trace.log](https://github.com/user-attachments/files/17894282/evcc-20241124-174240-trace.log)

Danke!