evcc-io/evcc #17309: BMW Climate function

hyperbart, 2024-11-17:
**Is your feature request related to a problem? Please describe.**
I don't see my BMW i5 with a climate status in EVCC. As far as I know this should be possible?

**Describe the solution you'd like**
Add climate visibility/control of BMW newer models to EVCC

andig, 2024-11-18:
Unless somebody wants to add this- not planned.

closed 2024-11-18 by andig

hyperbart, 2025-01-09:
@mfuchs1984 i have the new version of evcc including your fix and a BMW i series car but I don’t see anything happening or visible for clima. What can I investigate?

mfuchs1984, 2025-01-09:
You need to setup the poll mode of the loadpoint to `connected` and reduce the poll interval to e.g. 15 minutes, otherwise evcc does not get the information from the API.

https://docs.evcc.io/en/docs/reference/configuration/loadpoints#soc

hyperbart, 2025-01-10:
Thanks @mfuchs1984 but I am still not seeing anything visually on climate in EVCC for my car. I have looked at logs but also don't see any event passing by which indicates that EVCC is aware of the climate function. What is the log content/rule/line I should be able to find and on what area?

Config is as you mentioned:

[image]

mfuchs1984, 2025-01-10:
Mode should be `connected`, not `connect`.

mfuchs1984, 2025-01-10:
I also think the loadpoint has to be in PV mode.

hyperbart, 2025-01-10:
That was the error, indeed. Connect should have been connected. Thank you VERY much!

[image]

tilibibi, 2025-01-11:
I was reading this and I was curious to try. So I put the following in my config: 
```
loadpoints:
- title: Garage door
  charger: wallbox4
  vehicle: ev1
  mode: pv
  soc:
    poll:
      mode: connected
      interval: 15m
    estimate: true
```
and I'm in PV mode, but still I do not see any option for my BMW X1. 
My trace log for BMW looks like this: 

```
[bmw ] TRACE 2025/01/11 21:17:19 POST https://customer.bmwgroup.com/gcdm/oauth/token
[bmw ] TRACE 2025/01/11 21:17:19 grant_type=refresh_token&redirect_uri=com.bmw.connected%3A%2F%2Foauth&refresh_token=xxxxx -- {"gcid":"0d234107-8be7-4141-ab9c-f2bbc355ce3e","token_type":"Bearer","access_token":"xxxxx","refresh_token":"xxxxx","scope":"smacc vehicle_data perseus dlm svds openid profile vsapi remote_services authenticate_user cesim offline_access email fupo","expires_in":3599,"id_token":"xxxxx"}
[bmw ] TRACE 2025/01/11 21:17:19 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1736626639040
[bmw ] TRACE 2025/01/11 21:17:19 {"state":{"isLeftSteering":true,"lastFetched":"2025-01-11T20:17:19.427Z","lastUpdatedAt":"2025-01-11T15:55:37Z","isLscSupported":true,

......

"climateControlState":{"activity":"INACTIVE"},"climateTimers":[{"isWeeklyTimer":false,"timerAction":"DEACTIVATE","timerWeekDays":[],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}},{"isWeeklyTimer":true,"timerAction":"DEACTIVATE","timerWeekDays":["MONDAY"],"departureTime":{"hour":7,"minute":0}}],"departurePlan":{},"requiredServices":[],"checkControlMessages":[{"type":"TIRE_PRESSURE","severity":"LOW"},{"type":"ENGINE_OIL","severity":"LOW"}],"chargingProfile":{"chargingControlType":"WEEKLY_PLANNER","reductionOfChargeCurrent":{"start":{"hour":10,"minute":0},"end":{"hour":14,"minute":0}},"chargingMode":"IMMEDIATE_CHARGING","chargingPreference":"NO_PRESELECTION","departureTimes":[{"id":1,"timeStamp":{"hour":0,"minute":0},"action":"DEACTIVATE","timerWeekDays":[]},{"id":2,"timeStamp":{"hour":0,"minute":0},"action":"DEACTIVATE","timerWeekDays":[]},{"id":3,"timeStamp":{"hour":0,"minute":0},"action":"DEACTIVATE","timerWeekDays":[]},{"id":4,"timeStamp":{"hour":8,"minute":18},"action":"DEACTIVATE","timerWeekDays":[]}],"climatisationOn":true,"chargingSettings":{"targetSoc":100,"acCurrentLimit":6,"idcc":"AUTOMATIC_INTELLIGENT","hospitality":"HOSP_INACTIVE","isAcCurrentLimitActive":false}},"electricChargingState":{"chargingLevelPercent":38,"range":19,"isChargerConnected":true,"chargingStatus":"NOT_CHARGING","chargingTarget":100},"combustionFuelLevel":{"remainingFuelPercent":80,"range":556},"driverPreferences":{"lscPrivacyMode":"OFF"},"securityOverviewMode":"ARMED","vehicleSoftwareVersion":{"puStep":{"month":11,"year":24},"iStep":{"seriesCluster":"U006","year":24,"month":11,"iStep":540}},"pwf":"STANDING_CUSTOMER_NOT_IN_VEH"},"capabilities":{"remoteServices":{"departureTimerControl":{"id":"departureTimerControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"inCarCamera":{"id":"inCarCamera","state":"NOT_AVAILABLE","executionApi":"RS_360_V4"},"inCarCameraDwa":{"id":"inCarCameraDwa","state":"NOT_AVAILABLE","executionApi":"RS_360_V4"},"remote360":{"id":"remote360","state":"ACTIVATED","executionApi":"RS_360_V4","functions":{"encryptionMethod":["ENCRYPTION_METHOD_A"]}},"surroundViewRecorder":{"id":"surroundViewRecorder","state":"ACTIVATED","executionApi":"RS_360_V4","functions":{"encryptionMethod":["ENCRYPTION_METHOD_A"]}},"windowControl":{"id":"windowControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"centralLockControl":{"id":"centralLockControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"batteryPreconditioningControl":{"id":"batteryPreconditioningControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"doorLock":{"id":"doorLock","state":"ACTIVATED","executionApi":"RS_API_V6"},"doorUnlock":{"id":"doorUnlock","state":"ACTIVATED","executionApi":"RS_API_V6"},"hornBlow":{"id":"hornBlow","state":"ACTIVATED","executionApi":"RS_API_V6"},"lightFlash":{"id":"lightFlash","state":"ACTIVATED","executionApi":"RS_API_V6"},"telematicsWakeup":{"id":"telematicsWakeup","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"wakeup":{"id":"wakeup","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"},"interiorPreconditioningControl":{"id":"interiorPreconditioningControl","state":"NOT_AVAILABLE","executionApi":"RS_API_V6"}},"a4aType":"NOT_SUPPORTED","climateNow":true,"isClimateTimerWeeklyActive":false,"climateFunction":"AIR_CONDITIONING","horn":true,"isBmwChargingSupported":true,"isCarSharingSupported":false,"isChargeNowForBusinessSupported":true,"isChargingHistorySupported":true,"isLocationBasedChargingSettingsSupported":false,"isChargingHospitalityEnabled":true,"isChargingLoudnessEnabled":true,"isChargingPlanSupported":true,"isChargingPowerLimitEnabled":true,"isChargingSettingsEnabled":true,"isBatteryPreconditioningSupported":false,"isChargingTargetSocEnabled":true,"isCustomerEsimSupported":false,"isDataPrivacyEnabled":false,"isDCSContractManagementSupported":true,"isEasyChargeEnabled":false,"isMiniChargingSupported":false,"isEvGoChargingSupported":true,"isRemoteHistoryDeletionSupported":false,"isRemoteEngineStartSupported":false,"isRemoteServicesActivationRequired":false,"isRemoteServicesBookingRequired":false,"isScanAndChargeSupported":true,"lastStateCallState":"ACTIVATED","lights":true,"lock":true,"remote360":true,"remoteSoftwareUpgrade":true,"sendPoi" [... cut ...] d":false,"personalPictureUpload":{"state":"NOT_AVAILABLE","aspectRatio":"2:1"},"isPlugAndChargeSupported":false,"isOptimizedChargingSupported":false,"alarmSystem":true,"isThirdPartyAppStoreSupported":false,"thirdPartyAppStore":{"state":"NOT_AVAILABLE"},"locationBasedCommerceFeatures":{"parking":true,"fueling":true,"reservations":false}}} 

```
I see some "climate ..." entries but not sure ...   
Any ideas?

BTW ... We also have a Mini ... could this be implemented there as well? 

Thanks and BR.

mfuchs1984, 2025-01-11:
`"climateControlState":{"activity":"INACTIVE"}`
Climating was not active.

Just to make sure there is not misconception: it is not possible to start climating through evcc. The climater API in evcc just makes sure to start the charger when you start climating through the car app or from the car infotainment while it is connected to the charger.

tilibibi, 2025-01-11:
Ok, got it. Thanks for the reply. 
I actually thought, you could start the climating via evcc. Would be a nice option to have it included in the planner :-)
Just to get it right. If I program a pre-conditioning via the BMW app at a certain hour, evcc is switching then from e.g. PV mode to Min+PV?

PR #18114 (2025-01-07): BMW: add climater
This PR implements the api.VehicleClimater interface for BMW.

Fixes #17309.