evcc-io/evcc #17321: Sponsortoken als Github Organisation

nick-b81, 2024-11-18:
### Describe the bug

Hi,

für unsere 5x Mennekes Wallboxen nutzen wir EVCC. Getestet hatten wir mit dem Test-Token. Nun haben wir über die Organisation https://github.com/Lueling-Altena einen Sponsor-Token aktivieren wollen. Allerdings bekomme ich über die sponsor.evcc.io keinen Token. Ich muss mich dort natürlich mit meinem eigenen GitHib-User anmelden. 

### Steps to reproduce

1. Anlage meines Benutzers
2. Anlage Organisation
3. GitHub Sponsoring über das Unternehmen aktiviert
[image]
5. Abwarten, bis die Zahlung erfolgt ist
6. Token über sponsor.evcc.io abrufen > Anmelden (User Account)
7. Es wird kein Sponsoring gefunden
[image]

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
nicht relevant
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.3

closed 2024-11-18 by naltatis

nick-b81, 2024-11-18:
> Hi @nick-b81, magst du dich einmal über [info@evcc.io](mailto:info@evcc.io) melden? Dann schauen wir uns das an.

Habe ich gemacht, bislang aber keine Antwort bekommen. Deshalb melde ich mich hier

naltatis, 2024-11-18:
@nick-b81 Mail mit deinem Token ist raus. Der Fall "Sponsoring via Organisation" klappt aktuell bisher nicht automatisiert bei uns.