evcc-io/evcc #17327: Netzladen Batterie erzeugt Fehler und Störungen

amigaron, 2024-11-18:
### Describe the bug

Wenn ich die Funktion Netzladen aktiviere erhalte ich vom evcc die Fehlermeldung "battery mode: write tcp 192.168.3.200:44990->192.168.3.220:502: write: broken pipe"
Gestern Abend wurde die Batterie dennoch vom System geladen - allerdings auch über das angekündigte Zeitfenster von 1h hinaus.
Seitdem steht der Hausbatteriestatus bei 100% und wird nicht mehr entladen.
SYSTEM: Sungrow Sh5.0RT und BYD Batterien
Der WR meldet als Betriebsstatus: "Zwangsmodus"
Ein Neustart von evcc, Batterie und WR bräuchten keine Verbesserung.

### Steps to reproduce

1.Hausbatterie -> Netzladen einstellen
2. Fehlermeldung "broken pipe" erscheint spätestens wenn Grenzwert unterschritten
3.
...

### Configuration details

```yaml
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxxxx
sponsortoken: xxxx

interval: 30s # control cycle interval

mqtt:
  broker: localhost:1883
  topic: evcc
  user: xxx
  password: xxx
modbusproxy:
- port: 5200
  uri: 192.168.3.220:502
  # rtu: true
  # readonly: true

meters:
- type: template
  template: tibber-pulse
  usage: grid
  token: Xxxx
  homeid: xxxx
  name: grid1
type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.3.220
  port: 502
  usage: grid
  modbus: tcpip
  name: grid2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.3.220
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.3.220
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: sunspec-inverter
  id: 1
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv4
- type: template
  template: sunspec-inverter
  id: 2
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv5
- type: template
  template: sunspec-inverter
  id: 3
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv6
- type: template
template: sunspec-inverter
  id: 4
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv7
- type: template
  template: sunspec-inverter
  id: 5
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv8
- type: template
  template: sunspec-inverter
  id: 6
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv9
- type: template
  template: sunspec-inverter
  id: 7
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  modbus: tcpip
  name: pv10
- type: template
  template: sunspec-inverter
  id: 8
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv11
- type: template
  template: sunspec-inverter
  id: 9
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv12
- type: template
  template: sunspec-inverter
  id: 10
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv13
- type: template
  template: sunspec-inverter
  id: 11
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv14
- type: template
  template: sunspec-inverter
  id: 12
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv15
- type: template
  template: sunspec-inverter
  id: 13
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv16
- type: template
  template: sunspec-inverter
  id: 14
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv17
- type: template
  template: sunspec-inverter
  id: 15
host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv18
- type: template
  template: sunspec-inverter
  id: 16
  host: 192.168.3.218
  port: 502
  usage: pv
  modbus: tcpip
  name: pv19

chargers:
- type: template
  template: go-e-v3
  host: 192.168.3.215
  name: Phoenixfeder
- type: template
  template: go-e-v3
  host: 192.168.3.216
  name: Einhornhaar

#circuits:
#- name: Wallbox
#  title: Wallboxen
#  maxCurrent: 16
#  maxPower: 11000
#  meter: grid

loadpoints:
- title: Phoenixfeder
  charger: Phoenixfeder
  priority: 1
  mode: off
  phases: 0
#  mincurrent: 6
#  maxcurrent: 16
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
- title: Einhornhaar
  charger: Einhornhaar
  priority: 0
  mode: off
  phases: 0
#  mincurrent: 6
#  maxcurrent: 16
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

vehicles:
- title: Marvel
  name: ev1
type: template
  template: mg
  icon: car # Icon in der Benutzeroberfläche (optional)
  user: xxxxxx# Benutzerkonto (bspw. E-Mail Adresse, User I
d, etc.)
  password: xxx# Passwort des Benutzerkontos (bei führenden Nullen
bitte in einfache Hochkommata setzen)
  vin: xxxxxx# Erforderlich
  capacity: 70 # Akkukapazität in kWh (optional)
  phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (op
tional)
  cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt wer
den (optional)
  mode: PV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner defin
iert werden soll (optional)
  minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phas
e mit welcher das Fahrzeug geladen werden soll (optional)
  maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Pha
se mit welcher das Fahrzeug geladen werden soll (optional)
  identifiers: # Kann meist erst später eingetragen werden, siehe: https://do
cs.evcc.io/docs/features/vehicle (optional)
  priority: 1 # Priorität des Ladepunktes oder Fahrzeugs in Relation zu ander
en Ladepunkten oder Fahrzeugen für die Zuweisung von PV-Energie (optional)Mar

site:
  title: Home
  meters:
    grid: grid2
    pv:
    - pv4
    - pv5
    - pv6
    - pv7
    - pv8
    - pv9
    - pv10
    - pv11
    - pv12
    - pv13
    - pv14
    - pv15
    - pv16
    - pv17
    - pv18
    - pv19
    battery:
    - battery3
  residualPower: 100
#  bufferSoC: 90
#  prioritySoC: 50

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: tibber
   # cheap: 0.20 # [currency]/kWh # soll ab 0,20 € laden
    token: xxxx
    homeid: xxxx
#  grid:
#    # static grid price
#    type: fixed
#    price: 0.3331 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07343 # [currency]/kWh
```

### Log details

```text
site ] DEBUG 2024/11/18 12:01:26 pv 16 power: 37W
[site ] DEBUG 2024/11/18 12:01:26 pv 8 power: 21W
[site ] DEBUG 2024/11/18 12:01:26 pv 4 power: 16W
[site ] DEBUG 2024/11/18 12:01:26 pv 9 power: 26W
[site ] DEBUG 2024/11/18 12:01:26 pv 5 power: 24W
[site ] DEBUG 2024/11/18 12:01:26 pv 1 power: 12W
[site ] DEBUG 2024/11/18 12:01:26 pv 10 power: 18W
[site ] DEBUG 2024/11/18 12:01:26 pv 6 power: 21W
[site ] DEBUG 2024/11/18 12:01:26 pv 2 power: 16W
[site ] DEBUG 2024/11/18 12:01:26 pv 11 power: 35W
[site ] DEBUG 2024/11/18 12:01:26 pv 7 power: 18W
[site ] DEBUG 2024/11/18 12:01:26 pv 12 power: 34W
[site ] DEBUG 2024/11/18 12:01:26 pv 3 power: 19W
[site ] DEBUG 2024/11/18 12:01:26 pv 13 power: 37W
[site ] DEBUG 2024/11/18 12:01:26 pv 14 power: 37W
[site ] DEBUG 2024/11/18 12:01:26 pv 15 power: 33W
[site ] DEBUG 2024/11/18 12:01:26 pv power: 404W
[site ] DEBUG 2024/11/18 12:01:26 site power: 2118W
[lp-2 ] DEBUG 2024/11/18 12:01:26 charge total import: 1121.512kWh
[lp-2 ] DEBUG 2024/11/18 12:01:26 charger status: A
[site ] DEBUG 2024/11/18 12:01:27 set grid charge limit: 0.3
[site ] DEBUG 2024/11/18 12:01:44 ----
[lp-1 ] DEBUG 2024/11/18 12:01:44 charge power: 0W
[lp-1 ] DEBUG 2024/11/18 12:01:44 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/11/18 12:01:44 charge power: 0W
[lp-2 ] DEBUG 2024/11/18 12:01:44 charge currents: [0 0 0]A
[site ] ERROR 2024/11/18 12:01:44 battery mode: write tcp 192.168.3.200:46794->192.168.3.220:502: write: broken pipe
[site ] DEBUG 2024/11/18 12:01:44 grid power: 1856W
[site ] DEBUG 2024/11/18 12:01:44 battery soc: 98%
[site ] DEBUG 2024/11/18 12:01:44 battery power: 0W
[site ] DEBUG 2024/11/18 12:01:44 grid currents: [1.21 1 5.61]A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.6

naltatis, 2024-11-18:
Ist evcc die einzige Software, die mit der Modbus Schnittstelle spricht oder hast du noch irgend eine andere Integration (bspw. Home Assistant, ...) laufen, die darauf zugreift?

amigaron, 2024-11-18:
Nein, noch nicht. Ich plante genau diese Funktion mit OpenHab abzubilden (zusätzlich mit Wetterforecast), habe das aber noch nicht realisiert. Daher die Aktivierung des Modbus-Proxys

Ich glaube ich habe auch die Ursache gefunden: 
evcc schaltet den Sungrow in den Zwangsmodus zum Laden der Batterie, schaltet aber am Ende nicht mehr zurück auf Eigenverbrauchsmodus.
Somit wartet der WR auf den Entladebefehl, statt selber dies zu regeln.
Unglücklicher Weise kann die den Modus aber scheinbar nicht selber wieder im WR ändern..

andig, 2024-11-18:
Duplikat zu https://github.com/evcc-io/evcc/discussions/17069. Leider ist niemand in der Lage, das auf Kommando zu reproduzieren.