evcc-io/evcc #17352: EVCC lädt Auto obwohl kein Überschuss vorhanden ist

skyguy2002, 2024-11-19:
### Describe the bug

EVCC lädt das auto trotz "PV" modus, obwohl kein Überschuss erkennbar war.

### Steps to reproduce

1. Auto anstecken (10 Uhr am Vormittag)
2. Modus auf PV
3. Plan erstellt, damit das Auto morgens definitiv voll ist (Geplant für nächsten morgen um 7 Uhr)

4. EVCC lädt das Auto obwohl noch genug Zeit vorhanden wäre das eventuell Sonne noch rauskommen könnte

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 30s

sponsortoken: XXX

meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.40.150 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: pv
    type: template
    template: sungrow-hybrid
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.40.150 # Hostname
    port: 502 # Port
    timeout: 10s # optional

  - name: battery
    type: template
    template: sungrow-hybrid
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.40.150 # Hostname
    port: 502 # Port
    timeout: 10s # optional

vehicles:
  - name: Kuga2
    type: custom
    title: Kuga
    icon: car
    capacity: 14
    phases: 1
    soc:
      source: http
      uri: http://192.168.40.150:8123/api/states/sensor.fordpass_elveh
      method: GET
      headers:
        - authorization: Bearer XXX
        - content-type: application/json
      jq: .attributes."Battery Charge" | tonumber
    range:
      source: http
      uri: http://192.168.40.150:8123/api/states/sensor.fordpass_elveh
      method: GET
      headers:
        - authorization: Bearer XXX
        - content-type: application/json
      jq: .state | tonumber
    odometer:
      source: http
      uri: http://192.168.40.150:8123/api/states/sensor.fordpass_odometer
      method: GET
      headers:
        - authorization: Bearer XXX
        - content-type: application/json
      jq: .state | tonumber

site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
    battery: battery

loadpoints:
  - title: Garage # display name for UI
    charger: sungrow # charger reference
    mode: off # charge mode (off, now, minpv, pv)
chargers:
  - name: sungrow
    type: template
    template: sungrow
    
    # RS485 via adapter (Modbus RTU)
    modbus: rs485serial
    id: 248
    device: /dev/ttyACM0 # USB-RS485 Adapter Adresse
    baudrate: 9600 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter für den Adapter

# mqtt message broker
mqtt:
  broker: 192.168.40.150:1883
  topic: evcc
  user: evcc
  password: XXX

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.2858
  feedin:
    type: fixed
    price: 0.082 # EUR/kWh
  co2:
    type: grünstromindex
    zip: XXX
```

### Log details

```text
siehe anhang
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.6

skyguy2002, 2024-11-19:
[ EVCC.txt](https://github.com/user-attachments/files/17820367/EVCC.txt)

skyguy2002, 2024-11-19:
Settings Hausbatterie:

[image]

StefanSchoof, 2024-11-19:
Du hast einen CO2 Tarif und dann lädt evcc CO2 optimiert. Das kann bedeuten, dass eine Ladung jetzt weniger CO2 verursacht als eine direkt vor der Abfahrt. Das sollte dann im Planungsansicht gezeigt werden.

skyguy2002, 2024-11-20:
> Du hast einen CO2 Tarif und dann lädt evcc CO2 optimiert. Das kann bedeuten, dass eine Ladung jetzt weniger CO2 verursacht als eine direkt vor der Abfahrt. Das sollte dann im Planungsansicht gezeigt werden.

Das heißt das EVCC CO2 Verbrauch über PV Überschuss sieht ?