evcc-io/evcc #17367: `evcc tariff` ohne Ergebnis, wenn Tarife im UI konfiguriert

VolkerK62, 2024-11-20:
### Describe the bug

Ich habe meine Tarife im UI konfiguriert.
Es wird alles richtig angezeigt und smartes Laden funktioniert.
Rufe ich aber auf der Kommandozeile `evcc tariff` auf, kommt kein Ergebnis.
bug or feature?

[image]

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
pi@raspberrypi:~ $ evcc tariff
[main  ] INFO 2024/11/20 17:49:39 evcc 0.131.6 (9083a846)
[main  ] INFO 2024/11/20 17:49:39 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/20 17:49:39 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2024/11/20 17:49:40 connecting evcc-1070015338 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/11/20 17:49:40 tcp://localhost:1883 connected
pi@raspberrypi:~ $
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-11-20:
@VolkerK62 was bedeutet der Screenshot?

VolkerK62, 2024-11-21:
Als "Beleg" für die Konfig.

andig, 2024-11-21:
@naltatis die Tarife funktionieren ja etwas anders- willst Du das evtl. anschauen?

naltatis, 2024-11-22:
Erwartungskonform wäre es natürlich wenn `evcc tariff` hier auch die UI konfigurierten Werte nutzt. Funktioniert das denn bei `vehicle` und `meter` aktuell wie erwartet?

VolkerK62, 2024-11-22:
`meter` und `vehicle`(offline) funktionieren wie erwartet.

andig, 2024-11-25:
@naltatis ich hab grad einen Klemmer. Wo werden die Tarife aus der DB _überhaupt_ initialisiert? Ich hab eher das Gefühl, dass es den Teil im Code noch gar nicht gibt? Siehe `configureTariffs`?

naltatis, 2024-11-25:
@andig in `configureTariffs` passiert doch genau das:
https://github.com/evcc-io/evcc/blob/d7e87bf207dc0cc7b483453924c8bac1466335d2/cmd/setup.go#L807

Wenns einen Eintrag in der DB gibt wird der in conf übernommen.

andig, 2024-11-29:
@VolkerK62 hast Du die falsche DB erwischt? Sonst bräuchte ich bitte mal Deine zum Test.

VolkerK62, 2024-11-29:
Ich hab nur eine DB (in der yaml fixiert)
DB schick ich dir in Slack.

andig, 2024-11-29:
Mal wieder wunderbar gefunden, danke @VolkerK62