evcc-io/evcc #17374: SMA Sunny Island grid charge leads to illegal data value

deadrabbit87, 2024-11-20:
### Describe the bug

Grid charge with SMA Sunny Island over modbus leads to `illegal data value`

### Steps to reproduce

n/a

### Configuration details

```yaml
meters:
  - name: my_battery
    type: template
    template: sma-si-modbus
    usage: battery
    modbus: tcpip
    id: 3
    host: 192.168.178.48
    port: 502 
```
```

### Log details

```text
Output of `.\evcc.exe -c .\evcc.yaml meter -l trace --battery-mode charge`: 

[main  ] INFO 2024/11/20 20:17:52 evcc 0.131.6
[main  ] INFO 2024/11/20 20:17:52 using config file: .\evcc.yaml
[db    ] INFO 2024/11/20 20:17:52 using sqlite database: C:\Users\maximilian.hauser\.evcc\evcc.db
[db    ] TRACE 2024/11/20 20:17:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT * FROM `settings` 6 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/11/20 20:17:52 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[my_battery] TRACE 2024/11/20 20:17:52 modbus: send 00 01 00 00 00 0b 03 10 9c d5 00 02 04 80 00 00 01
[my_battery] TRACE 2024/11/20 20:17:52 modbus: recv 00 01 00 00 00 03 03 90 03
[main  ] FATAL 2024/11/20 20:17:52 set battery mode: modbus: exception '3' (illegal data value), function '16'
```
```

### What type of operating system are you running?

Windows

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.6

deadrabbit87, 2024-11-20:
Hierzu gibts mehrere Issues, ich blicke da aber noch nicht ganz durch. 

Nach der [Doku ](https://files.sma.de/downloads/SI-Modbus-BA-en-12.pdf) passen die Register. 

Hier https://github.com/evcc-io/evcc/pull/11909 wurde das ganze ja schon mal umgebaut. 

Lässt sich bei dem Batteriewechselrichter nicht gleich Sunspec nutzen? 

Ausgabe von `.\evcc.exe -l trace --id 126 sunspec 192.168.178.48:502`: 
[info.txt](https://github.com/user-attachments/files/17836171/info.txt)

Herbert23, 2024-11-21:
Ich habe was den Sunny Island angeht diese Woche etwas herumprobiert.

Mit der aktuellsten FW-Version 3.30.37.R auf dem SI solle es ausreichen, wie beim Sunny Boy Storage, die Wirkleistungsbegrenzung zu setzen (Register 41255 beim SI).

deadrabbit87, 2024-11-21:
Konntest du das bei dir mit Erfolg testen? 

Ich habe das mal so gebaut, Fehler erhalte ich da dennoch: 
`[main  ] FATAL 2024/11/21 13:19:15 set battery mode: modbus: exception '2' (illegal data address), function '6'`

Herbert23, 2024-11-21:
Welche Firmware Version läuft denn auf deinem Sunny Island? Das Problem hatte ich glaub ich auch, bevor ich das FW-Update gemacht habe.

Bei mir läuft das jetzt seit dieser Woche und funktioniert so

Herbert23, 2024-11-21:
> Konntest du das bei dir mit Erfolg testen?
> 
> Ich habe das mal so gebaut, Fehler erhalte ich da dennoch: `[main ] FATAL 2024/11/21 13:19:15 set battery mode: modbus: exception '2' (illegal data address), function '6'`
> 
> Nach der Doku die ich gefunden habe, gibt es das Register `41255` gar nicht.
> 
> > Nach der [Doku ](https://files.sma.de/downloads/SI-Modbus-BA-en-12.pdf) passen die Register.

Durch den edit:

[Doku ](https://files.sma.de/downloads/MODBUS-HTML_SI44M-80H-13_32009R_V10.zip) hab ich diese genutzt

deadrabbit87, 2024-11-21:
Danke. Ich überprüfe das nochmal und melde mich.

deadrabbit87, 2024-11-21:
Hm, die Firmware wäre sogar `4.0.0.R`.

Sunny Island 6.0H-11 wäre das genaue Modell.

Herbert23, 2024-11-21:
Das erklärt das ganze, ich hab nen SI 80H-12, da ist die FW ne andere und dann auch die Modbusregister.

Die Doku, die du gelinkt hast, gilt aber auch nur für die 12er und 13er Serie.

Für die 11er Serie wäre dann hier die [Doku](https://files.sma.de/downloads/SMA_Modbus-de-en_V25.zip?_gl=1*lyyyg4*_gcl_au*MTA0MzkwOTE2LjE3MzAxMjA0ODI.) mit der hab ich aber dementsprechend noch nicht gearbeitet

premultiply, 2024-11-21:
Lass uns das dann bei allen WR konfigurierbar machen und als Default 4200W nehmen damit keine zusätzliche dynamische Dimmung nach §14a EnWG erforderlich ist.

Eigentlich müsste man das wohl auch perspektivisch dynamisch durch das LM regeln lassen...

deadrabbit87, 2024-11-21:
Und wie die Unterscheidung zwischen Version -11 und -12 des WR?

@Herbert23 oder funktioniert das bestehende Template bei dir auch, wenn du niedrigere Werte schreibst?

andig, 2024-11-21:
> Lass uns das dann bei allen WR konfigurierbar machen und als Default 4200W nehmen damit keine zusätzliche dynamische Dimmung nach §14a EnWG erforderlich ist.

Ich wiederhole mich da- lasst uns die Themen bitte nicht mischen. Danke.

premultiply, 2024-11-21:
Das Ding ist halt dass SMA offensichtlich selbst den Überblick verloren hat welcher WR bzw. welche Hard- und Softwareversion was kann. Dazu gab es bei aktuelleren Versionen gerade in dem Bereich Änderungen bei manchen Modellen - gerade im Zusammenspiel mit dem SHM.
Die SMA-Doku ist leider abweichend, unvollständig und fehlerhaft.

Echt schwierig da einen Überblick zu behalten.

PR #17393 (2024-11-25): SMA: fix battery control
Fixes #17115
Fixes #17374
Fixes #17201
closed 2024-11-25 by andig