evcc-io/evcc #17376: Min+PV Modus lädt mit maximaler Leistung

Ziltoidium, 2024-11-20:
### Describe the bug

Laut [Doku](https://docs.evcc.io/docs/features/solar-charging#minpv-modus) soll evcc die Ladung mit minimaler Leistung beginnen. Das wären in meinem Fall 6A.  Es wird aber unerwünschterweise mit 16A geladen.  Lediglich wenn ich die Wallbox pauschal in den Einstellungen begrenze, dann wird auf 6A runtergeregelt. Die Option [mincurrent in der evcc.yaml ist laut Doku obsolet](https://docs.evcc.io/docs/reference/configuration/loadpoints#mincurrent).

### Steps to reproduce

1.Minimaler Ladestrom 6A, maximal 16A.  Ich habe eine Heidelberg mit 3phasigem Anschluss. Der Dacia kann jedoch nur 1phasig laden
[image]

2.Der Dacia lädt mit 16A statt der minimalen erwarteten 6A
[image]

3.Begrenzung in den Einstellungen auf maximal 6A
[image]

4. erst jetzt durch manuelle Festlegung laden mit 6A
[image]

...

### Configuration details

```yaml
loadpoints:
  - title: Wallbox Garage
    charger: wallbox5
    mode: minpv
```

### Log details

```text
evcc -l debug meter
[main  ] INFO 2024/11/20 21:26:27 evcc 0.131.6
[main  ] INFO 2024/11/20 21:26:27 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/20 21:26:27 using sqlite database: /root/.evcc/evcc.db
my_grid
-------
Power: 1718W

my_pv
-----
Power: 0W

my_battery
----------
Power:        -0W
Soc:          1%
Controllable: true
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.6

andig, 2024-11-20:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-11-20 by andig

Ziltoidium, 2024-11-20:
Log file
[evcc-20241120-213055-debug.log](https://github.com/user-attachments/files/17836437/evcc-20241120-213055-debug.log)

andig, 2024-11-20:
Does the log show any peoblem? At which timestamp?

Ziltoidium, 2024-11-20:
it shows this, but no idea if this is about min+pv settings!?

time="2024-11-20T20:53:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:54:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:54:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:54:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:54:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:54:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:54:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:54:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:54:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:55:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:55:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:55:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:55:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:55:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:55:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:55:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:55:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:56:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:56:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:56:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:56:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:56:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:56:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:56:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:56:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:57:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:57:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:57:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:57:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:57:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:57:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:57:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:57:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:58:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:58:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:58:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:58:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:58:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:58:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:58:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:58:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:59:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:59:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:59:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:59:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T20:59:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T20:59:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T20:59:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T20:59:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T21:00:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T21:00:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T21:00:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T21:00:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T21:00:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[... cut ...]
time="2024-11-20T21:34:47+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T21:34:47+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-11-20T21:35:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-20T21:35:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-20T21:35:17+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-20T21:35:17+01:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"

Ziltoidium, 2024-11-20:
> Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

@andig 
i added the log. i hope this was the right kind of information. actually i am not sure this causes errors in logs, it is more like an unexpected behaviour. no idea, if my interpretation of min+pv is right, what it should do

mfuchs1984, 2024-11-21:
Das liegt wohl an der eingestellten CO2 Grenze für sauberes Netzladen.

eckerse, 2024-11-21:
@Ziltoidium As already stated, you selected a CO2 limit. If actual CO2 content is below your selected limit it will charge in fast mode - it should do it in PV as well as Min+PV mode as this is your preferred charging regime.