evcc-io/evcc #17377: Polestar add vehicle failed with 404 after restart of docker container

ottomitschke, 2024-11-20:
### Describe the bug

Hi, 
after having restarted the container a few times after testing some configs, evcc is not longer able to create the vehicle and is only sending: 
ERROR 2024/11/20 22:48:36 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot get vehicles: Message: 404 Not Found, Locations: [], Extensions: map[code:request_error], Path: []

Since there is no option in the app or web of Polestar to maybe reset all connections, I am moreless locked in. 

The integration was working for some hours, so not an issue in general indeed.

Complete setup is a KEBA P30-x Wallbox and a Polestart 2 ( Yes I am a Sponsor and I have successfully set my token )

### Steps to reproduce

1. Create vehicle in the config
2. works like a sharm
3. Restart container (perhaps restart login at Polestar.com too often)
...

### Configuration details

```yaml
vehicles:
- type: template
  template: polestar 
  title: Polestar  
  icon: car  
  user: email@user.cc  
  password: STRONGPW
  vin: LPSVS...
  capacity: 78  
  phases: 3  
  cache: 15m  
  mode: minpv  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4
```

### Log details

```text
INFO 2024/11/20 22:48:34 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/20 22:48:34 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/11/20 22:48:34 listening at :7070
[main  ] ERROR 2024/11/20 22:48:36 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot get vehicles: Message: 404 Not Found, Locations: [], Extensions: map[code:request_error], Path: []
[lp-1  ] DEBUG 2024/11/20 22:48:37 charge total import: 16678.903kWh
[site  ] INFO 2024/11/20 22:48:37 site config:
[site  ] INFO 2024/11/20 22:48:37   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/11/20 22:48:37     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/11/20 22:48:37     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/11/20 22:48:37     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/11/20 22:48:37   vehicles:
[site  ] INFO 2024/11/20 22:48:37     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/11/20 22:48:37 loadpoint 1:
[lp-1  ] INFO 2024/11/20 22:48:37   mode:        minpv
[lp-1  ] INFO 2024/11/20 22:48:37   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/11/20 22:48:37   meters:      charge ✓
[lp-1  ] INFO 2024/11/20 22:48:37     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/11/20 22:48:37 phase timer inactive
[lp-1  ] DEBUG 2024/11/20 22:48:37 pv timer inactive
[lp-1  ] DEBUG 2024/11/20 22:48:37 max charge current: 6A
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.6

andig, 2024-11-21:
Please show trace log.

kptkip, 2024-11-21:
It seems that Polestar again changed the API during several days without communication.

I assume this because today I got the information from Home Assistant, that there is a new version of the Polestar Integration. No clue, what they changes this time 🙄

ottomitschke, 2024-11-21:
> trace log.

Hi, its a docker container, not sure how to show more log as the provided one ?

andig, 2024-11-21:
Please read docs how to do this.

closed 2024-11-21 by andig

ottomitschke, 2024-11-21:
> Please read docs how to do this. We're not providing Docker support- it's a complex technology.

`    environment:
      - log_level=trace
    restart: unless-stopped`

The output is moreless the same like in the initial post:
`[+] Running 1/0
 ✔ Container EVCC  Recreated                                                                                                                                                                              0.1s 
Attaching to EVCC
EVCC  | [main  ] INFO 2024/11/21 20:17:02 evcc 0.131.6
EVCC  | [main  ] INFO 2024/11/21 20:17:02 using config file: /etc/evcc.yaml
EVCC  | [db    ] INFO 2024/11/21 20:17:02 using sqlite database: /root/.evcc/evcc.db
EVCC  | [mqtt  ] INFO 2024/11/21 20:17:02 connecting evcc-1118776634 at tcp://172.17.17.10:1883
EVCC  | [mqtt  ] DEBUG 2024/11/21 20:17:02 tcp://172.17.17.10:1883 connected
EVCC  | [main  ] INFO 2024/11/21 20:17:02 listening at :7070
EVCC  | [main  ] ERROR 2024/11/21 20:17:05 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': cannot get vehicles: Message: 404 Not Found, Locations: [], Extensions: map[code:request_error], Path: []
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:05 charge total import: 16711.442kWh
EVCC  | [site  ] INFO 2024/11/21 20:17:07 site config:
EVCC  | [site  ] INFO 2024/11/21 20:17:07   meters:      grid ✓ pv ✓ battery ✓
EVCC  | [site  ] INFO 2024/11/21 20:17:07     grid:      power ✓ energy ✗ currents ✗
EVCC  | [site  ] INFO 2024/11/21 20:17:07     pv 1:      power ✓ energy ✗ currents ✗
EVCC  | [site  ] INFO 2024/11/21 20:17:07     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
EVCC  | [site  ] INFO 2024/11/21 20:17:07   vehicles:
EVCC  | [site  ] INFO 2024/11/21 20:17:07     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
EVCC  | [lp-1  ] INFO 2024/11/21 20:17:07 loadpoint 1:
EVCC  | [lp-1  ] INFO 2024/11/21 20:17:07   mode:        minpv
EVCC  | [lp-1  ] INFO 2024/11/21 20:17:07   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
EVCC  | [lp-1  ] INFO 2024/11/21 20:17:07   meters:      charge ✓
EVCC  | [lp-1  ] INFO 2024/11/21 20:17:07     charge:    power ✓ energy ✓ currents ✓
EVCC  | [site  ] DEBUG 2024/11/21 20:17:07 set buffer soc: 90
EVCC  | [site  ] DEBUG 2024/11/21 20:17:07 set priority soc: 60
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:07 phase timer inactive
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:07 pv timer inactive
EVCC  | [site  ] DEBUG 2024/11/21 20:17:08 ----
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:09 charge power: 0W
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:11 charge currents: [0 0 0]A
EVCC  | [site  ] DEBUG 2024/11/21 20:17:11 grid power: 36W
EVCC  | [site  ] DEBUG 2024/11/21 20:17:11 pv power: 4W
EVCC  | [site  ] DEBUG 2024/11/21 20:17:11 battery soc: 26%
EVCC  | [site  ] DEBUG 2024/11/21 20:17:11 battery power: 383W
EVCC  | [site  ] DEBUG 2024/11/21 20:17:11 site power: 519W
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:12 charge total import: 16711.442kWh
EVCC  | [lp-1  ] DEBUG 2024/11/21 20:17:13 charger status: A
EVCC  | [lp-1  ] INFO 2024/11/21 20:17:13 car disconnected`

ottomitschke, 2024-11-21:
No idea why Andig closed my Bug here. Allow me to remind you that my setup is working well in docker, so not a docker issue at all but an issue with EVCC -> Polestar. 

May I please ask to reopen this Bug? - Its obvious that Polestar has changed the API. - Thanks!

andig, 2024-11-21:
Please read the issue template:

> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.

ottomitschke, 2024-11-21:
[EVCC.trace.log](https://github.com/user-attachments/files/17851327/EVCC.trace.log)

[image]

Please see the trace file attached. (Reopen of issue is still appreciated! ) 
I hope this helps. - My apologies the picture with the error message is in German, however, just to give much details as possible.

andig, 2024-11-21:
Schön, dass wir nach endlicher Diskussion doch ein Log haben:

ˋˋˋ
polestar] TRACE 2024/11/21 20:52:44 \{"query":"query getCars\{getConsumerCarsV2\{vin,internalVehicleIdentifier\}\}","operationName":"getCars"\}\
--\
Not found\
ˋˋˋ

Das könnte eine Apiänderung sein. In dem Fall bleibt nur abearten bis jemand die neue Api heraus findet.

loebse, 2024-11-22:
I think I was able to trace down the issue. Would it make sense to open this issue again @andig

loebse, 2024-11-22:
@andig: I have created the following PR to fix the issue: https://github.com/evcc-io/evcc/pull/17398

kptkip, 2024-11-22:
@andig: Issue still exists and is *NOT* closed at all.

ottomitschke, 2024-11-23:
Just want to confirm, with the actual nightly build the car function with Polestar 2 is working again. - Big thank you @loebse

flo9127, 2024-12-09:
It's broken again.

**> [main ] ERROR 2024/12/09 22:30:10 creating vehicle db:1 failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: resumePath not found**

**Trace / Log:**
`[polestar] TRACE 2024/12/09 22:30:09 GET https://polestarid.eu.polestar.com/as/authorization.oauth2?access_type=offline&client_id=l3oopkc_10&redirect_uri=https%3A%2F%2Fwww.polestar.com%2Fsign-in-callback&response_type=code&scope=openid+profile+email+customer%3Aattributes&state=jQ1NYOjBwb1T9l7h
[polestar] TRACE 2024/12/09 22:30:09 GET https://www.polestar.com/sign-in-callback?error_description=code_challenge+is+required.&state=jQ1NYOjBwb1T9l7h&error=invalid_request#.
[polestar] TRACE 2024/12/09 22:30:10 <!DOCTYPE html><html><head><meta charset="utf-8"><meta http-equiv="X-UA-Compatible" content="IE=edge"><meta name="viewport" content="width=device-width,initial-scale=1"><title>Pure progressive performance | Polestar</title><meta name="description" content="Passion and emotion drive us, electricity and innovation drive our cars. We are all in, guiding our industry forward through pure, progressive, performance."><meta name="application-name" content="Polestar"><meta name="mobile-web-app-capable" content="yes"><meta name="theme-color" content="#ffffff"><meta name="apple-mobile-web-app-capable" content="yes"><meta name="apple-mobile-web-app-status-bar-style" content="black"><meta name="apple-mobile-web-app-title" content="Polestar"><link rel="shortcut icon" href="https://www.polestar.com/w3-assets/favicon.ico"><link rel="apple-touch-icon" href="https://www.polestar.com/w3-assets/apple-touch-icon.png"><link rel="icon" type="image/png" sizes="32x32" href="https://www.polestar.com/w3-assets/favicon-32x32.png"><link rel="icon" type="image/png" sizes="16x16" href="https://www.polestar.com/w3-assets/favicon-16x16.png"><link rel="mask-icon" href="https://www.polestar.com/w3-assets/safari-pinned-tab.svg" color="#000000"><link rel="icon" type="image/png" sizes="228x228" href="https://www.polestar.com/w3-assets/coast-228x228.png"><meta name="msapplication-TileColor" content="#ffffff"><meta name="msapplication-TileImage" content="https://www.polestar.com/w3-assets/mstile-144x144.png"><meta name="msapplication-config" content="https://www.polestar.com/w3-assets/browserconfig.xml"><link rel="yandex-tableau-widget" href="https://www.polestar.com/w3-assets/yandex-browser-manifest.json"><link rel="preconnect" crossorigin="anonymous" href="https://cdnjs.cloudflare.com/"><link rel="preload" as="script" crossorigin="anonymous" href="https://cdnjs.cloudflare.com/polyfill/v3/polyfill.min.js?version=3.111.0&features=document.querySelector%2Cdefault%2Cfetch"><script defer="defer" src="/auth/static/b9a766d1.js" integrity="sha256-66yj5R6AjAlQMZ3oBpdx2cXhnkzLYVfuDYKQ99uMyXk= sha384-VTVKzhBXMsCgO5+HLEhbG2RFLMRvxksZwsQa0FMOW+V4t6fxgIAbUzqOayUZ94if" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/2da361e0.js" integrity="sha256-JN6EccKnshC2oPhsokT/9AvIV9emqbg99fwfBn0KTDg= sha384-2MIGdcOHvSqSAjD2VnRyg9qtyZtQySVT1Nhav1liHE5wWy8mjlT3YwUGopYABTaJ" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/0169f248.js" integrity="sha256-2j6kNkeXhV/PEWSyNiTk2CQkMn3v9BTWiOyV6IDc5Ag= sha384-ESFcBbyJG4a3PKqUWJw+r8cRchyaITXHbsVKWyb3hzi+v5AHrq/CTr+kmyzWV+yV" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/919f1409.js" integrity="sha256-GT6xj44RUd3EtPlO817cN92kWZQPR003Fx/037K6cBc= sha384-9dSCzllMkZWuPdsRYTAvPyibPcei5aduq9nAkigWOr5eTJgIlPxKz186HF1vNrcg" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/5609ba1d.js" integrity="sha256-RWNdC+5XcvPheQ3KcmSTEpTHuWD0b4UjE4YSlW4Ea3Y= sha384-u1kXYAmUlPnhDQnXhZUpA5/FqX7rfczWpiDkOCIWJsmQPlq1gvWoqm5ErN0BJkPj" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/78d1d6de.js" integrity="sha256-tLwjGQmj5pXG+uOUevSDfJ/oy2N7lmqI7F/dkMDKQEA= sha384-5Gxb9D+4k1fBE+fWnMADxtsI4SYsBrTj8qSD+QBAAyk9VkN/nhO8BbrhKGgnl6P1" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/135023cf.js" integrity="sha256-SOnyMd+KweocOr+ldISGd7PxTzKNs/7DTa5AO8MRBU0= sha384-pAr0662NlRnz2ClXVfVIs1lDOhGokDfjXxwF24KkzpDqjYASJQI89RBxuiROFifk" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/a58a3a80.js" integrity="sha256-Ud2a4U5718O1tdS6/3KdqZBmfMmGWbg3ZwwCUX1uwFE= sha384-h69nQ/XXifoITsLT31zz4FnXZp3x8feBp1VrgOKNiwCV9H+YCL9LbHsD7hKD0Nab" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/6a9a1c64.js" integrity="sha256-HEwJCzYD7awtxksW+DjpeFV4n+vVC9HWXn5kKykxUao= sha384-KGc3wH9BVxjQUZMG5CQ+f3KPIAN/tNqze50vfZ75zB7C3bqhnwJ8xXnuQBnyJHLq" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/6784b01c.js" integrity="sha256-JaCZzo01ARtUPcMtqXMMZu/TiiUTgxU1sTYs68ZQFW4= sha384-wwAjkmSbQZExziic00hltl+OJGSm94QTS347FbyrKvOPlJ9DHE7Oq5FkXf+wbivY" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/c2703543.js" integrity="sha256-+Y6x5l7eZAH8m9s2CGdmEhI0tRYtyC7fJ4v+xNrr/yU= sha384-AF/+3w4Crpe7+N7h1fBs5mqDU/xLwuyPInfC8CokDeGVODDobSLaknT6sfYbsBfC" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/1026da5c.js" integrity="sha256-IeU9YK8U8QPXa/ATm5cMgLFbzBASPVEIXFbiZz9pBDI= sha384-GaSjsfZnavEDyrZjc2pOG31f9JrWxSmQZxbXJra9l7JKdp1oYDzorKB3GPglFoVG" crossorigin="anonymous"></script><script defer="defer" src="/auth/static/7b2d1fc3.js" integrity="sha256-pRz4s/AObe9IwK2ku+QULLGeb0SnzrdhuhaChCUC39M= sha384-+A+QHO1HDAanri4L [... cut ...] ead><body><div id="root"></div><noscript>JavaScript seems disabled in your browser. We are deeply sorry to disturb your experience, but this application require JavaScript to work properly. Please&nbsp;:&nbsp;<a href="https://enable-javascript.com" target="_blank" rel="noreferrer">enable JavaScript</a>.</noscript><script crossorigin="anonymous" src="https://cdnjs.cloudflare.com/polyfill/v3/polyfill.min.js?version=3.111.0&features=document.querySelector%2Cdefault%2Cfetch"></script></body></html>`