evcc-io/evcc #17381: Feature Request: Manual Entry and Editing of Charging Log Entries for External Charging Sessions

rschueler, 2024-11-21:
I would like to request a feature that allows users to manually enter and modify entries in the charging log to document charging sessions conducted at public charging stations. The goal is to obtain a consolidated cost overview for my vehicle. While I predominantly charge at home using evcc, I occasionally charge externally. Including these costs in my statistics would provide a more comprehensive understanding of my total charging expenses.

**Key Features:**

Ability to manually add and edit charging sessions with the following fields:
- Date and Time of the charging session
- Charged kWh
- Cost
- Duration
- Vehicle (selectable via dropdown)
- Comments (optional)
 
**Use Case:**

This feature would be particularly useful for users who primarily charge at home but occasionally use external charging stations. By allowing manual entry of external charging sessions, users can maintain a complete and accurate charging log, facilitating better cost tracking and analysis.

**Describe alternatives you've considered**
Alternatively, providing API access to the charging sessions would also achieve this goal. This would allow users to:

- Retrieve charging data to create a consolidated view in an external system.
- Insert charging session entries via API.

andig, 2024-11-21:
Thank you for the suggestion. We don‘t plan to work on this, its beyond scope of evcc.

/cc @naltatis

naltatis, 2024-11-22:
Agree with @andig. The ability to manually create sessions would, beside the obvious data entry/management UI, open up a lot of special cases we'd have to address (optional fields for charging speed, duration, co2, ...). Sounds like a good opportunity for a third-party application. evcc charging data can be easily fetched from REST API or as CSV.

closed 2024-11-22 by naltatis