evcc-io/evcc #17401: Mein Enyaq wird seit einigen Versionen nicht mehr erkannt!

Makio74, 2024-11-22:
### Describe the bug

Der Skoda Enyaq wird nicht mehr automatisch erkannt! Wenn ich ihn von Hand auswähle kommt nicht mehr der Ladestand etc.
Evcc scheint die Angaben des Autos nicht mehr abzuholen! Mail und Pass stimmen noch, wurden nicht geändert. Agb sind alle zugestimmt!

### Steps to reproduce

Es wird zwar nach einstellen des Enyaq´s im evcc von Hand das Laden gestartet, aber nur mit +Prozentangabe. Nicht mehr die vorherige Anzeige: Ladestand und Grenze bis zu der geladen werden soll. Die ich dann auch verschieben kann!

### Configuration details

```yaml
vehicles:                                                                                                               - type: template                                                                                                         template: enyaq                                                                                                         title: Skoda Enyaq                                                                                                      user: ###                                                                                              password: ###                                                                                                    capacity: 50                                                                                                            timeout: 10s                                                                                                            name: Enyaq                                                                                                             icon: car
phases: 3
cache: 15m
mode: off
minCurrent: 4 
maxCurrent: 16 
identifiers: 
priority: 1
```

### Log details

```text
[site ] DEBUG 2024/11/22 15:32:02 ----
[lp-1 ] DEBUG 2024/11/22 15:32:02 charge power: 11040W
[site ] ERROR 2024/11/22 15:32:02 battery mode: write tcp 192.168.178.31:44248->192.168.178.85:502: write: broken pipe
[site ] DEBUG 2024/11/22 15:32:02 grid power: 7773W
[site ] DEBUG 2024/11/22 15:32:02 pv power: 0W
[site ] DEBUG 2024/11/22 15:32:02 battery soc: 0%
[site ] DEBUG 2024/11/22 15:32:02 battery power: 0W
[site ] DEBUG 2024/11/22 15:32:02 grid currents: [16.5 1.13 15.2]A
[site ] DEBUG 2024/11/22 15:32:02 site power: 7873W
[lp-1 ] DEBUG 2024/11/22 15:32:02 charger status: C
[eebus ] DEBUG 2024/11/22 15:32:03 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 718 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:05 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 719 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:07 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 720 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:09 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 721 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:11 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 722 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:13 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 723 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:15 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 724 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:17 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 725 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:19 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 726 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:21 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 727 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:23 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 728 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:25 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 729 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:27 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 730 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:29 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 731 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/11/22 15:32:31 Send: d:_i:47859_Elli-Wallbox-2209A2468W notify 732 DeviceDiagnosisHeartbeatData
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.6

roedeer82, 2024-11-22:
Gleiches Problem bei mir nach Update auf Version v0.131.6

andig, 2024-11-22:
Logfile fehlr

closed 2024-11-22 by andig

Makio74, 2024-11-23:
[evcc-20241123-124122-debug.log](https://github.com/user-attachments/files/17879518/evcc-20241123-124122-debug.log)

andig, 2024-11-23:
Lass mich raten- du hast nichts geändert und plötzlich gehts nicht mehr? Der Fehler ist offensichtlich:

> [main  ] ERROR 2024/11/23 12:18:56 creating vehicle Enyaq failed: invalid vehicle type: templateu

Makio74, 2024-11-23:
Doch ich hatte, als nichts mehr ging von der Seite noch Zeilen angefügt. Alles unter "icon: car". Dabei ist mir das mit dem u passiert habe ich gerade auch gesehen und behoben! Nach Neustart geht es nun wieder! Es war aber vorher irgend ein Fehler drin, denn ich hatte ja immer diese Meldung im Log gehabt, hatte jemand kann mir sagen wo der 'Fehler zu suchen ist: 
ERROR 2024/11/22 15:32:02 battery mode: write tcp 192.168.178.31:44248->192.168.178.85:502: write: broken pipe

roedeer82, 2024-11-23:
Problem konnte ich gerade lösen:
1. Login auf https://skodaid.vwgroup.io/account
2. Unter "Fahrzeug-Einwilligungsmanagement" "E-Privacy-Informationen (ePrivacy)" bestätigen.
3. evcc neu starten

Makio74, 2024-11-24:
Hab das mit der ePrivacy auch noch gemacht, obwohl ich auch so schon allem anderen zugestimmt hatte. Nun kommen keine Fehlermeldungen mehr.
Problem ist gelöst!