evcc-io/evcc #17415: Cannot find recently added vehicles (via cofig UI) in the db with 'evcc vehicle'

TangoTaker, 2024-11-23:
### Describe the bug

I cannot find recently via the config UI added vehicles with 'evcc vehicle' It shows me two entries, whilst I have 4 cars working in the UI. The two cars it finds are outdated. I would like to add the parameter 'vehicle' to my loadpoints and now I cant find the respective name/identifier

### Steps to reproduce

1. configure a vehicle in the config UI
2. run 'evcc vehicle'

### Configuration details

```yaml
configured the vehicles via config UI
```

### Log details

```text
no error displayed
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.6

TangoTaker, 2024-11-23:
This is what running evcc vehicle looks like:

[image]

This is how it looks in the config UI:

[image]

If there is a new way to get the vehicle names/identifiers, please let me know. Thank you.

andig, 2024-11-23:
Seems you're not using the same DB then.

Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-11-23 by andig