evcc-io/evcc #17418: Charging home battery by the loweste price interval of the day

timpetersson81, 2024-11-24:
It would be great if there was an option to choose to leave the battery at home based on the lowest price, and not that you have to define a price.

Together with the automatic management of the use of the home battery during the expensive hours of the day.

Perhaps combined with reading the typical average consumption over the day, and releasing and charging the battery in the hours when the electricity is expensive, and the consumption is high, and then charging again in the cheap hours, in order to be able to release the battery again when the electricity is expensive.

In Denmark, electricity is typically cheap at night and expensive between 7-9, and very expensive between 17-21, and it is therefore an advantage to be able to charge the battery during the cheap hours between 00-06, and then use the electricity during the expensive hours.

The current model where a price below the set value will charge the battery, needs ajustment everyday due to dynamic energy prices.

I have made. own automations for this, but only 1 instance can control the battery @andig said.

andig, 2024-11-24:
If you want to do this yourself (which is fine) disable battery control in evcc.

naltatis, 2024-11-24:
@timpetersson81 The fixed price limit is our first step into this topic. We're planning on implementing more flexible and smart ways. This discussion thread https://github.com/evcc-io/evcc/discussions/16364 (sorry, mostly German) collects possible solutions. So right now, we have not decided which route to go, but there will be a next step :)

closed 2024-11-24 by naltatis