evcc-io/evcc #17422: SessionsUI - kWh im Tooltip anzeigen?

maatinh, 2024-11-24:
Hallo,

die Sessios-UI ist wirklich super! Ich vermisse nur leider die gelandene Energiemenge zusammen mit den Kosten in einer Anzeige. Energiemenge gibt es nur bei Sonnenanteil.
Könnte man die kWh im Tooltip bei der Kostenanzeige pro Fahrzeug mit anzeigen?

naltatis, 2024-11-24:
Hi @maatinh,

was ist denn dein Anwendungsfall dafür? Bzw. für welchen Kontext möchtest du das haben? Abrechnung, mit Leuten teilen, ...

Generell kannst du ja mit dem Umschalter Sonne/Preis direkt zwischen Menge und Kosten hin- und herwechseln. Aber ja, in einer Ansicht gibt es diese beiden Werte momentan nicht. Man könnte das mit in den Tooltip aufnehmen, aber das fühlt sich etwas willkürlich an. Gerade weil man dann auch sagen könnte, dass man CO2 und Preis/kWh vielleicht auch noch direkt mit reinhaben möchten und dann wird es komisch.

Was es in der Tat nicht ins erste Release der Visualisierung geschafft hat, ist eine tabellarische Ansicht der gruppierten Daten. Hier gibts ja aktuell nur den Link auf die Einzelladevorgänge. Ich kann mir aber gut vorstellen, dass wir hier auch eine Tabelle mit allen Daten in der jeweiligen Gruppierung (Fahrzeug, Ladepunkt) unter den Charts anzeigen. Darin könnten wir dann alle Werte (Kosten, kWh-Preis, Sonnenanteil, Menge CO2, ...) pro Fahrzeug/Zeitraum gruppiert anzeigen - unabhängig von der Sonne/Preis/CO2 Auswahl oben.

maatinh, 2024-11-25:
Ja, mir fehlt tatsächlich EINE Übersicht, über die Ladestatistik. Energiemenge/Kosten. In der Kostenansicht sehe ich den Gesamtpreis und den Durchschnitt, weiß aber gar nicht wieviel ich geladen habe. Dafür muss ich dann wechseln indem ich auf die Sonne klicke. Da merke ich mir dann die Energiemenge vielleicht noch für einen anderen Tag und muss wieder zurück auf Preis wechseln.

Die Tabelle ist auf dem Telefon so bedingt hilfreich. Wenn ich im Porträtmodus auf "Ladevorgänge" klicke, sehe ich in der Tabelle nur Datum/LP/Fzg/kWh. Wenn ich es dann auf Landscape kippe ändert sich an der Tabelle nichts. Schließe ich dann die Ansicht Ladevorgänge, halte das Telefon quer und öffne die Ladevorgänge, dann hat die Tabelle auch mehr Spalten. Bis ich es einmal hochkant drehe, dann kommen die Spalten auch nicht mehr zurück, wenn ich es wieder quer halte.
Desweiteren muss ich die Tabelle dann auch erst filtern, um die Infos zu bekommen, die aus den Grafiken viel schöner abzulesen sind.

Ein Hinweis noch zum Umschalter Sonne/Preis. Klicke ich auf Preis, ist das Diagram mit "Ladekosten" beschriftet. Das macht Sinn. Klicke ich auf die Sonne, heißt das Diagramm "Geladene Energie". Wäre da nicht ein Blitz o.ä. als Symbol besser? Für die Ladepunkt/Fahrzeug Darstellung wird ja der Sonnenanteil auch gar nicht aufgeschlüsselt, sondern nur für die Summendarstellung.

naltatis, 2024-11-25:
@maatinh in deinem Kommentar waren jetzt sehr viele unterschiedliche Punkte auf einmal.

Tabelle: Ich rede hier in der Tat von einer anderen Tabelle als die der Einzelvorgänge. Die kann man auch noch mal anders optimieren um auch auf kleinem Platz viele Daten anzuzeigen. 

Das was du mit dem Landscape/Porträit-Modus und Neuladen beschreibst, klingt nach einem Bug. Magst du das mal in einem separaten Issue mit technischen Details (Browser, Gerät, ...) schildern.

Zum Thema Sonne vs. Energie gabs im originalen PR schon einiges an Diskussion. Ich würde hier erst einmal bei der jetzigen Lösung bleiben. Nehme dein Feedback aber mit. Ggf. ändern wir das noch mal.

closed 2024-11-25 by maatinh