evcc-io/evcc #17430: Messaging: vehicleSoc & vehicleRange bei 0

lmfgpeace, 2024-11-24:
### Describe the bug

**Whats the issue?**
Täglich kommt es vor, dass bei `${vehicleSoc}` & `${vehicleRange}` in Ausgabe über Pushover (aber vermutlich auch bei allen anderen Services) 0 angezeigt wird.

Dies kommt vermutlich zustande, da der aktuelle Wert nicht von der vehicle API abgerufen werden kann.

**Wann kommt es zu dem Problem?**
1. wenn das Auto neu verbunden wurde `connect` (Immer)
2. wenn das Auto getrennt wurde `disconnect‘ (immer)
3. wenn Ladevorgang gestartet `start` (sporadisch, keine Logik erkennbar)
4. wenn Ladevorgang beendet wird `stop` (sporadisch, keine Logik erkennbar)

### Steps to reproduce

siehe oben

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 5s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjkyNzQ0MDAsImlhdCI6MTcyODQ5NjgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.gnfLjIPCnFt3TmCUhoXCmA3nHK5RkQcOrLIejCQfc4w
# trial token, valid until 2024-10-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3Mjk4NjQ4MDAsImlhdCI6MTcyOTA4NzIwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.3iF50HEbpf7hF7nob075jJxseDt4c2rFeSS-ZHpxzuE

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
[... 280 lines cut ...]
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
n.e.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.7

andig, 2024-11-24:
Das kann passieren...

closed 2024-11-24 by andig

andig, 2024-11-24:
> Das stimmt, ist aber nicht wirklich ein Lösungsorientiertes Herangehen

Vielleicht hab ich was überlesen, aber ich habe hier gar kein Problem gesehen, dass einer Lösung bedürfte? Anders gesagt: was ist das Problem?

lmfgpeace, 2024-11-26:
> > Das stimmt, ist aber nicht wirklich ein Lösungsorientiertes Herangehen
> 
> 
> 
> Vielleicht hab ich was überlesen, aber ich habe hier gar kein Problem gesehen, dass einer Lösung bedürfte? Anders gesagt: was ist das Problem?

Das Problem ist, dass die Werte `vehicleSoc` & `vehicleRange` praktisch im Messaging unbrauchbar sind, da sie nie korrekt ausgegeben werden, außer beim Ereignis `soc`.

lmfgpeace, 2024-11-26:
Manchmal klappts dann aber auch bei den anderen Ereignissen neben `soc`.

Ist halt n sehr sporadischer Fehler, der aber täglich auftritt:

[image]

svn2208, 2024-12-20:
Ich möchte auch noch mal das Thema "stressen". Bei mir werden beim Ereignis: 'connect' nicht der richtige SOC angezeigt. Ich vermute, dass die Nachricht rausgeht, bevor das Auto von Guest auf "per API gescanntes" Auto springt.