evcc-io/evcc #17436: SMA Tripower SE Grid Charging not working

ulrichsc, 2024-11-25:
### Describe the bug

SMA Tripower 10.0 SE (Firmware-Version: 3.5.25.R) 
Dynamic pricing with Tibber.
When setting the experimental feature for grid-charging the battery to a price-level greater than the current price, the battery will stop charging/discharging entirely.

### Steps to reproduce

1. Set the price limit for grid charging to a price level greater the current price.
2. Wait for the next update-intervall

### Configuration details

```yaml
- name: my_battery
    type: template
    template: sma-hybrid
    usage: battery
    chargepower: 3000
    modbus: tcpip
    id: 3
    host: x.x.x.x
    port: 502 # Port
    capacity: 10
    watchdog: 30s
```

### Log details

```text
Nov 25 10:11:25 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:25 set grid charge limit: 0.4
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 ----
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 set battery mode: charge
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 grid power: 64W
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 grid powers: [10 44 10]W
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 grid currents: [1.44 1.09 0.884]A
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 pv 2 power: 64W
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 battery soc: 18%
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 battery power: 0W
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 pv 1 power: 769W
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 pv power: 833W
Nov 25 10:11:32 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:32 site power: 164W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 ----
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 grid power: -54W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 grid powers: [-29 7 -32]W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 grid currents: [-1.46 1.08 -0.905]A
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 pv 2 power: 64W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 battery soc: 18%
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 battery power: 0W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 pv 1 power: 781W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 pv power: 845W
Nov 25 10:11:53 openhabian evcc[1855710]: [site  ] DEBUG 2024/11/25 10:11:53 site power: 46W
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.7

andig, 2024-11-25:
>  I have verified that the issue is reproducible with the latest nightly build

Did you?

closed 2024-11-25 by andig

ulrichsc, 2024-11-25:
Thanks for the quick reply.
So apparently it was fixed 2h ago. Works now with latest nightly:
evcc version 0.131.7 (f4111139)

andig, 2024-11-25:
Perfect, thanks for confirming!