evcc-io/evcc #17439: Re-enable Porsche API

MarkusGH, 2024-11-25:
Follow-up for issue https://github.com/evcc-io/evcc/issues/14371

Referenced in discussions https://github.com/evcc-io/evcc/discussions/14404, https://github.com/evcc-io/evcc/discussions/12445

pyporscheconnect has a working implementation for the new api:

https://github.com/CJNE/pyporscheconnectapi/tree/new_api

The pyporscheconnect commandline interface has a debug mode that shows all requests made.

I would greatly appreciate if the Porsche API changes could be implemented so the Porsche API is available again in evcc.

closed 2024-12-07 by andig

farbfoto, 2025-05-01:
Have the same question - The Macan becoming more and more available. Can integrate it but withiut any SoC. I could pull SoC over MQTT but this doesn't seem to work either.

andig, 2025-05-01:
> is there an update on the Porsche API for evcc?

We've been in lengthy contact with Porsche but eventually didn't receive API access. That said, we won't invest any more time in this (two years is enough). 

You can still interface creating a custom vehicle.

Dervorino, 2025-05-02:
Vielen Dank!

Custom Vehicle heisst: in der evcc. yaml diesen code hinzufuegen (siehe
unten)?

Ueber das experimentelle UI Menu kann ich den Wagen
naemlich hinzufuegen (über ISO5...), aber muss ihn immer manuell auswaehlen
und die 100kWh scheint evcc auch nicht zu erkennen. Beim MiniCooper geht
das alles perfekt ueber die API.

vehicles:
  - name: porsche-macan4
    title: Porsche Macan 4
    type: custom
    capacity: 100
    phases: 3

ivoks, 2025-06-11:
FWIW; I had a meeting with Porsche Digital this week about the overall digital experience in new Porsches. I pointed out that they are being unreasonably difficult while all other relevant vendors provide access.