evcc-io/evcc #17440: Netzgeladener Batteriestrom

Ralf-Th, 2024-11-25:
**Is your feature request related to a problem? Please describe.**
Der Preis für den entnommenen Batteriestrom stimmt nicht, wenn aus dem Netz gefüllt wurde

**Describe the solution you'd like**
Der Opportunitätspreis von 8ct, den ich für die Einspeisung hinterlegt habe passt nicht bei selbst erzeugtem Nicht-PV-Strom

**Describe alternatives you've considered**
Der Preis für Batteriestrom sollte einer Mischkalkulation zugrunde legen, der die netzgeladene Energie gem. dem Bezugspreis berücksichtigt

naltatis, 2024-11-25:
Das ist, Stand heute, erwartetes Verhalten.

Siehe auch https://docs.evcc.io/docs/faq#ersparnisberechnung

> Was ist Sonnenenergie?
> 
> Die direkte PV-Nutzung und die vom Batteriesystem bereitgestellte Energie werden bei dieser Berechnung gleich behandelt. evcc geht davon aus, dass der Hausspeicher primär zur Speicherung von überschüssigem, selbstproduzierten Sonnenstrom verwendet wird. Erfüllt der Speicher auch netzdienliche Leistungen (bspw. Sonnen) oder führt eine Kalibrierung mit Netzstrom durch, ist diese Annahme nicht immer korrekt. Batterieverluste durch die Umwandlung werden nicht berücksichtigt.

Wenn das Batterienetzladen aus dem experimentellen Stadium raus ist, werden wir dieses Annahmen sicher noch einmal überarbeiten. Ist aber ein etwas größeres Thema, da wir permanent den "Wert des Speichereinhalts" berechnen müssen.

closed 2024-11-25 by naltatis

Ralf-Th, 2024-11-25:
Deswegen habe ich doch ein feature request gestellt!Diese Nachreicht wurde mobil verfasst.

naltatis, 2024-11-25:
> Deswegen habe ich doch ein feature request gestellt!

Und deswegen hab ich drunter geschrieben, dass wir das angehen, wenn das Feature nicht mehr experimentell ist 😄