evcc-io/evcc #17446: "broken pipe" zu E3DC WR mit RSCP Protokoll

Dirk77777, 2024-11-26:
### Describe the bug

Der Fehler in #17247 ist jetzt auch mit evcc als Service, d.h. ohne Docker aufgetreten.
Die Logs hänge ich hier nochmal an.

### Steps to reproduce

Kann ich nicht sagen. Gefühlt tritt es einfach sporadisch irgendwann auf.

### Configuration details

```yaml
Alles was geht wurde über das UI konfiguriert, auch der E3DC WR, d.h. RSCP Protokoll mit Port 5033.
evcc läuft auf einem Raspberry Pi als Linux Service. Pi und E3DC sind im gleichen Subnet und über LAN verbunden.
Auto, WR etc. wurde über das UI konfiguriert.

192.168.1.18 = Raspberry mit evcc Service
192.168.1.16 = E3DC
192.168.1.30 = Wallbox (Tesla Wall Connector)
```

### Log details

```text
[site  ] ERROR 2024/11/25 19:59:38 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:38 pv power: 0W
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:38 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:38 ----
[lp-1  ] DEBUG 2024/11/25 19:59:38 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 19:59:38 charge currents: [0.1 0 0.1]A
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T19:59:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:39 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:39 pv power: 0W
time="2024-11-25T19:59:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:39 grid power: write tcp 192.168.1.18:32970->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:46 ----
[lp-1  ] DEBUG 2024/11/25 19:59:46 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 19:59:46 charge currents: [0.1 0 0.1]A
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:46 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:46 pv power: 0W
time="2024-11-25T19:59:46+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:46 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:56 ----
[lp-1  ] DEBUG 2024/11/25 19:59:56 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 19:59:56 charge currents: [0.1 0 0.1]A
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T19:59:56+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:56 pv 1 power: write tcp 192.168.1.18:32982->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 19:59:56 pv power: 0W
time="2024-11-25T19:59:57+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 19:59:57 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 20:00:06 ----
[lp-1  ] DEBUG 2024/11/25 20:00:06 charge power: 0W
[lp-1  ] DEBUG 2024/11/25 20:00:06 charge currents: [0.1 0 0.1]A
[... 173 lines cut ...]
time="2024-11-25T20:02:27+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
[site  ] ERROR 2024/11/25 20:02:27 battery 1 power: write tcp 192.168.1.18:32980->192.168.1.16:5033: write: broken pipe
[site  ] DEBUG 2024/11/25 20:02:27 ----
[lp-1  ] ERROR 2024/11/25 20:02:37 charge power: Get "http://192.168.1.30/api/1/vitals": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-1  ] ERROR 2024/11/25 20:02:38 charge currents: Get "http://192.168.1.30/api/1/vitals": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-11-25T20:02:38+01:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-11-25T20:02:39+01:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.1.31.6

andig, 2024-11-26:
Der WR schliesst die Netzwerkverbindung. Bitte Ticket beim Hersteller aufmachen.

Dirk77777, 2024-11-26:
Ok, das mach ich.
Aber kann man evcc an dieser Stelle nicht etwas robuster machen - Fehler fangen, und z.B. eine neue Connection aufmachen? Nach einem Neustart von evcc über das UI (Konfiguration -> Neustart) geht es meistens wieder.

maikl-muc, 2024-11-26:
Hallo Zusammen,
bei mir kommt genau der selbe Fehler auch regelmässig (alle 3/4 Tage) & ein Neustart von evcc behebt den Fehler immer. Würde mich der Frage von @Dirk77777 anschließen, ob man nicht einfach eine neue Connection aufmachen könnte, wenn die aktuelle geschlossen wurde....

Dirk77777, 2024-11-29:
Ticket @ E3DC ist geöffnet.

andig, 2024-11-29:
@docolli sieht Du evtl. was in den Nachrichten oben falsch sein könnte? Sonst muss ich von Fehler bei e3dc ausgehen...

andig, 2024-11-29:
> Aber kann man evcc an dieser Stelle nicht etwas robuster machen - Fehler fangen, und z.B. eine neue Connection aufmachen? 

@Dirk77777 der Punkt ist gut. Willst Du damit sagen, dass nach einmaligem Fehler alle (?!) weiteren Zugriffe fehlschlagen?

/cc @spali falls Du das nicht in der Library machen willst müsste evcc nach jedem Fehler ein Disconnect machen?

maikl-muc, 2024-11-29:
@andig Ja, danach schlagen alle weiteren Zugriffe fehl & es hilft nur evcc neu zu starten. Danach funktioniert der Zugriff dann wieder problemlos.

andig, 2024-11-29:
Das ist sehr unschön- mal schauen was @spali dazu sagt.

spali, 2024-11-30:
Definitiv unschön. Denke würde Sinn machen dies im go-rscp `Client` zu lösen.
@andig  Läuft der Client in EVCC wirklich die ganze Zeit ohne mal neu zu verbinden seitens EVCC... Falls ja, bin ich ehrlich gesagt positiv überrascht, da ich den Client nie "langzeit" getestet habe, sonder der immer nur währen CLI aufrüfen gelebt hat ;)
Ich schau mir deine PR's so schnell und gut ich kann an, komme aber vermutlich nicht so schnell dazu diese zu testen für mein Primär use-case mit der cli.

andig, 2024-12-01:
@spali ja- evcc ist "long-running". We need to handle connection drops. If you don't want to do this in the library we can do in evcc. Doing this in evcc has the disadvantage that we can no longer decide if the root cause was a TCP error or something else. Inside the library you'd only need to check during network activities, but not e.g. on CRC operations.

spali, 2024-12-01:
fixed on my end and released 0.2.1.

andig, 2024-12-02:
Closed in 45169e54eb412cd9ab25bd9c3bc2ad50c67e0d18

closed 2024-12-02 by andig