evcc-io/evcc #17448: dev-Container go version mismatch

mha1, 2024-11-26:
### Describe the bug

Hi,

I set up the dev environment using the dev-container method. Builds failed with error go 1.23.3 expected, 1.23.2 found.

After manually deinstalling 1.23.2 and installing 1.23.3 evcc can be built successfully.

### Steps to reproduce

1. Setup dev environment using dev-container
2. start build

### Configuration details

```yaml
VSCode/docker on W11, cross compiling for arm (Pi4)
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

andig, 2024-11-26:
I‘m not maintaining the devcontainer, but the error is strange. Go should be able to download the required toolchain itself. Anybody wants to grab this feel free.

andig, 2024-11-26:
I guess the GOTOOLCHAIN breaks it. No idea where that comes from, sorry. Shouldn‘t be there.

tfenster, 2024-11-26:
#17451 fixes the problem for me by changing the GOTOOLCHAIN value

tfenster, 2024-11-26:
sorry, wrong fix, that only worked initially

tfenster, 2024-11-26:
Digging a bit deeper, the `Beispiel: '...'` and `Example: '...'` parts in https://github.com/evcc-io/evcc/commit/5da0a943bf57a2a6a9af2177fa4055dd806cb9ba break it for me. Removing the `:` solved that for me

PR #17451 (2024-11-26): fix(devcontainer): change GOTOOLCHAIN to auto
allow go to pickup the right toolchain

Fixes #17448