evcc-io/evcc #17453: e-208 SoC not updated when charging

bigbig6, 2024-11-26:
### Describe the bug

When the vehicle is plugged, the charge is started.
The car SoC is updated and periodically there is a comparison between the SoC and the target SoC.
But never since the charge this vehicle SoC is updated.
So the charge finished when it reach 100%.

I have no grid meter, I use a demo-meter, and the car is configured via the GUI. No cache parameters is available in the interface.

Information about the car is only updated when the car is plugged and unplugged.

### Steps to reproduce

1. plug the car (Peugeot e-208)
2.wait all the night
3.in the morning the car is 100% full

### Configuration details

```yaml
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# unique installation id
plant: xxxxxxx

sponsortoken: xxxxxxx

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/co>

log: trace
levels:
  site: trace
  lp-1: trace
  lp-2: trace
  cache: trace
  e-208: trace
  peugeot: trace
  ocpp: trace

site:
  title: Maison # display name for UI
  meters:
    grid: reseau # grid meter
  residualPower: 120 # additional household usage margin

meters:
  - name: reseau
    type: template
    template: demo-meter
    usage: grid
    power: 1243

loadpoints:
 - title: chargeur_gauche
    charger: chargeur_gauche
#    soc:
#      poll:
#        mode: charging
#       interval: 60m
#     estimate: false

  - title: chargeur_droit
    charger: chargeur_droit
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: false

chargers:
  - name: chargeur_gauche
    type: template
    template: demo-charger

  - name: chargeur_droit
    type: template
    template: elvi
#    stationid: EVB-P20234226 # Station ID of the charging point. Only required if multiple OCPP charging stations are set up to assign them correctly. A sing>
#    connector: 1
#    connecttimeout: 5m # optional
#    timeout: 2m
#    metervalues: Current.Import.L1,Current.Import.L2,Current.Import.L3,Energy.Active.Import.Register,Power.Active.Import,Voltage.L1,Voltage.L2,Voltage.L3
#    meterInterval: 4s
```

### Log details

```text
b
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.5

bigbig6, 2024-11-26:
log in attach
[evcc-20241120-224346-trace.log](https://github.com/user-attachments/files/17926806/evcc-20241120-224346-trace.log)
ment

mfuchs1984, 2024-11-27:
You can see the API calls and responses triggered by evcc during charging in the log, but the car only updates the SOC after charging. See als https://github.com/evcc-io/evcc/discussions/16547#discussioncomment-10878580

bigbig6, 2024-11-27:
thank you for your answer.

So the problem is on the peugeot API side...  really, on this point I hate this car! :(

And like my wallbox is an EVBox Elvi, without meters, I cannot have any feedback on kW charged in the car.
I will try estimate: true, but I think I did it in the past and the result was the same.

Regards,

mfuchs1984, 2024-11-27:
Maybe you can use `estimate: true` with a huge cache, so evcc will not query the API during charging at all.

https://docs.evcc.io/docs/reference/configuration/vehicles#cache

bigbig6, 2024-11-27:
Thanks, I will try so.

closed 2024-11-27 by andig