evcc-io/evcc #17455: Fahrzeug (Mini Cooper) wird nicht mehr erkannt

Effendi68, 2024-11-27:
### Describe the bug

Vorweg: habe die yaml seit einem Jahr unverändert, bisher ohne Probleme.
Aktuell wird mein Fahrzeug (Mini Cooper SE) nicht mehr erkannt. 
Bekomme eine Fehlermeldung im Log, dass die Verbindung nicht funktioniert.
evcc läuft bei mir im Docker-Container auf einem Synology NAS.

Habe evcc bereits mehrfach neu gestartet, das Fahrzeug auch ab- und angesteckt. Die Login-Daten haben sich übrigens nicht geändert, ich kann mich damit bei Mini problemlos einloggen.

Ist das Problem bekannt? Bin für jeden Tipp dankbar.
Tom

### Steps to reproduce

1. Fahrzeug wird angesteckt
2. Keine Verbindung von evcc zum Fahrzeug
3. Neustarts bringen nichts
...

### Configuration details

```yaml
vehicles:
- type: template
  template: mini
  title: Mini Cooper SE rot
  user: ***
  password: ***
  vin: ***
  capacity: 29 
  name: ev4
```

### Log details

```text
creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'mini': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.7

lightning-e, 2024-11-27:
BMW hat den Log-In verändert... Näheres und Lösung: #17445

markusjnagel, 2024-11-27:
> I have the exact same issue with a Skoda Enyaq since update to latest version: "cannot create vehicle type 'template': cannot create vehicle type 'enyaq': missing id_token"

Me too. Also with a Kia - since I'm not entirely sure if I missed a change that I should know of, I have created a discussion instead of an issue here (sorry, in German). https://github.com/evcc-io/evcc/discussions/17463

ThomasWup, 2024-11-27:
Gleiche Situation bei mir. Sowohl der Mini SE als auch der iX1 werden seit kurzen mit der Ergänzung (offline) angezeigt. Habe ebenfalls keinerlei Änderungen vorgenommen.

closed 2024-11-27 by andig