evcc-io/evcc #17461: Sungrow SH6.ORT Update führt zu falscher Auswertung des Batteriespeichers

JuMa1967, 2024-11-27:
### Describe the bug

Nach dem Update der unten aufgeführten Sungrow-Module Wird die Batterie falsch ausgewertet. -> Siehe Grafik
Der wert unter In mit 6,0kW ist korrekt. Im grünen Balken darüber wird es jedoch so dargestellt als ob 2,7kW davon aus der Batterie kommen. In Wahrheit wird die Batterie mit 2,7kW geladen und nicht entladen.

Der Bug is auch in evcc Version v0.131.5 und auch nach dem Update auf Version v.0.131.7 vorhanden

- Battery auf Version SBRBCU-S_22011.01.24
- Communication Module auf Version WINET-SV200.001.00.P026
- Energy Storage System 
       -> MDSP auf Version SAPPHIRE-H_03011.95.03
       -> LCD auf Version SAPPHIRE-H_01011.95.03

Falsche Auswertung in EVCC
[image]

Sungrow Module nach dem Update:
[image]
[image]
[image]

### Steps to reproduce

Fehler ist direkt nach dem Start von evcc zu sehen. Auch ein Neustart von evcc oder ein Reboot des PI bringen keine Verbesserung.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error

# Sponsortoken GitHub
sponsortoken: xxx
# unique installation id
plant: xxx

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.133.80  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.133.80  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.133.80  
  port: 502  
  usage: battery  
[... 820 lines cut ...]
[lp-1 ] DEBUG 2024/11/27 12:07:00 charge currents: [0 0 0]A
[site ] DEBUG 2024/11/27 12:07:00 pv power: 3498W
[site ] DEBUG 2024/11/27 12:07:00 battery soc: 47%
[site ] DEBUG 2024/11/27 12:07:00 battery power: 2854W
[site ] DEBUG 2024/11/27 12:07:00 grid power: -0W
[site ] DEBUG 2024/11/27 12:07:00 grid currents: [0.212 0.242 -0.423]A
[site ] DEBUG 2024/11/27 12:07:00 site power: 2954W
[lp-1 ] DEBUG 2024/11/27 12:07:00 charge voltages: [230 228 229]V
[lp-1 ] DEBUG 2024/11/27 12:07:00 detected connected phases: 3p
[lp-1 ] DEBUG 2024/11/27 12:07:00 charge total import: 935.800kWh
[lp-1 ] DEBUG 2024/11/27 12:07:00 charger status: B
[lp-1 ] DEBUG 2024/11/27 12:07:00 pv charge current: 0A = 0A + -4.28A (2954W @ 3p)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.131.7

twixfix, 2024-11-27:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/406

sanwaldx6x, 2024-11-27:
Since update of Sungrow I have similar issue. My Battery soc is always at 100%. But erverthing else works.

Bloodjunkie, 2024-11-28:
I have the same problem with my sh10rt. Ive already  restarted the whole system but it didnt help.

jroegner, 2024-11-28:
I have the same behavior here. The reason is, Sungrow has changed the data type of two Modbus registers.
These are 13021 (battery current) and 13022 (battery power raw). The data type was unsigned 16bit big endian and is NOW signed 16bit big endian. So there ist a mismatch with positive and negative values.
In homeassistant I solved it changing the type from unit16 to int16 as you see in my example below. I guess something similar needs to be done in the sungrow template in cvcc:

- name: Battery current
        unique_id: sg_battery_current
        device_address: 
        address: 13020 # reg 13021
        input_type: input
        data_type: int16
        precision: 1
        unit_of_measurement: A
        state_class: measurement
        device_class: Current
        scale: 0.1
        scan_interval: 10

- name: Battery power raw
        unique_id: sg_battery_power_raw
        device_address:
        address: 13021 # reg 13022
        input_type: input
        data_type: int16
        precision: 0
        unit_of_measurement: W
        device_class: power
        state_class: measurement
        scale: 1
        scan_interval: 10

andig, 2024-11-29:
> Nach dem Update der unten aufgeführten Sungrow-Module Wird die Batterie falsch ausgewertet. -> Siehe Grafik

> Der Bug is auch in evcc Version v0.131.5 und auch nach dem Update auf Version v.0.131.7 vorhanden

Was für ein Bug? Anscheinend hat ja Sungrow etwas geändert. Was? Bitte beim Hersteller ein Ticket öffnen!

naltatis, 2024-11-30:
Wenn jemand Informationen hat ob das eine bewusste Umstellung war oder ein Bug in der neuen Sungrow Firmware gerne her damit. Ich konnte bislang kein aktualisiertes Modbus Protokolldokument finden.

Bloodjunkie, 2024-11-30:
Hat jemand vielleicht eine Idee für einen Workaround bis geklärt ist ob das bewusst umgestellt wurde?

ralf-walk, 2024-11-30:
Hallo,

bei mir ist ein ähnliches Problem mit der Auswertung der Batterie allerdings etwas anders gelagert.

Ich habe den Sungrow SR8.0RT-V112, Batterie: SBR096.

Bei der Batterie wird nur der Ladestand angezeigt, aber nicht wie diese ausgewertet wird.
[image]

Es sieht so aus, als ob die ganzen 1,6kh direkt verbraucht werden, es wird aber ein Teil in die Batterie gespeist. Vgl. iSolarCloud:

[image]

Ich habe mal die Register mit der Doku https://community.symcon.de/uploads/short-url/8AQhNuy0LHYTsM8awDaWMbYiWeD.pdf verglichen. Viele Register sind um 1 verschoben was aber vielleicht mit 0 vs 1 basiert zusammenhängt (Danke für den Hinweis @VolkerK62).

Aber ein paar Register unterscheiden sich doch vom Typ s. auch Kommentar von @jroegner 

 ```
     register:
        address: 13021 # Battery power -> 13022 in docu
        type: input
        decode: uint16 # vorher int16

    register:
      address: 13022 # Battery level -> 13023 in docu
      type: input
      decode: uint16 # vorher int16
```

zachelnet, 2024-11-30:
Hallo @ralf-walk,
über welche verbindung erfolgt bei dir der zugriff auf Modbus, über den wifi dongel habe ich ja das selbe Problem. Aktuell nehme ich gerade LAN Schnittstelle, darüber passen die werte. (Bitte nicht wündern wenn die werte von der icloud abweichen, da hier anschein ein mittel wert genommen wird. Und beim Lan Port der Smartmeter aktuell abgefragt wird.)

Bloodjunkie, 2024-11-30:
Bei mir kommen die falschen Werte aktuell über LAN. Ich schaue mal morgen, was bei mir über den wifi Dongle zurückkommt.

ralf-walk, 2024-11-30:
Ich habe gar keinen Wifi Dongle. Zumindest bisher nicht gesehen :-). Dachte der kommt ohne Dongle ins Wifi.
Kann ich die Lan Schnittstelle direkt and den RaspberryPI anschließen?

zachelnet, 2024-11-30:
@ralf-walk

Sollte gehen. (Leider kann ich dir gerade nichts sagen ob eine Default IP-Adresse auf den LAN-Port vom Wechselrichter vergeben ist. Bei mir bezieht er diese vom DHCP-Server,) evtl muss du ein DHCP-Server auf den Lan-Interface vom RaspberryPI einrichte.

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/doc/images/Inverter_LAN_ports.drawio.svg

lmfgpeace, 2024-11-30:
Wenn es wirklich an nem neuen FW Update liegen sollte, kann das verursachende Update ziemlich genau eingrenzen (falls es hilft). Checke jede Woche nach Updates und am 25.11. habe ich folgendes Update vom WR SH10.0 RT installiert (alle anderen FW-Stände waren aktuell):

MDSP
Neue Version:SAPPHIRE-H_03011.95.03
Originalversion:SAPPHIRE-H_03011.95.01

LCD
Neue Version:SAPPHIRE-H_01011.95.03
Originalversion:SAPPHIRE-H_01011.95.01

Allerdings ist mir der Fehler erst heute (30.11.) aufgefallen, nachdem evcc auf Version v0.131.8 geupdatet wurde.

FW-Update vom 25.11.:
[image]

Aktueller Softwarestand (Stand: 30.11.2024):
[image]

lmfgpeace, 2024-12-01:
@andig 

Wurde der Fehler nicht von @jroegner und @ralf-walk bereits identifiziert und es müsste lediglich in der evcc Environment geändert werden?

andig, 2024-12-01:
Mach gerne einen PR. Ich kenne keine neue Doku von Sungrow die das bestätigen würde. Ich kenne auch keine Aussage, dass alle Geräte diese Änderung hätten. Wie sollen wir das also differenzieren? Wir wissen nichtmal, ob das Absicht von Sungrow ist oder ein Fehler den sie beheben werden.

Insofern: bitte Sungrow für Klarheit sorgen!

naltatis, 2024-12-01:
@andig +1

Zudem gibt es hier aktuell widersprüchliche Aussagen ob dieses umgestellte Verhalten an allen Ports oder nur über den Wifi Dongle auftritt.

twixfix, 2024-12-01:
Habe einen SH10RT-20 mit neuester Firmware und genau das Problem wie der Ersteller am hinteren LAN Port. Wenn ich den vorderen nutze werden dauerhaft 100%SOC beim Speicher angezeigt. Also irgendwie sind beide Ports gerade nicht zu gebrauchen. Was müsste man denn machen um den Speicher beim laden von jetzt fälschlicherweise "IN" wieder zu "OUT" zu bekommen? Scale vom register 13021 auf -1? Nur übergangsweise, denn so ist das ganze nicht zu gebrauchen.

andig, 2024-12-01:
See https://github.com/evcc-io/evcc/discussions/17406

sanwaldx6x, 2024-12-01:
> Habe einen SH10RT-20 mit neuester Firmware und genau das Problem wie der Ersteller am hinteren LAN Port. Wenn ich den vorderen nutze werden dauerhaft 100%SOC beim Speicher angezeigt. Also irgendwie sind beide Ports gerade nicht zu gebrauchen. Was müsste man denn machen um den Speicher beim laden von jetzt fälschlicherweise "IN" wieder zu "OUT" zu bekommen? So ist das ganze nicht zu gebrauchen.

Exakt das kann ich auch bestätigen. Hatte den vorderen LAN Port benutzt. Der Battery SOC war immer auf 100%.
Nach einem Neustart vom EVCC funktioniert aus unbekannten Gründen, die Modbusverbindung darüber jetzt nicht mehr.
Habe das ganze jetzt umkonfiguriert auf den hinteren Port. Jetzt passt der Batterie Soc wieder. Aber jetzt stimmt der Grif wert nicht mehr.

andig, 2024-12-01:
Gibts denn schon Tickets bei Sungrow? Das ist ja grauenhaft…

lmfgpeace, 2024-12-01:
> Gibts denn schon Tickets bei Sungrow? Das ist ja grauenhaft…

@andig 
Wenn mir jemand den Inhalt des Tickets formuliert, kann ichs über meinen Installateur-Acc erstellen.

Mir fehlt leider aber die fachliche Tiefe um hier das Problem technisch einwandfrei zu beschreiben - und das auch noch in 500 Zeichen max.

andig, 2024-12-01:
Ich denke im Kern:
- Register passen nicht zur Doku
- Es gibt keine passende Doku für die Unterschiede der Schnittstellen vorne/hinten

Beides müsste behoben werden.

lmfgpeace, 2024-12-02:
> Ich denke im Kern:
> 
> - Register passen nicht zur Doku
> 
> - Es gibt keine passende Doku für die Unterschiede der Schnittstellen vorne/hinten
> 
> 
> 
> Beides müsste behoben werden. 

Erledigt. Sowohl als Mail als auch als Ticket.

naltatis, 2024-12-02:
@ralf-walk Token ist aktualisiert.

ralf-walk, 2024-12-03:
Ich habe es jetzt zumindest für mich passend hinbekommen.
-> Anschluss am hinteren LAN Port (sonst wird keine Batterie angezeigt)
-> Änderung von int auf uint für 13021
-> Die Berechnung des Vorzeichens muss trotzdem drin bleiben

```
meters:
  - type: custom
    name: battery3
    power:
      source: calc
      mul:
        - source: modbus
          uri: ****:502
          id: 1
          register:
            address: 13021 # Battery power
            type: input
            decode: uint16
        - source: calc
          add:
            - source: modbus
              uri: ****:502
              id: 1
              register:
                type: input
                address: 13000 # Battery running state
                decode: bool16
                bitmask: 2 # Charging
              scale: -1
            - source: modbus
              uri: ****:502
              id: 1
              register:
                type: input
                address: 13000 # Battery running state
                decode: bool16
                bitmask: 4 # Discharging
```

Batterie wird sowohl beim laden als auch entladen richtig angezeigt.

zachelnet, 2024-12-05:
Hallo @ralf-walk 
hab mal deine Config getestet, dort liefert evcc beim charging folgende werte zurück "battery power: -65277W",
die leider nicht mit den werte von isolarcloud übereinstimmt.

Folgende Konfiguration funktioniert bei mir:
```
- type: custom
  name: battery3
  power:
    source: modbus
    id: 1
    uri: 192.168.100.100:502
    register:
            address: 13021 # Battery power
            type: input
            decode: int16
```

Dies funktioniert am hinteren LAN-Port, da entsprechende Vorzeichen mit übertragen werden.
Über den WLAN-Dongle werden nur positive Werte übertragen, d.h. es wird nicht zwischen Charging und Discharging unterschieden.

ralf-walk, 2024-12-05:
Hallo @zachelnet, danke für die Info. Es sieht so aus, als ob 13021 bei mir auch am hinteren Lan Port (hinter dem Dongle) als uint16 ausgegeben wird. Aktuell sehe ich im Log folgende Werte mit meiner Konfiguration:

```
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:16 charge power: 0W
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:16 charge currents: [0 0 0]A
evcc             | [site  ] DEBUG 2024/12/05 13:27:16 pv power: 935W
evcc             | [site  ] DEBUG 2024/12/05 13:27:16 battery soc: 98%
evcc             | [site  ] DEBUG 2024/12/05 13:27:16 battery power: -663W
evcc             | [site  ] DEBUG 2024/12/05 13:27:17 grid power: 34W
evcc             | [site  ] DEBUG 2024/12/05 13:27:17 grid currents: [-0.197 -0.196 0.542]A
evcc             | [site  ] DEBUG 2024/12/05 13:27:17 site power: -629W
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:17 charge total import: 2564.422kWh
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:17 charger status: A
evcc             | [site  ] DEBUG 2024/12/05 13:27:46 ----
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:46 charge power: 0W
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:46 charge currents: [0 0 0]A
evcc             | [site  ] DEBUG 2024/12/05 13:27:46 pv power: 831W
evcc             | [site  ] DEBUG 2024/12/05 13:27:46 battery soc: 98%
evcc             | [site  ] DEBUG 2024/12/05 13:27:46 battery power: -559W
evcc             | [site  ] DEBUG 2024/12/05 13:27:47 grid power: 37W
evcc             | [site  ] DEBUG 2024/12/05 13:27:47 grid currents: [-0.198 -0.199 0.53]A
evcc             | [site  ] DEBUG 2024/12/05 13:27:47 site power: -522W
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:47 charge total import: 2564.422kWh
evcc             | [lp-1  ] DEBUG 2024/12/05 13:27:47 charger status: A
```

Entspricht auch dem was ich bei iSolarClous sehe.
Ich hoffe dass Sungrow mit der Anfrage von @sanwaldx6x für Klarheit sorgt und dann die Templates ensprechend angepasst werden.

zachelnet, 2024-12-06:
Hallo zusammen,

habe den Fehler gefunden, nach dem Update werden auf Register 13021 (am hinteren Port) positive (Discharging) und negative (Charging) Werte ausgegeben. und dadurch entstehen nach der Multiplikation die falschen Werte.

Außerdem wird bei der aktuellen Firmware des WiNet-S Dongle der Status über Register 13000 nur am hinteren Port korrekt ausgegeben. Hier müssen wir wohl auf eine neue Firmware von Sungrow warten.

hab mal ein PR erstellt, dass den ersten Fehler behebt:
https://github.com/evcc-io/evcc/pull/17618/

andig, 2024-12-06:
Wir _können_ das nicht "beheben" da wir leider nicht wissen, welche WR davon betroffen sind. Es fehlt weiter eine Aussage und Doku von Sungrow. Bitte den Support kontaktieren.

lmfgpeace, 2024-12-06:
> Wir _können_ das nicht "beheben" da wir leider nicht wissen, welche WR davon betroffen sind. Es fehlt weiter eine Aussage und Doku von Sungrow. Bitte den Support kontaktieren.

Ich habe eine konkrete Antwort erhalten:

_Wir entschuldigen uns für die lange Wartezeit auf unsere Antwort. Im Anhang senden wir Ihnen das neueste Modbus-Protokoll. Gleichzeitig möchten wir darauf hinweisen, wie wir in den Schulungen und im Protokoll selbst informieren: Communication address = protocol address – 1. If data of address 5000 is to be inquired, the corresponding sending address data is 4999 (0x1387). Weitere Informationen finden Sie im Anhang._

_Bitte lassen Sie uns wissen, wie wir Ihnen noch helfen können. Wir beantworten gerne alle entstandenen Fragen. Sobald es keine weiteren Probleme mit der Installation gibt, werden wir dieses Ticket schließen.
Wir bitten um ein kurzes Feedback._

Neuste Doku:
[](https://www.icloud.com/iclouddrive/070nof6Hy-GI4kitxZ5hEDSRw#TI_20240924_Communication_Protocol_of_Residential_Hybrid_Inverter-V1.1.5_for_SHxxRT_and_SHxT_actuell)

[https://www.icloud.com/iclouddrive/0f6S3RiBYKFLsFX4ETzeEOVPQ]()

[image]

[image]
[image]

ralf-walk, 2024-12-06:
Hallo @lmfgpeace, vielen Dank für das Einholen der Informationen!

D.h. laut aktueller Doku sind beide Register auf uint.

[image]

Das würde aber wahrscheinlich zu Problemen führen für Nutzer älterer Versionen.

Dann doch PR von @zachelnet ? Weil das ABS sollte für beide Versionen passen (habe ich aber noch nicht ausprobiert).

andig, 2024-12-06:
Die 1.1.5er Doku ist gut. Wissen wir, ob diese FW kurzfristig für alle WRs ausgerollt wird oder ist?

RaphaelBarabas, 2024-12-06:
Habe vor wenigen Tagen meine Solateure (Klarsolar) um ein Firmware-Update meines SH10RT-112 (steht in DE) gebeten. Grund war die irgendwo aufgeschnappte Aussage, daß neuere Firmware besser via Modbus am Winet-S abgefragt werden kann. Was angeblich stabiler sein soll.

Es wurde vor 3 Tagen die P11 aka 95.03 aufgespielt, welche eben diesen breaking Change auf Register 13021 hat. → Zonk :)

Die Firmware ist vom November '24, die 1.1.5er Doku vom Link oben ist vom September '24. Um ehrlich zu sein traue ich dieser Sache noch nicht ganz. Mir ist immer noch nicht klar, ob das Absicht oder ein Versehen vom Hersteller ist.

In der iSolarCloud Software gibt es seit ein paar Monaten eine Konfiguration, mit der man automatische Firmware Updates erlauben kann. Hab ich vor 2 Wochen mal aktiviert, aber es kam kein Update. Ich vermute es gibt da (noch) keine Automatik.

So wie ich die Situation also verstehe, bekommen die wenigsten hybriden WR von Sungrow jemals ein Update. Nur bei Bedarf oder auf Nachfrage.

andig, 2024-12-06:
Gibts noch irgendwelche Bedenken gegen den PR?

closed 2024-12-06 by andig