evcc-io/evcc #17465: Energy flow displayed incorrectly with house battery

jeriox, 2024-11-27:
### Describe the bug

We recently installed a hybrid inverter and a house battery. I included it into evcc and everything works fine. However, there is a quirk for our setup: we also have an old PV/inverter that does not support any reasonable method of communication, so it is not included in evcc. This leads to the following situation, where energy seems to appear out of nowhere (the old PV).

[image]

I was pleased to see that evcc renders the "?" to account for that mysterious energy. However, the proportions are off. evcc knows that 1.3kW are consumed, so I would expect the bar to account for that value, with around half of it (513W) marked in green as evcc knows the source of the energy. The other part labeled with the question mark should then be gray or a similar color. Also note how the small part of grid usage is displayed with a black color to the far right, but the icon (for whatever reason shrunk to a dot) is displayed between the sun and the question mark. I would also expect this to be aligned.

### Steps to reproduce

1. Set up a hybrid inverter with PV and battery within evcc
2. Set up a grid meter within evcc
3. Connect another energy generator that is not included within evcc
4. Have the generator that is not included in evcc feed energy into the battery

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 1234

interval: 10s # control cycle interval

sponsortoken: 1234

meters:
  - type: template
    template: sma-energy-meter
    usage: grid
    name: grid1
    host: 1.2.3.5
  - name: pv_neu
    type: template
    template: sma-hybrid
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 1.2.3.4
    port: 502
  - name: battery
    type: template
    template: sma-hybrid
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 1.2.3.4
    port: 502

site:
  title: title
  meters:
    grid: grid1
    pv:
      - pv_neu
    battery: battery

chargers:
  - name: wallbox1
    type: template
    template: heidelberg
    id: 1
    device: /dev/ttyAMA0
    baudrate: 19200
    comset: 8E1
    modbus: rs485serial

loadpoints:
  - title: Garage
    charger: wallbox1
    vehicle: Kona
    disable:
      delay: 10m

tariffs:
  grid:
    type: tibber
    token: token
  feedin:
    type: fixed
    price: 0.39

vehicles:
  - name: Kona
    type: template
    template: hyundai
    capacity: 65.4
```

### Log details

```text
Nov 27 13:00:06 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:06 ----
Nov 27 13:00:06 power evcc[26117]: [lp-1  ] DEBUG 2024/11/27 13:00:06 charge power: 0W
Nov 27 13:00:07 power evcc[26117]: [lp-1  ] DEBUG 2024/11/27 13:00:07 charge currents: [0 0 0]A
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 grid power: 10W
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 grid powers: [-100 -322 431]W
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 grid currents: [-0.862 -1.57 2.04]A
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 pv power: 972W
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 battery soc: 54%
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 battery power: -1744W
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 battery has priority at soc 54% (< 80%)
Nov 27 13:00:07 power evcc[26117]: [site  ] DEBUG 2024/11/27 13:00:07 site power: 110W
Nov 27 13:00:07 power evcc[26117]: [lp-1  ] DEBUG 2024/11/27 13:00:07 charge voltages: [236 239 237]V
Nov 27 13:00:07 power evcc[26117]: [lp-1  ] DEBUG 2024/11/27 13:00:07 detected connected phases: 3p
Nov 27 13:00:07 power evcc[26117]: [lp-1  ] DEBUG 2024/11/27 13:00:07 charge total import: 5836.943kWh
Nov 27 13:00:07 power evcc[26117]: [lp-1  ] DEBUG 2024/11/27 13:00:07 charger status: A
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.7

premultiply, 2024-11-28:
Just add an additional meter on your second inverter to sample its production correctly.

jeriox, 2024-11-28:
> Just add an additional meter on your second inverter to sample its production correctly.

yeah thats what I'm planning to do long term, would still be happy to have it displayed correctly until then :)

andig, 2024-11-29:
@naltatis why is this labeled as enhancement? Afaikt we're still trying to find an actual issue here?

naltatis, 2024-11-29:
Gute Frage. Möglicherweise das falsche issue erwischt.

jeriox, 2024-12-03:
Hier die state json:

```
{"result":{"auth":{},"battery":[{"power":-776,"energy":420.671,"soc":11,"controllable":true}],"batteryCapacity":1,"batteryDischargeControl":true,"batteryEnergy":420.671,"batteryGridChargeActive":false,"batteryGridChargeLimit":0.25,"batteryMode":"unknown","batteryPower":-776,"batterySoc":11,"bufferSoc":0,"bufferStartSoc":0,"currency":"EUR","fatal":null,"greenShareHome":1,"greenShareLoadpoints":0,"gridConfigured":true,"gridCurrents":[-0.76,-0.965,1.458],"gridEnergy":1633.481,"gridPower":4.8,"gridPowers":[-71.10000000000001,-162.20000000000002,238.10000000000002],"hems":{},"homePower":0,"influx":{"url":"","database":"","org":"","user":"","insecure":false},"interval":10,"loadpoints":[{"batteryBoost":false,"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeTotalImport":5891.2,"chargeVoltages":[234,232,233],"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"chargerPhases1p3p":false,"chargerPhysicalPhases":null,"chargerStatusReason":"unknown","charging":false,"connected":false,"connectedDuration":2147483647,"disableDelay":600,"disableThreshold":0,"effectiveLimitSoc":85,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":null,"effectivePriority":0,"enableDelay":60,"enableThreshold":0,"enabled":false,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":3,"phasesEnabled":3,"planEnergy":0,"planOverrun":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"smartCostActive":false,"smartCostLimit":0.2,"smartCostNextStart":null,"title":"Garage","vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleLimitSoc":0,"vehicleName":"Kona","vehicleOdometer":20356.3,"vehicleRange":0,"vehicleSoc":0,"vehicleWelcomeActive":false}],"mqtt":{"broker":"","topic":"evcc"},"network":{"schema":"http","host":"evcc.local","port":7070},"prioritySoc":80,"pv":[{"power":698,"energy":659.201}],"pvEnergy":659.201,"pvPower":698,"residualPower":0,"siteTitle":"title","smartCostType":"priceforecast","sponsor":{"name":"jeriox","expiresAt":"2027-06-13T23:00:00Z"},"statistics":{"30d":{"avgCo2":0,"avgPrice":0.2747106698775433,"chargedKWh":401.3060000000014,"solarPercentage":9.2389082298878},"365d":{"avgCo2":0,"avgPrice":0.3135313556530989,"chargedKWh":1395.9513562341388,"solarPercentage":53.7460476627952},"thisYear":{"avgCo2":0,"avgPrice":0.3135313556530989,"chargedKWh":1395.9513562341388,"solarPercentage":53.7460476627952},"total":{"avgCo2":0,"avgPrice":0.3135313556530989,"chargedKWh":1395.9513562341388,"solarPercentage":53.7460476627952}},"tariffFeedIn":0.14,"tariffGrid":0.33,"tariffPriceHome":0.14,"tariffPriceLoadpoints":0.33,"vehicles":{"Kona":{"title":"Kona","icon":"car","capacity":65.4,"limitSoc":85}},"version":"0.131.8"}}
```

closed 2024-12-05 by naltatis

jeriox, 2024-12-05:
danke für den Fix! ich berichte dann nochmal ob alles passt sobald das released ist :)

jeriox, 2024-12-18:
[image]
sieht gut aus, danke!