evcc-io/evcc #17466: Hausbatterie aus Netz laden nur ab bestimmtem Delta zwischen Tiefst- und Höchstpreis

OliS1975, 2024-11-27:
Es ist ja jetzt schon experimentell möglich, die Hausbatterie abhängig vom Strompreis zu laden, ähnlich der preisabhängigen Ladung des Fahrzeugs.
Ich fände es super, wenn man definieren könnte, dass die Hausbatterie nicht generell bei Unterschreiten einer bestimmten Preisgrenze geladen wird, sondern nur, wenn feststeht, dass in den nächsten Stunden das Delta zwischen niedrigem Preis und hohem Preis derart hoch ist, dass das Laden des Hausakkus überhaupt wirtschaftlich sinnvoll ist. Stichwort Ladeverluste der Hausbatterie.

dvbhannes, 2024-11-27:
Das wäre ein Traum....

KDU300, 2024-11-27:
In diesem Zug sollte eine Funktion gleich mit eingebaut werden, die die Batterie nur entlädt, wenn der Strom aus dem Netz einen bestimmten Preis überschreitet. So wäre es möglich, besonders günstig zu laden und diesen Strom nur dann zu verbrauchen, wenn er besonders teuer ist. So wie ich das sehe, wird jetzt sofort entladen, wenn der Ladevorgang  beendet ist. Das ist meistens unwirtschaftlich.

naltatis, 2024-11-27:
Für diese Themen haben wir einen Ideen-Thread aufgenmacht. https://github.com/evcc-io/evcc/discussions/16364

Duplicate of https://github.com/evcc-io/evcc/issues/17424

closed 2024-11-27 by naltatis