evcc-io/evcc #17479: Add Tariff-template to dynamic price api for  NORDPOOL NORDIC Regions

lka-hh, 2024-11-29:
Hej aus dem Norden,
ich nutze evcc in Schweden seit ein paar Monaten und würde nun gern die funktion Dynamischer Strompreis hinzufügen, da ich einen entsprechenden Tarif mit meinem Versorger habe. 

Bin weder Tibber- noch Awattar-kunde , nutze bisher die Daten von Nordpool über die Webseite um die Preise manuell zu checken und die Ladezeiten zu konfigurieren. Screenshot unten angefügt.
https://data.nordpoolgroup.com/auction/day-ahead/prices?deliveryDate=2024-10-29&currency=EUR&aggregation=DeliveryPeriod&deliveryAreas=SE4,SE4

Habe versucht das Template für Nordpool elering zu nutzen und stattdessen die Region SE4 anzugeben. Das liefert aber keine Daten und ich bekomme eine Fehlermeldung : smartCostActive: no matching rate

Vielleicht wäre eine Möglichkeit das Elering-template so anzupassen, das es auch andere Regionen von Nordpool liefern kann? 
Oder davon ausgehend ein eigenes Nordpool Nordic-Template zu erstellen um die verschiedene Strompreis-Regionen in Skandinavien zu erfassen?
 Schweden hat allein 4 verschiedene Preiszonen aktuell. 

Dieses Feature würde die Möglichkeiten für evcc in Skandinavien enorm erweitern denke ich. Börsenpreisbezogene variable Stromtarife sind hier seit geraumer Zeit eingeführt und werden von vielen Kunden angenommen, bzw sie sind sogar verpflichtet diese zu nutzen, wenn beispielsweise Solarüberschuss ins Netz eingespeist werden soll, das E-Auto günstig geladen werden soll ...

Insofern wäre solch ein Template sicher auch ein Booster für evcc in ganz Skandinavien und nicht nur für meine ganz persönliche Situation ein Zugewinn.
Ich hoffe, da lässt sich etwas zaubern. Der Anfang scheint gemacht.

Vielen Dank für die Entwicklung bisher und im Voraus für alles was ihr noch hinzufügen werdet.
Ich hoffe, dass ich mit meinen rudimentären Kenntnissen trotzdem etwas beitragen kann.

Wintergrüße aus Schweden,

Lasse K

[image]
[image]

HolgerMiara, 2024-11-29:
An der Nordpool Börse wird der Strom aller Nordischen und Baltischen Länder gehandelt (siehe [Karte](https://data.nordpoolgroup.com/map?deliveryDate=latest&currency=EUR&market=DayAhead&mapDataType=Intraday&resolution=60)). Ab ca. 14 Uhr wird der stündliche Preis des Folgetages veröffentlicht (Day-ahead market). In absehbarer Zukunft sollen mehrere der Märkte auf viertelstündliche Abrechnungszeiträume und -preise umgestellt werden. 

Zu den gehandelten Regionen gehören unter anderem _Dänemark West_ und _Dänemark Ost_ (derzeit abrufbar mit dem Template `Energinet` und den Regionen `DK1` und `DK2`, sowie Finnland und die Baltsichen Länder (Estland, Lettland und Littauen, abrufbar mit dem Template `Elering` und den Regionen `FI`, `EE`, `LV` und `LT`). 

Der Name des Templates "Elering" für diese Regionen ist übrigens nicht ganz richtig. Elering ist lediglich der Übertragungsnetzbetreiber in Estland, auf dess Webseite man die Nordpool Preise (für Estland) abrufen kann. 

Falls sich also jemand erbarmt, wäre es bestimmt sinnvoll, die beiden Templates zu vereinen und mindestens die Regionen `DK1`, `DK2`, `FI`, `EE`, `LV` und `LT` verfügbar zu machen; außerdem die Regionen `SE1` bis `SE4` sowie `NO1` bis `NO5`. Sollte alles über das gleiche API gehen, habe leider keine Ahnung, wie viel Arbeit das ist. 

Ich helfe gerne beim Update der Doku, doch beim Template muss ich leider passen.

andig, 2024-11-29:
> Falls sich also jemand erbarmt, wäre es bestimmt sinnvoll, die beiden Templates zu vereinen 

Wie denn? Die Frage ist doch ob die APIs die jeweiligen Daten anbieten. Kann ich nicht beurteilen...

HolgerMiara, 2024-11-29:
Das tun sie.

andig, 2024-11-29:
Na dann müssen doch einfach nur die Wertehilfen in den entsprechenden Dateien ergänzt werden? Oder sollten nicht beide zugunsten von https://data-api.nordpoolgroup.com/index.html#/PriceCurves/get_api_v2_Auction_N2EX_DayAhead_PriceCurves_UK raus fliegen?

andig, 2024-11-29:
> Na dann müssen doch einfach nur die Wertehilfen in den entsprechenden Dateien ergänzt werden

Also https://github.com/evcc-io/evcc/blob/master/templates/definition/tariff/elering.yaml und https://github.com/evcc-io/evcc/blob/master/templates/definition/tariff/energinet.yaml. Einfach editieren und fertig ;)

Du schaffst das!

lka-hh, 2024-11-29:
Bin mir da nicht so sicher... , wo finde ich denn diese dateien zum editieren im filesystem ? ich denke ich kann die ja nicht hier direkt editieren ?

andig, 2024-11-29:
Doch, kannst Du...

Millox, 2024-12-05:
I did implement this based on a go program I wrote a couple of months ago when I realized that they changed the api so that the nord-red implementation no longer worked.

You can unpack the attached file into a clone of the repository and compile it and it should most likely work. The code is basically identical to the energinet/elering versions but supports more areas and local currency with the official exchange rate from Nordpool.

As I am not very fluent in git it will probably take quite some time to make a proper pull request.

[nordpool.tgz](https://github.com/user-attachments/files/18018045/nordpool.tgz)

closed 2024-12-07 by andig