evcc-io/evcc #17488: Authentication for changing values

throny, 2024-11-29:
**Is your feature request related to a problem? Please describe.**
It's great to have the EVCC Dashboard available to everyone. However, it's possible for everyone to change the charging profiles and limits too.

**Describe the solution you'd like**
It'd be great to lock these features. Changing these values should only be possible when authenticated with the password provided.

**Describe alternatives you've considered**
Restrict access to the entire EVCC dashboard (with something like HTTP Basic Auth)

naltatis, 2024-11-29:
We already have the account system (admin password) in place. We could provide a user setting to require authentication for all endpoints. Right now, it's only required for accessing logs or changing configuration.

Can you be more specific on what you would like protected or not? Read-only mode? Should it be allowed for everyone to see charging situation and sessions/stats?

throny, 2024-11-29:
Thanks for the quick response. Yes, it'd be great to lock more functions behind the admin password.

Basically, a read-only mode would be great. 
I'd love to see the configuration of  chargings protected. Like changing charging limits, the profiles (off/solar/fast), changing the wallbox settings (phases, current).

naltatis, 2024-11-29:
Ok, thanks for your clarification. Implementing this should not be too complicated. However, we've a lot of other topics right now.

For the meantime, the way to go would be to add basic auth via a reverse proxy.

andig, 2025-01-07:
Its not clear to me how evcc should do this since read-only mode could not be left. Imho wontfix.

naltatis, 2025-01-08:
My solution would be quite simple. We'd add auth requirement to all http endpoints that mutate state. When the user triggers a UI action (mode change, limit change, ...) it receives a "401 Unauthorized" HTTP response and will then prompt the users with the Login-Dialog. The described 401 > Login Mechanism already exists for all config-Endpoints.

What needs to be done in this issue is make this behavior configurable (Config UI) and ensure and test that we've address all the right endpoints.

cschwarz007, 2025-05-11:
To add my 2 cents here: I have the use case of making evcc available via internet (reverse proxy), hence would not want anyone to see anything without prior authentication. Easy fix would be to require auth (same as currently implemented for config) for all endpoints. This would then be the same password for the entire system. This would save a full blown user management...

On a related note: I saw the iOS app allows for setting "login required" and once ticked allows to enter username and password. Anything planned here...?

cryptedbv, 2025-06-07:
> To add my 2 cents here: I have the use case of making evcc available via internet (reverse proxy), hence would not want anyone to see anything without prior authentication. Easy fix would be to [...]

Exactly, i saw this in the iOS app as well. I just launched a Vultr VM with this as I need it online 24/7 without interuptions, however it now is accessible for everyone.

jellevanerven, 2025-12-28:
is there any news on this? remote access is a must imo but don't want strangers to see my charging sessions or let alone change limits and so on.