evcc-io/evcc #17489: Elli Pro Gen1: Laden 1-phasig fehlerhaft

CarG70, 2024-11-29:
### Describe the bug

#17157 beschreibt den selben Fehler wie bei mir. Issue wurde jedoch seinerzeit mangels Log geschlossen und wurde auch vom Ersteller nicht nachgeliefert. Mangels Sonne nun etwas verspätet von mir.
Ich nutze zum 1-phasigen laden ein entsprechendes Verlängerungskabel und meine Elli hat keinen separaten Zähler. Das Problem trat bei mir nach dem Wechsel auf Version 0.131.0 auf.
Ich verbinde meinen ID.4 mit den 1-phasigen Kabel und gebe den Ladevorgang mittels RFID-Card frei. Standard ist der PV-Modus. Fahrzeug lädt mit nur 1.3 kW obwohl mehr zur Verfügung steht. Im UI wird angezeigt "Ladebereit. Warte auf Fahrzeug", obwohl mit 1,3 kW geladen wird (siehe Screenshot). Anschließend habe ich auf min+PV gestellt, was auch zu keiner Änderung führte. Dann auf Schnell, wodurch das Fahrzeug mit 3,6 kW lädt. Zurück auf PV und alles läuft wie gewünscht. Der der Nutzer mit dem Issue #17157 einen Lasttrennschalter und einen separaten Zähler für die Elli nutzt, sollte es nicht damit zusammen hängen. 
Nightly habe ich nicht probiert, konnte das Issue aber ohne den Klick nicht veröffentlichen... ;-)
 
[image]

[evcc-20241126-121519-debug.log](https://github.com/user-attachments/files/17959000/evcc-20241126-121519-debug.log)

### Steps to reproduce

1. Ladevorgang mit 1-phasigem Kabel bei ausreichend Überschuss starten (es wird nur mit 1,4 kW geladen)
2. Auf schnell und anschl. wieder auf PV. Dann wird auch wieder an Überschuss angepasst
3. 
...

### Configuration details

```yaml
[main  ] INFO 2024/11/29 12:43:00 evcc 0.131.7
[main  ] INFO 2024/11/29 12:43:00 using config file: /etc/evcc.yaml
[db    ] INFO 2024/11/29 12:43:00 using sqlite database: /home/pi/.evcc/evcc.db
[eebus ] INFO 2024/11/29 12:43:01 Local SKI: 71385000451683c18171182ce79a5f8483bf7b80
[eebus ] DEBUG 2024/11/29 12:43:01 starting websocket server on :4712
[eebus ] ERROR 2024/11/29 12:43:01 websocket server error: listen tcp :4712: bind: address already in use
[eebus ] DEBUG 2024/11/29 12:43:01 mdns: announce
[eebus ] DEBUG 2024/11/29 12:43:01 mdns: using avahi
[eebus ] DEBUG 2024/11/29 12:43:01 mdns: new - ski: 829ce7814380c70379edd6a82dd6cf28b1d31ad1 name: Elli-Wallbox-2045A11TNB brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2045A11TNB register: false host: wallbox-2045A11TNB.local port: 4712 addresses: []
[eebus ] DEBUG 2024/11/29 12:43:01 trying to connect to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at wallbox-2045A11TNB.local
[eebus ] DEBUG 2024/11/29 12:43:01 initiating connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at wallbox-2045A11TNB.local:4712/ship/
[eebus ] DEBUG 2024/11/29 12:43:01 mdns: update - ski: 829ce7814380c70379edd6a82dd6cf28b1d31ad1 name: Elli-Wallbox-2045A11TNB brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2045A11TNB register: false host: wallbox-2045A11TNB.local port: 4712 addresses: [192.168.3.17]
[eebus ] DEBUG 2024/11/29 12:43:01 trying to connect to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at wallbox-2045A11TNB.local
[eebus ] DEBUG 2024/11/29 12:43:01 initiating connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at wallbox-2045A11TNB.local:4712/ship/
[eebus ] DEBUG 2024/11/29 12:43:01 connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 failed: dial tcp: lookup wallbox-2045A11TNB.local on [fe80::7ce2:516b:dfb0:3a47%eth0]:53: no such host
[eebus ] DEBUG 2024/11/29 12:43:01 connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 failed: dial tcp: lookup wallbox-2045A11TNB.local on [fe80::7ce2:516b:dfb0:3a47%eth0]:53: no such host
[eebus ] DEBUG 2024/11/29 12:43:01 trying to connect to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at 192.168.3.17
[eebus ] DEBUG 2024/11/29 12:43:01 initiating connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at 192.168.3.17:4712/ship/
[eebus ] DEBUG 2024/11/29 12:43:01 mdns: announce
[eebus ] DEBUG 2024/11/29 12:43:01 mdns: using avahi
[eebus ] DEBUG 2024/11/29 12:43:01 trying to connect to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at wallbox-2045A11TNB.local
[eebus ] DEBUG 2024/11/29 12:43:01 initiating connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at wallbox-2045A11TNB.local:4712/ship/
[eebus ] DEBUG 2024/11/29 12:43:01 connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 failed: dial tcp: lookup wallbox-2045A11TNB.local on [fe80::7ce2:516b:dfb0:3a47%eth0]:53: no such host
[eebus ] DEBUG 2024/11/29 12:43:01 trying to connect to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at 192.168.3.17
[eebus ] DEBUG 2024/11/29 12:43:01 initiating connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 at 192.168.3.17:4712/ship/
[eebus ] DEBUG 2024/11/29 12:43:02 closing outgoing double connection, as the existing connection will be used
[eebus ] DEBUG 2024/11/29 12:43:02 connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1 failed:  closing connection to 829ce7814380c70379edd6a82dd6cf28b1d31ad1: ignoring this connection
[eebus ] DEBUG 2024/11/29 12:43:02 ski 829ce7814380c70379edd6a82dd6cf28b1d31ad1 connected

Fehlermeldung:

    failed configuring loadpoints: failed configuring loadpoint: not found: db:1

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: #error

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: 192.168.3.53  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: 192.168.3.53  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: 192.168.3.53  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- name: Elli
  type: eebus
  ski: 829C E781 4380 C703 79ED D6A8 2DD6 CF28 B1D3 1AD1
  meter: true
  chargedEnergy: false
#- name: Tasmota
  #type: template
  #template: tasmota
  #host: 192.168.3.116 # IP-Adresse oder Hostname
  #user: *****
  #password: *****
  #channel: 1 # Schaltkanal (1-8)
  #standbypower: 1 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional)  
  
loadpoints:
- title: ID. Charger Pro
  charger: Elli
  mode: pv
  vehicle: db:1
#- title: Tasmota # display name for UI
  #charger: Tasmota # charger
  #vehicle: db:3
  #mode: off  
  ##mincurrent: 9 # 9A*220V=1980W minimum charge current (default 6A)
  ##maxcurrent: 10 # 10A*220V=2200W maximum charge current (default 16A)
  #phases: 1 # ev phases (default 3)
    
site:
  title: CUBo
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
    
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2606 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh
        
messaging:
  events:
    stop: # charge stop event
      title: "{{.title}}"
      msg: |
        Laden beendet von {{.vehicleTitle}} um {{printf `%02d:%02d` now.Hour now.Minute}} Uhr mit {{round (divf .chargedEnergy 1000) 2 }} kWh. 
  services:
  - type: ntfy
    uri: https://ntfy.sh/***********          
  
eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          MHcCAQEEIAz3Hc6YK8a+xmhi56BDlMJLOxnYDtSEKEppT1ev++bqoAoGCCqGSM49
          AwEHoUQDQgAEpzCngeYST+PZ5jiNFx3tEi2VLNYTeiMCHIYNbvVtP5GzanznhwX4
          2iPHH7u8352z09edSIgVH4OqjdmfL3g+KQ==
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          MIIBvjCCAWOgAwIBAgIRAvPcuFkn3uSifBVoU8egSCQwCgYIKoZIzj0EAwIwPjEL
          MAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0MxCTAHBgNVBAsTADEVMBMGA1UEAwwM
          RVZDQ19IRU1TXzAxMB4XDTIzMTAwNDEzMjAzM1oXDTMzMTAwMTEzMjAzM1owPjEL
          MAkGA1UEBhMCREUxDTALBgNVBAoTBEVWQ0MxCTAHBgNVBAsTADEVMBMGA1UEAwwM
          RVZDQ19IRU1TXzAxMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEpzCngeYST+PZ
          5jiNFx3tEi2VLNYTeiMCHIYNbvVtP5GzanznhwX42iPHH7u8352z09edSIgVH4Oq
          jdmfL3g+KaNCMEAwDgYDVR0PAQH/BAQDAgeAMA8GA1UdEwEB/wQFMAMBAf8wHQYD
          VR0OBBYEFHE4UABFFoPBgXEYLOeaX4SDv3uAMAoGCCqGSM49BAMCA0kAMEYCIQD1
          La6Sn/zowJlSMCirvGK5YEq9By/xNHp92RJQBgWu7AIhAPKaHn6TxyZ4wxVW5G8m
          NPs/L7BKthzDbWDzNt86Z75a
          -----END CERTIFICATE-----
          
sponsortoken: *****

          

Version: `0.131.7`
```

### Log details

```text
Log als Datei angehängt
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.7

VolkerK62, 2024-11-29:
Es wird geladen, aber Charger hat Status B.
Evtl., weil auf Phase 2 geladen wird?
[lp-1  ] DEBUG 2024/11/26 12:13:47 charge power: 1304W
[lp-1  ] DEBUG 2024/11/26 12:13:47 charge currents: [0.06 5.61 0]A
[lp-1  ] DEBUG 2024/11/26 12:13:47 charger status: B

CarG70, 2024-11-29:
bitte kurze Info, was an der Konfiguration fehlerhaft ist. Vor 0.131.0 lief das System problemlos. Lediglich die Fehlermeldung der Elli hatte genervt.

CarG70, 2024-11-29:
Da im UI beim Laden mit 1-phasigem Kabel immer die mittlere Phase angezeigt wird gehe ich davon aus, dass auch darüber geladen wird. War auch schon immer so. Das lief auch bis 0.131.0 problemlos. Wenn ich wie beschrieben kurz "Schnell" auswähle und anschl. wieder "PV" ist ja auch alles wieder wie gehabt (siehe Screenshot). Das kann aber nicht plötzlich die Lösung sein.
[image]

andig, 2024-11-29:
>  failed configuring loadpoints: failed configuring loadpoint: not found: db:1

Hier geht gar nichts!

VolkerK62, 2024-11-29:
Warum auch immer, die Wallbox merkt nicht, dass geladen wird und bleibt im Status B.
Erst wenn auf "now" umgeschaltet wird, geht der Status auf C.
In #17157 war es auch ein VW ID und auch Elli und auch "Min+PV" mit 6A@1p.
Hast du es mal mit einem höheren "Min.Ladestrom" versucht.