evcc-io/evcc #17497: MG Vehiclde Config: Please Add Region and/or API URL option

mjhgmailcom, 2024-11-30:
### Problem
MG has Multiple Regions, but the evcc MG Vehicle configuration only allows for EU.

MG accounts for one region do not work in other regions.

### Solution
1. Have a "REGION" option on the MG Vehicle configuration to chose the region (e.g. EU or AU)
2. AND/OR allow URL to be specified for MG's API (e.g.  https://gateway-mg-au.soimt.com/api.app/v1/ )

### Background Information
MG configuration is hardcoded to EU URL:
`https://gateway-mg-eu.soimt.com/api.app/v1/`

[image]

However, there is also an AU region (and possibly more to come)
`https://gateway-mg-au.soimt.com/api.app/v1/` <-- Works for Australia and New Zealand.

This does not allow MG vehicles to work in from the AU Region, as they receive the following error:
```
Request failed with status code 400.
cannot create vehicle type 'template': cannot create vehicle type 'mg': 1000036: The account is not registered..
```

However when using the [said-python-mqtt-gateway](https://github.com/SAIC-iSmart-API/saic-python-mqtt-gateway) with the following configuration, the account will log in.

```
services:
  saic-mqtt-gateway:
    image: "saicismartapi/saic-python-mqtt-gateway:latest"
    container_name: "saic-mqtt-gateway"
    environment:   
      - SAIC_REST_URI=https://gateway-mg-au.soimt.com/api.app/v1/
      - SAIC_REGION=au
```

Happy to assist if you need anyone to test or debug.

BTW I am a developer (Windows), but could not find a guide on compiling evcc that worked (and I attempted several different ways) - so I couldn't make the change myself.  Let me know if there is a guide (on Windows?) that I have missed and I might be able to make the change myself.

Thankyou!

andig, 2024-11-30:
For developing please see https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md or contact us on Slack.

mjhgmailcom, 2024-11-30:
Thankyou, I will give it ago.

mjhgmailcom, 2024-12-02:
@jomach 

> @mjhgmailcom I use the integration over home assistant. This means evcc -> mqtt . This means evcc does not connect to mg at all.

Yes, thanks.  I've tried Home assistant and set that up aswell via mqtt using my own solar rules.   Just wanted to see how evcc works, but couldn't do it properly because no AU region.

mjhgmailcom, 2024-12-02:
@andig 
Added Pull Request here: https://github.com/evcc-io/evcc/pull/17544
First, time working with go, I think I'll stick with C# :)

closed 2024-12-04 by andig