evcc-io/evcc #17500: Race Condition in modbus access (RS485 RTU)

rpoisel, 2024-11-30:
### Describe the bug

Reading meter values from RS485 devices results in sporadic `read failed: serial: timeout` error messages (see below for logs and timestamps) without values being read. Reading values works better on one of the two busses than the other one. I am using evcc on a Raspberry Pi 4 (ARM64 architecture).

Bisecting the commits in this repository revealed that the errors occur since this commit: [GitHub](https://github.com/evcc-io/evcc/commit/03f62e559314f4671170d48441062e93539db1dc).

Currently, I guess this issue is related to using 2 devices on 1 RS485 bus. This is my meters configuration:

```yaml
meters:
- name: pv
  usage: pv
  type: template
  template: abb-ab
  modbus: rs485serial
  id: 1
  device: /dev/ttyUSB0  # <-- same as grid
  baudrate: 19200
  comset: "8E1"
- name: grid
  usage: grid
  type: template
  template: abb-ab
  modbus: rs485serial
  id: 2
  device: /dev/ttyUSB0  # <-- same as pv
  baudrate: 19200
  comset: "8E1"
```

I confirmed that this error persists in the latest nightly build on Docker Hub. It prevents me from updating my 0.129.0 installation, which is the most recent release unaffected by this bug.

Furthermore, I guess a potential workaround could be to install an additional RS485 bus to my Raspberry Pi so that there is a 1:1 relationship between RS485 devices and busses.

### Steps to reproduce

1. Copy the configuration pasted below into a file callled `evcc.yaml`.
2. Adjust `sponsortoken` (or remove it) as well as the value of the `plant` key.
3. Start evcc with the file created in steps 1 and 2.
4. Observe logs of evcc

### Configuration details

```yaml
---
sponsortoken: ""  # adjust
plant: ""  # adjust
network:
  schema: http
  host: host.lan
  port: 7070

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.4

log: info
levels:
  cache: error

interval: 30s

meters:
- name: grid
  usage: grid
  type: template
  template: abb-ab
  modbus: rs485serial
  id: 2
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: "8E1"
- name: pv
  usage: pv
  type: template
  template: abb-ab
  modbus: rs485serial
  id: 1
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: "8E1"

chargers:
- name: wallbox
  type: template
  template: go-e-v3
  host: 192.168.1.1

loadpoints:
- title: Wallbox
  charger: wallbox
  mode: pv
  enable:
    threshold: -1000
    delay: 2m
  disable:
    threshold: 450
    delay: 15m

site:
  title: Haus
  meters:
    grid: grid
    pv: pv
```

### Log details

```text
[main  ] INFO 2024/11/30 08:41:38 evcc 0.131.7 (fe68c915)
[main  ] INFO 2024/11/30 08:41:38 using config file: /evcc/evcc.yaml
[db    ] INFO 2024/11/30 08:41:38 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/11/30 08:41:39 listening at :7070
[site  ] INFO 2024/11/30 08:41:39 site config:
[site  ] INFO 2024/11/30 08:41:39   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/11/30 08:41:39     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/11/30 08:41:39     pv 1:      power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2024/11/30 08:41:39 loadpoint 1:
[lp-1  ] INFO 2024/11/30 08:41:39   mode:        pv
[lp-1  ] INFO 2024/11/30 08:41:39   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/11/30 08:41:39   meters:      charge ✓
[lp-1  ] INFO 2024/11/30 08:41:39     charge:    power ✓ energy ✓ currents ✓
[site  ] ERROR 2024/11/30 08:41:44 pv 1 power: read failed: serial: timeout
[lp-1  ] INFO 2024/11/30 08:41:44 car disconnected
[site  ] ERROR 2024/11/30 08:42:14 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:42:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:43:14 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:43:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:44:14 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:44:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:45:14 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:45:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:46:14 grid power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:46:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:47:14 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:47:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:48:14 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:48:44 pv 1 power: read failed: serial: timeout
[site  ] ERROR 2024/11/30 08:49:14 pv 1 power: read failed: serial: timeout
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.7 (fe68c915)

andig, 2024-11-30:
Thanks for the bug report @rpoisel. Can I ask you to contact us on Slack for quicker discussion?

rpoisel, 2024-11-30:
> Thanks for the bug report @rpoisel. Can I ask you to contact us on Slack for quicker discussion? /cc @premultiply

Hi @andig thanks for your quick response! Of course, I'd be happy to do so and just joined the evcc workspace. I'll drop you a message there.

andig, 2024-12-11:
Kein Feedback, closing.

closed 2024-12-11 by andig