evcc-io/evcc #17503: Falsche Darstellung grid, pv und battery

lmfgpeace, 2024-11-30:
### Describe the bug

Die Darstellung in evcc ist inkorrekt.
Die Batterie wird nicht entladen gerade, sondern geladen.
Der Verbrauch ist somit auch nicht bei 5,4 kW.

[image]
[image]

### Steps to reproduce

n.e.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 5s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjkyNzQ0MDAsImlhdCI6MTcyODQ5NjgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.gnfLjIPCnFt3TmCUhoXCmA3nHK5RkQcOrLIejCQfc4w
# trial token, valid until 2024-10-18
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3Mjk4NjQ4MDAsImlhdCI6MTcyOTA4NzIwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.3iF50HEbpf7hF7nob075jJxseDt4c2rFeSS-ZHpxzuE

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri:
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors
[... 276 lines cut ...]
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
n.e.
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

StefanSchoof, 2024-11-30:
Ist das https://github.com/evcc-io/evcc/issues/17461?

naltatis, 2024-11-30:
@StefanSchoof Ja, sieht aus wie ein Duplikat. Hier scheint ein Softwareupdate des Wechselrichters die Bedeutung der Felder geändert zu haben. Ob bewusst oder unbewusst vom Hersteller ist uns aktuell noch nicht klar.

closed 2024-11-30 by naltatis

lmfgpeace, 2024-12-05:
@andig 
@naltatis 

Ich habe von Sungrow eine Antwort erhalten.

Haben mir nur eine Datei zugeschickt:

_Communication Protocol of Residential Hybrid Inverter_

Siehe:

https://distribution-sungrow-europe.intercom-attachments.eu/i/o/d1eu4lj5/28489731/e2c00bb8a778eda92a9d0d0a2e36/TI_20231019_Communication+Protocol+of+Residential+Hybrid+Inverter_V1_1_2_EN+%281%29.pdf?expires=1751410836&signature=f9e95e30e8f82fd1df9129b92863b7ccc82f23b2f8f8373d77e24ed3bc20c4b6&req=0t1rzVH9rzZk2h%2Fz9NJkoaPzt7%2B0A%2BjiGI1gihVemqCCMPpn%0A

andig, 2024-12-05:
Letztes Update 2023/10/18. Wie soll das mit den aktuellen Problemen korrelieren?