evcc-io/evcc #17508: Rechtschreibfehler in "Auswertung Ladeenergie"

brand1011, 2024-11-30:
### Describe the bug

Wenn man auf der EVCC Seite rechts unten auf "x% Sonnenenergie" klickt, erscheint eine kleine Statistik namens "Auswertung Ladeenergie". Dort steckt ein Rechtschreibfehler:

> Unsere Mission: Sonne tanken zum Standard machen. Zusammen mit Ihrer finanziellen Unterstützung, können **dir** es ermöglichen.

### Steps to reproduce

1. Auf EVCC Seite, rechts unten auf "x% Sonnenenergie" klicken

### Configuration details

```yaml
unnötig
```

### Log details

```text
unnötig
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

naltatis, 2024-12-01:
Ja, das was @Maschga sagt :D

closed 2024-12-01 by naltatis

Maschga, 2024-12-01:
Hab's geändert.

brand1011, 2024-12-01:
Danke für den Hinweis zu Weblate! Das kann ich noch nicht.