evcc-io/evcc #17520: Climater function not working

Lightmanster, 2024-12-01:
### Describe the bug

If I activate precondition function in Tesla. The MQTT set pre_condition flag but evcc doesnt react on it.
In pvmode only 6A is used. If I switch to Fast then 11KW is activated. I logs no response at climater flag.

### Steps to reproduce

1. put in min pvmode 
2. set car in precondition
3. only 6 A when there is no PV power
4.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
#  cache: error
   dsmr: fatal
#  lp-1: fatal
#  script: fatal

interval: 20s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

influx:
  url: http://localhost:8086
  database: evcc

mqtt:
  broker: 192.168.100.18:1883
  topic: evcc

meters:
- name: pv1
  type: custom
  power:
     source: mqtt
     topic: domoticz/out/MQTT/Huidige productie Panelen
     jq: .svalue1

- name: my_grid
  type: template
  template: dsmr
  usage: grid
  host: 192.168.100.18
  port: 7801

chargers:
- type: template
  template: alfen 
  id: 1  
  host: 192.168.100.140  
  port: 502  
  modbus: tcpip  
  name: alfen
- type: template
  name: Ebike_shelly
  template: shelly
  host: 192.168.121.199
[... 496 lines cut ...]
Dec 01 08:57:51 raspberrypi evcc[28645]: [lp-2  ] DEBUG 2024/12/01 08:57:51 charge power: 0W
Dec 01 08:57:51 raspberrypi evcc[28645]: [lp-1  ] DEBUG 2024/12/01 08:57:51 charge power: 10891W
Dec 01 08:57:51 raspberrypi evcc[28645]: [lp-1  ] DEBUG 2024/12/01 08:57:51 charge currents: [15.8 15.9 15.9]A
Dec 01 08:57:51 raspberrypi evcc[28645]: [circuit-main] DEBUG 2024/12/01 08:57:51 power: 12550W
Dec 01 08:57:51 raspberrypi evcc[28645]: [circuit-main] DEBUG 2024/12/01 08:57:51 current: 21A
Dec 01 08:57:51 raspberrypi evcc[28645]: [site  ] DEBUG 2024/12/01 08:57:51 grid power: 12550W
Dec 01 08:57:51 raspberrypi evcc[28645]: [site  ] DEBUG 2024/12/01 08:57:51 grid currents: [16 21 16]A
Dec 01 08:57:51 raspberrypi evcc[28645]: [site  ] DEBUG 2024/12/01 08:57:51 pv power: 82W
Dec 01 08:57:51 raspberrypi evcc[28645]: [site  ] DEBUG 2024/12/01 08:57:51 site power: 12150W
Dec 01 08:57:51 raspberrypi evcc[28645]: [lp-2  ] DEBUG 2024/12/01 08:57:51 charge total import: 3.852kWh
Dec 01 08:57:51 raspberrypi evcc[28645]: [lp-2  ] DEBUG 2024/12/01 08:57:51 charger status: B
Dec 01 08:57:51 raspberrypi evcc[28645]: [lp-2  ] DEBUG 2024/12/01 08:57:51 pv charge current: 0A = 0A + -52.8A (12150W @ 1p)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

v 0.131.8 / 0.131.7

andig, 2024-12-01:
Please show

    evcc vehicle —log trace,db:error

Lightmanster, 2024-12-01:
here you go

[main  ] INFO 2024/12/01 12:26:54 using config file: /root/evcc.yaml
[mqtt  ] INFO 2024/12/01 12:26:54 connecting evcc-415928090 at tcp://192.168.100.18:1883
[mqtt  ] DEBUG 2024/12/01 12:26:54 tcp://192.168.100.18:1883 connected
[mqtt  ] TRACE 2024/12/01 12:26:54 recv teslamate/cars/1/battery_level: '69'
[mqtt  ] TRACE 2024/12/01 12:26:54 recv teslamate/cars/1/plugged_in: 'true'
[mqtt  ] TRACE 2024/12/01 12:26:54 recv teslamate/cars/1/charger_actual_current: '6'
[mqtt  ] TRACE 2024/12/01 12:26:54 recv teslamate/cars/1/rated_battery_range_km: '344.0'
[mqtt  ] TRACE 2024/12/01 12:26:54 recv teslamate/cars/1/is_climate_on: 'true'
Soc:            69%
Capacity:       82.0kWh
Charge status:  strconv.ParseBool: parsing "6": invalid syntax
Range:          344km
Climate active: true
OnIdentified:   Mode:now
Features:       [CoarseCurrent]

Lightmanster, 2024-12-01:
No clue where to look. Where does the failure point to ?