evcc-io/evcc #17552: Implementierung Solplanet Wechselrichter und Speicher

mi-ch-a, 2024-12-02:
Hi!

Ich habe den Solplanet Wechselrichter ASW10k H-T3 und einen Solplanet Speicher der Ai-HB G2 Series. Bei den Unterstützen Geräten sind diese nicht aufgeführt. Gibt es eine Möglichkeit diese mit evcc zu verbinden?

Der Wechselrichter ist über einen Dongle mit dem Netzwerk verbunden und hat eine IP Adresse zugeteilt bekommen.

In Homeassistant konnte ich die Geräte über eine HACS Erweiterung einbinden.

mi-ch-a, 2024-12-02:
In einer anderen [Diskussion](https://github.com/evcc-io/evcc/discussions/16666#discussion-7323199) habe ich dies gefunden.

[MB001_ASW.GEN-Modbus-en_V2.1.4.pdf](https://github.com/user-attachments/files/17982869/MB001_ASW.GEN-Modbus-en_V2.1.4.pdf)

andig, 2024-12-04:
Das sieht fast wie normales Modbus aus. Probier gerne mal, Dir die als "custom" Geräte anzulegen. Wenn Du mit den Messwerten zufrieden bist können wir das in ein Template verpacken.

mi-ch-a, 2024-12-07:
Danke, damit hatte ich leider keinen Erfolg. Kann aber sein das ich es falsch gemacht habe. 

Ich habe aber auch noch weiter recherchiert:
- Solplanet würde zur SunspecAllianz gehören. Der Versuch es als Sunspec Hybrid Inverter einzurichten hat aber nicht funktioniert.
- Offener Port ist 8484.  Darüber kann man mithilfe der Seriennummer eine JSON abfragen.

Die Information stammt aus der HomeAssistant sowie iobroker Community. 

Kann man in evcc mit der abgefragten json was machen?

andig, 2024-12-07:
> Kann aber sein das ich es falsch gemacht habe.

Das ist arg unspezifisch

mi-ch-a, 2024-12-07:
Das stimmt. Bin totaler Laie. Hab also versucht die yaml Datei anhand der Infos auf der Internetseite zu bearbeiten. Da habe ich aber zum einpflegen als custom Gerät nichts gefunden. 

Dann hatte ich gelesen das solplanet sunspec nutzt. 
Deshalb habe ich es mit dem Konfigurationstool als sunspec inverter und als sunspec hybrid inverter versucht. 

Das habe ich als TCP/IP und als Serial/Lan Adapter versucht. Dabei kam es zu folgender Fehlermeldung:
Error: cannot create meter type 'custom': power: not a SunSpec device 
bzw. 
connect: connection refused

Bei Port 8484 mit TCP/IP kam als Antwort
Error: cannot create meter type 'custom': power: read tcp 192.168.1.250:42440->192.168.1.237:8484: read: connection reset by                peer
 
habe auch noch folgende Abfragen auspobiert:

evcc sunspec 192.168.1.237:502 Antwort:  dial tcp 192.168.1.237:502: connect: connection refused
evcc sunspec 192.168.1.237:8484 Antwort:   read tcp 192.168.1.250:37178->192.168.1.237:8484: read: connection reset by peer

evcc detect 192.168.1.237 Die Tabelle die dann erscheint ist leer.

andig, 2024-12-07:
Was auch immer du da tust- es gibt nichtmal eine Netzwerkverbindung. Da ist Sunspec das kleinste Problem.

mi-ch-a, 2024-12-10:
Habe den Ansatz mit dem offenen Port 8484 weiterverfolgt. 

Unter Angabe der Seriennummer und einer Devicenummer kann man die benötigten Daten Abrufen.

http://ipadresse:8484/getdevdata.cgi?device=3&sn=Seriennummer

Wobei Device 2 der Wechselrichter, Device 3 der Smartmeter und Device 4 die Batterie ist. Die Ausgaben sehen bei mir so aus: 

Wechselrichter:
{"flg":1,"tim":"20241210142328","tmp":344,"fac":5000,"pac":415,"sac":415,"qac":0,"eto":818,"etd":7,"hto":297,"pf":100,"err":0,"vac":[2364,2354,2384],"iac":[8,9,9],"vpv":[4253,2511,0],"ipv":[45,51,0],"str":[],"stu":1,"pac1":134,"qac1":0,"pac2":128,"qac2":0,"pac3":149,"qac3":0}

Smartmeter
{"flg":1,"tim":"20241210142327","pac":-11,"itd":776,"otd":34,"iet":1134,"oet":275,"mod":1}

Batterie
{"flg":1,"tim":"20241210142420","ppv":319,"etdpv":21,"etopv":988,"cst":10,"bst":3,"eb1":65535,"wb1":65535,"vb":20940,"cb":11,"pb":230,"tb":190,"soc":19,"soh":100,"cli":250,"clo":300,"ebi":14,"ebo":5,"eaci":0,"eaco":0,"vesp":2366,"cesp":0,"fesp":4995,"pesp":0,"rpesp":0,"etdesp":0,"etoesp":52,"vl1esp":2360,"il1esp":0,"pac1esp":0,"qac1esp":0,"vl2esp":2368,"il2esp":0,"pac2esp":0,"qac2esp":0,"vl3esp":2369,"il3esp":0,"pac3esp":0,"qac3esp":0}

In der evcc.yaml habe ich die Abfrage so eingegeben:

```
meters:
  # replace with your real grid meter
  - name: grid
    type: custom
    power: #Power # default value, optionally override # Leistung (W)
      source: http
      uri: http://ipadresse:8484/getdevdata.cgi?device=3&sn=seriennummer
      jq: .pac # Converts a json integer to a boolean value      

  # replace with your real pv system
  - name: pv # Leistung PV
    type: custom
    power: #Power # default value, optionally override # Leistung (W)
      source: http
      uri: http://ipadresse/getdevdata.cgi?device=4&sn=seriennummer
      jq: .ppv # Converts a json integer to a boolean value   

  # replace with your real battery
  - name: battery
    type: custom
    capacity: 10.24 # Akkukapazität in kWh # Optional
    power: #Power # default value, optionally override # Leistung (W)
      source: http
      uri: http://ipadresse:8484/getdevdata.cgi?device=4&sn=seriennummer
      jq: .pb # Converts a json integer to a boolean value 
    soc: # Batterie SOC (%)
      source: http
      uri: http://ipadresse:8484/getdevdata.cgi?device=4&sn=seriennummer
      jq: .soc # Converts a json integer to a boolean value 
```

Das scheint zu funktionieren. 

Sind noch weitere Daten für die Nutzung sinnvoll? 
Im Log werden beim starten auch immer Custom und Energy als meters abgefragt.