evcc-io/evcc #17554: HomeWizard charger template issue

elRadix, 2024-12-02:
### Describe the bug

Unable to use a HomeWizard KWH meter as charger

cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'homewizard': unsupported product type: SDM230-wifi

### Steps to reproduce

create a homewizard meter as per guideline
https://docs.evcc.io/en/docs/devices/chargers#homewizard

### Configuration details

```yaml
chargers:

  - name: my_charger
    type: template
    template: homewizard
    host: 192.168.129.7 # IP address or hostname
    standbypower: 15 # Power values above this value will be considered as charging pow (optional)
```

### Log details

```text
[homewizard] TRACE 2024/12/02 20:20:59 GET http://192.168.129.7/api
[homewizard] TRACE 2024/12/02 20:20:59 {"product_name":"KWh Meter","product_type":"SDM230-wifi","serial":"xxxxxx","firmware_version":"4.06","api_version":"v1"}
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.131.8

andig, 2024-12-02:
/cc @thierolm braucht es den Check auf kompatible Produkte?

thierolm, 2024-12-04:
Das Teil kann, soweit ich das sehe, nicht schalten:
https://www.eastroneurope.com/products/view/sdm230-wifi
==> Kann nicht als `charger` genutzt werden ...

closed 2024-12-04 by andig

elRadix, 2024-12-04:
what's the solution, pls explain in English

thierolm, 2024-12-04:
SDM230 is only a meter. It has ni switch. So U cannot use it as charger, how should evcc switch on/off via this device?
U can define it as meter only.

thierolm, 2024-12-04:
Or U are using a HWE-SKT: https://www.homewizard.com/de-at/shop/wlan-energy-socket/