evcc-io/evcc #17567: No connection to vehicle (BMW)

luckytt69, 2024-12-03:
### Describe the bug

No connection to vehicle BMW since round about 2 weeks. 

### Steps to reproduce

1. EVCC software updated to newest version. 
2. new start of EVCC Server
3. Pluged and unpluged vehicle several time. 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070                                                                                   network:                                                                                                                  schema: http                                                                                                            host: evcc.local # .local suffix announces the hostname on MDNS                                                         port: 7070                                                                                                                                                                                                                                    log: debug                                                                                                              levels:                                                                                                                   cache: error                                                                                                                                                                                                                                  # unique installation id                                                                                                plant: 59a942717020c33deb7c8d3c3ce9cde2a6b262b0da0aed1bfc9987886d991ae1                                                                                                                                                                         interval: 30s # control cycle interval

sponsortoken: ###

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: go-e-controller
  usage: grid
  host: 192.168.178.110
  name: grid1
- type: template
  template: go-e-controller
  usage: pv
  host: 192.168.178.110
  name: pv2

chargers:
- type: template
  template: go-e-v3
  host: 192.168.178.108
  name: wallbox4

vehicles:
- type: template
  template: bmw
  title: 330E
  user: r.maurer1@gmx.net
  password: 
  vin: WBA6N31050FM00819
  capacity: 12
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox4
 vehicle: ev3
  mode: pv

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
```

### Log details

```text
[main ] INFO 2024/12/03 14:30:08 evcc 0.131.8
[main ] INFO 2024/12/03 14:30:08 using config file: /etc/evcc.yaml
[db ] INFO 2024/12/03 14:30:08 using sqlite database: /var/lib/evcc/evcc.db
[... 210 lines cut ...]
[lp-1 ] DEBUG 2024/12/03 14:37:09 charge power: 1333W
[lp-1 ] DEBUG 2024/12/03 14:37:09 charge currents: [6.04 0 0]A
[site ] DEBUG 2024/12/03 14:37:09 grid power: 1118W
[site ] DEBUG 2024/12/03 14:37:09 pv power: 449W
[site ] DEBUG 2024/12/03 14:37:09 site power: 1118W
[lp-1 ] DEBUG 2024/12/03 14:37:09 detected active phases: 1p
[lp-1 ] DEBUG 2024/12/03 14:37:09 charge total import: 816.885kWh
[lp-1 ] DEBUG 2024/12/03 14:37:09 charger status: C
[lp-1 ] DEBUG 2024/12/03 14:37:09 !! pvScalePhases DOWN activePhases: 1, available power: 215W, scalable: false
[lp-1 ] DEBUG 2024/12/03 14:37:09 !! pvScalePhases UP maxPhases: 1, available power: 215W, scalable: false
[lp-1 ] DEBUG 2024/12/03 14:37:09 !! pvScalePhases EXIT
[lp-1 ] DEBUG 2024/12/03 14:37:09 pv charge current: min 6A > 1.14A (1118W @ 1p, battery: false)
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

StefanSchoof, 2024-12-03:
Does https://github.com/evcc-io/evcc/discussions/17514#discussioncomment-11423632 help?

0x3d13f, 2024-12-03:
As far as I know something has changed with BMW. See https://docs.evcc.io/docs/devices/vehicles#bmw

andig, 2024-12-03:
See https://github.com/evcc-io/evcc/issues/17332

closed 2024-12-03 by andig