evcc-io/evcc #17580: Tariffs calculated incorrectly with data source from ENTSOE when currency is other than EUR

Millox, 2024-12-04:
### Describe the bug

The calculation of the current price for electricity is done incorrectly when using other than EUR for the ENTSOE tariff price source. The api always fetches the price in EUR, but the presentation is in local currency, which in the case of SEK leads to an underestimation of the electricity price by a factor of 10.

Currently (2024-12-04, 08:00-09:00) the price in SE4 is 3.13 SEK/kWh which is 0.27 EUR/kWh

Calculation from EVCC:
[image]

Expected (with configuration as below):
(3.13 + 0.73744) * 1.25 =  4.8 SEK / kWh

Actual:
(0.270 + 0.73744) * 1.25 = 1.26 SEK / kWh

### Steps to reproduce

1. Configure tariffs with currency other than EUR
2. Configure price fetching from entso-e

### Configuration details

```yaml
tariffs:
  currency: SEK
  grid:
    type: entsoe
    domain: BZN|SE4
    securitytoken: xxx
    charges: 0.73744
    tax: 0.25
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.137.7

andig, 2024-12-04:
> The api always fetches the price in EUR

I'm not familiar with it- does it offer anything else? Maybe we could also return local currency.

> but the presentation is in local currency

Presentation is as configured by user. If you have EUR data you'll need to configure EUR.

Alternative solution: use a formula for currency conversion using https://github.com/evcc-io/evcc/pull/17002

bartashevich, 2024-12-04:
I think ENTSO-e only provides EUR Tariff.

Try following to convert to your local currency:
```
tariffs:
  currency: SEK
  grid:
    type: entsoe
    domain: BZN|SE4
    securitytoken: xxx
    charges: 0
    tax: 0
    formula: (price*11.5765 + 0.73744) * 1.25
```

Millox, 2024-12-04:
Yeah, so... Obviously I wasn't very happy with this response, static exchange rate will always be a hack, and apparently ENTSO-E only provides EUR data as stated above. SOOO, I kinda fixed it in another way

./evcc -c evcc.yaml tariff -l debug
[main  ] INFO 2024/12/04 21:32:40 evcc 0.131.8 (55d3c8eb)
[main  ] INFO 2024/12/04 21:32:40 using config file: evcc.yaml
[db    ] INFO 2024/12/04 21:32:40 using sqlite database: /home/mattias/.evcc/evcc.db
[mqtt  ] INFO 2024/12/04 21:32:40 connecting evcc-586182962 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/12/04 21:32:40 tcp://localhost:1883 connected
[nordpool] DEBUG 2024/12/04 21:32:40 downloading price data
grid:
From                   To                     Price/Cost
2024-12-04 23:00:00    2024-12-05 00:00:00    0.652
2024-12-05 00:00:00    2024-12-05 01:00:00    0.646
2024-12-05 01:00:00    2024-12-05 02:00:00    0.648
2024-12-05 02:00:00    2024-12-05 03:00:00    0.642
2024-12-05 03:00:00    2024-12-05 04:00:00    0.642
2024-12-05 04:00:00    2024-12-05 05:00:00    0.664
2024-12-05 05:00:00    2024-12-05 06:00:00    1.059
2024-12-05 06:00:00    2024-12-05 07:00:00    1.620
2024-12-05 07:00:00    2024-12-05 08:00:00    1.595
2024-12-05 08:00:00    2024-12-05 09:00:00    1.481
2024-12-05 09:00:00    2024-12-05 10:00:00    1.405
2024-12-05 10:00:00    2024-12-05 11:00:00    1.207
2024-12-05 11:00:00    2024-12-05 12:00:00    0.894
2024-12-05 12:00:00    2024-12-05 13:00:00    0.833
2024-12-05 13:00:00    2024-12-05 14:00:00    0.739
2024-12-05 14:00:00    2024-12-05 15:00:00    0.736
2024-12-05 15:00:00    2024-12-05 16:00:00    0.721
2024-12-05 16:00:00    2024-12-05 17:00:00    0.709
2024-12-05 17:00:00    2024-12-05 18:00:00    0.615
2024-12-05 18:00:00    2024-12-05 19:00:00    0.483
2024-12-05 19:00:00    2024-12-05 20:00:00    0.371
2024-12-05 20:00:00    2024-12-05 21:00:00    0.375
2024-12-05 21:00:00    2024-12-05 22:00:00    0.416
2024-12-05 22:00:00    2024-12-05 23:00:00    0.346

This was based on the energinet tariff source, but uses Nordpool data instead which have SE/NO/DK and other areas as well. Will create pull request as soon as I have figured out how to. And cleaned up the code somewhat.

Maybe close this bug report with a documentation update only and then I will provide the Nordpool tariff source later.

andig, 2024-12-04:
Don‘t eneginet and elering also provide Nordpool data? If we can get them from the source that wouldn‘t hurt 👍🏻

closed 2024-12-04 by andig

Millox, 2024-12-05:
Nordpool provides data for
["EE","LT","LV","AT","BE","FR","GER","NL","PL","DK1","DK2","FI","NO1","NO2","NO3","NO4","NO5","SE1", "SE2","SE3","SE4","TEL","SYS"]
in currencies
["DKK","EUR","NOK","PLN","RON","SEK"]

Although some combinations do not report any values according to my testing. I just noticed the Nordpool feature request and will attach a patch to that thread when I've cleaned up the code a bit so that more people can test.