evcc-io/evcc #17583: Fronius Modbus Batteriesteuerung fehlerhaft bei Netzladen der Hausbatterie

mweihberg, 2024-12-04:
### Describe the bug

Ich verzeichne ein komisches Verhalten mit meinen zwei Fronius Wechselrichtern. siehe Diskussion #17505 

### Steps to reproduce

Beispiel: 
Hausverbauch 0,5kw
Fronius Gen24 mit Batterie Produktion 0,4kw
Fronius Symo über SolarApi Produktion 0,8kw
Kein Auto angeschlossen.

Erwartetes Verhalten:
Batterieladung 0,7kw.

1. Frisches System. Binde 2x Fronius Wechselrichter ein. Einer über Modbus mit aktiver Batteriesteuerung und einer über SolarApi. Dann wird erwartetes Verhalten erreicht.
2. Netzladen der Hausbatterie testen, dadurch aktive Batteriesteuerung und bei meinem einphasigen WR Gen24 wird dann auch korrekterweise mit 4,2kw aus dem Netz in die Batterie geladen => aktive Batteriesteuerung funktioniert.
3. Deaktivierung des Netzladens. Erwartetes Verhalten 0,7kw in Hausbatterie. Verhalten stattdessen: 0,4kw in Hausbatterie (von Gen24). 0,2kw Einspeisung (Rest von 0,7kw Symo abzgl. Hausverbauch 0,5kw).

Es scheint mir, als würde sich die Batteriesteuerung irgendwie nicht richtig zurücksetzen.

Ich würde gerne mit Logs helfen, weiß aber nicht was ich da collecten soll.

### Configuration details

```yaml
Database Konfiguration
```

### Log details

```text
Bitte um Hilfestellung was konkret collected werden soll. Alles?
```

### What type of operating system are you running?

HomeAssistant Add-on

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

mweihberg, 2024-12-04:
Btw. funktioniert die aktive Batteriesteuerung über die fronius-solarapi-v1 NICHT. 
Eine Aktivierung des Netzladens führt zwar zur korrekten Anzeige auf der UI, aber die Batterie wird trotzdem nicht aus dem Netz geladen.

Das Template setzt unter `charge` auch einfach nix.
[https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-solarapi-v1.yaml](https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-solarapi-v1.yaml)

mweihberg, 2024-12-04:
Und ich habe glaube ich die betroffene Stelle direkt gefunden.
[https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-gen24.yaml](https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-gen24.yaml)
```
batterymode: # model 124
    source: switch
    switch:
    - case: 1 # normal
      set:
        source: sequence
        set:
        - source: const
          value: 0 # off
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:ChaGriSet
        - source: const
          value: 0
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:StorCtl_Mod
        - source: const
          value: 100 # %
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:OutWRte
```

Hier wird offensichtlich im `normal` Mode die Ladeleistung auf den Output den Wechselrichters beschränkt. Besser wäre es den Output gar nicht zu beschränken und das SmartMeter diesen Job erledigen zu lassen.

mweihberg, 2024-12-04:
Wobei hier steht die Value bezieht sich immer auf `WchaMax`
[https://manuals.fronius.com/html/4204102649/de.html#0_m_0000016557](https://manuals.fronius.com/html/4204102649/de.html#0_m_0000016557)

Die Frage ist ja wie sieht es aus, wenn sich das System normal verhält bzw. gar keine Register von außen gesetzt wurden. Ich muss mal schauen ob ich es heute Abend hinbekomme den Modbus auszulesen.

andig, 2024-12-05:
Da du die Register gefunden hast und lesen kannst gerne PR falls dort etwas falsch ist. Ich sehe aktuell kein Todo.

mweihberg, 2024-12-05:
Ich könnte parallel mal schauen ob ich nebenher die Solarapi ertüchtigen kann die Batterie via Netz zu laden. Da bin ich deutlich fitter 😬

andig, 2024-12-07:
Ich mache zu bis es einen Vorschlag zur Änderung der Register gibt.

mweihberg, 2025-01-22:
Ich habe den Fehler mit `evcc sunpec` herausfinden können.
Wenn evcc an ist, dann wird im `ChaGriSet` bei `normal` und `hold` auf den Wert 0 gesetzt.

[image]

Wenn ich Modbus kurzzeitig am WR ausschalte um es dann wieder einzuschalten um das Default Verhalten wieder herzustellen, dann ist ChaGriState IMMER = 1.

[image]

Es gibt in Fronius WR in der WebUI eine Einstellung zum Thema Laden von Batterien aus anderen Quellen. Egal welche Einstellung ich dort setze, ChaGriState ändert sich nicht und ist immer = 1. 

[image]

mweihberg, 2025-01-22:
Somit wäre mein Vorschlag im Modus `normal` den `ChaGriState` auf = 1 zu belassen.
https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-gen24.yaml

andig, 2025-01-22:
Wir setzen das ja nur bei `charge`. Wenn es eh immer auf 1 ist könnten wir uns komplett sparen, da irgendwas zu setzen. Input kam damals unter anderen von @mikethetux in https://github.com/evcc-io/evcc/issues/10912.

Wie @premultiply schon sagte: PR welcome, bitte unbedingt testen.

mweihberg, 2025-01-22:
Ich kann es versuchen demnächst lokal mal zu bauen und zu testen - da ich beruflich wirklich stark eingebunden kann ich da nix versprechen.

Allerdings würde ich gerne nochmal revidieren was aktuell im Template passiert (https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/fronius-gen24.yaml):

```
batterymode: # model 124
    source: switch
    switch:
    - case: 1 # normal
      set:
        source: sequence
        set:
        - source: const
          value: 0 # off
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:ChaGriSet
        - source: const
          value: 0
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:StorCtl_Mod
        - source: const
          value: 100 # %
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:OutWRte
    - case: 2 # hold
      set:
        source: sequence
        set:
        - source: const
          value: 0 # off
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:ChaGriSet
        - source: const
          value: 2
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:StorCtl_Mod
        - source: const
          value: 0 # %
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:OutWRte
        - source: const
          value: 0 # s
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:InOutWRte_RvrtTms
    - case: 3 # charge
      set:
        source: sequence
        set:
        - source: const
          value: 1 # off
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:ChaGriSet
        - source: const
          value: 2
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:StorCtl_Mod
        - source: const
          value: -{{ .maxchargerate }} # %
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:OutWRte
        - source: const
          value: 0 # s
          set:
            source: sunspec
            uri: {{ .host }}:{{ .port }}
            id: 1
            value: 124:0:InOutWRte_RvrtTms
```

Fall 1 normal:
- ChaGriSet = 0 => Keine Ladung weiterer Erzeuger möglich
- StorCtl_Mod =0 => Keine Grenzwerte der Ladung/Entladung aktiv
- OutWRTe = 100% => Theoretisch 100% Ladung möglich

Fall 2 hold (keine Entladung):
- ChaGriSet = 0 => Keine Ladung weiterer Erzeuger möglich
- StorCtl_Mod = 2 => Entladegrenzwert aktiv
- OutWRTe = 0% => Entladegrenzwert = 0%

Fall 3 charge (Ladung je nach `maxchargerate`):
- ChaGriSet = 1 => Ladung Netz und weitere Generatoren aktiv
- StorCtl_Mod = 2 => Entladegrenzwert aktiv
- OutWRTe => -`maxchargerate` negative Entladung und somit Ladung

Selbst wenn jetzt im Modus `normal` ChaGriSet = 1 ist, dann heißt dass nicht, dass aus dem Netz geladen wird. Das macht der Fronius nicht. Nur wenn man dann aktiv Batterieprogramme setzt. Aber ohne ChaGriSet = 1 wird es auch weiteren Erzeugern im AC Netz des Hauses nicht erlaubt ihren Überschuss in die Batterie einzuspeisen. Der verfügbare Überschuss wird durch das Smart Meter automatisch geregelt.

andig, 2025-01-22:
Also einfach `ChaGriSet` komplett weg lassen wenns das eh nicht braucht (oder es sowieso 1 ist)...

TomF79, 2025-01-22:
Eventuell passt das hie dazu:

https://github.com/evcc-io/evcc/discussions/18277

MikeTheTux, 2025-01-22:
> Input kam damals unter anderen von [@MikeTheTux](https://github.com/MikeTheTux) in [#10912](https://github.com/evcc-io/evcc/issues/10912).

Ich habe damals den Issue https://github.com/evcc-io/evcc/issues/10912 gestellt, da ich Interesse am Discharge Control (Hold) hatte.
Um die Entwicklung zu enablen, habe ich dem EVCC Team Zugang zu meinem Inverter gegeben und Dokumente (PDF und Modubus-Register) zur Verfügung gestellt.
Im Anschluss habe ich nur den UseCase Discharge Control (Hold) getestet. Keine gezwungene Ladung aus dem Netz, keine Ladung mit maxChargeRate, usw.. Evtl. erklärt das, warum diverse Szenarien nicht funktionieren.

> Also einfach ChaGriSet komplett weg lassen wenns das eh nicht braucht (oder es sowieso 1 ist)...

In dem in https://github.com/evcc-io/evcc/issues/10912 referenzierten Dokument, kommt bei den einzelnen UseCases ChaGriSet ebenfalls nicht vor.

mweihberg, 2025-01-22:
> Eventuell passt das hier dazu:
> 
> [#18277](https://github.com/evcc-io/evcc/discussions/18277)

Ist interessant, die Zwischenspannungsfehler hatte ich auch vermehrt in letzter Zeit. Ich habe gerade nochmal in das Solarweb von Fronius geschaut. Die Zwischenspannungsfehler sind erst seitdem ich evcc benutze. Kann das aber gerade nicht in Korrelation mit einem bestimmten Zustand setzen.

> Ich habe damals den Issue [#10912](https://github.com/evcc-io/evcc/issues/10912) gestellt, da ich Interesse am Discharge Control (Hold) hatte. Um die Entwicklung zu enablen, habe ich dem EVCC Team [...]

Das Dokument kenne ich auch - daher habe ich die Informationen. Hier ist noch ein interessanter Link: [libe.net](https://www.libe.net/byd-modbus) - von dort kommt man hierhin [https://forum.iobroker.net/topic/65205/modbus-fronius-gen24](https://forum.iobroker.net/topic/65205/modbus-fronius-gen24) und da gibt es eine Excel mit allen Modbus Registern. 

[image]

Da es der Default Wert ist, wenn man Modbus deaktiviert und wieder aktiviert gehe ich davon aus, dass es keinerlei Auswirkung hat das grundsätzlich auf = 1 zu lassen.

TomF79, 2025-01-22:
> Ist interessant, die Zwischenspannungsfehler hatte ich auch vermehrt in letzter Zeit. Ich habe gerade nochmal in das Solarweb von Fronius geschaut. Die Zwischenspannungsfehler sind erst seitdem ich evcc benutze. Kann das aber gerade nicht in Korrelation mit einem bestimmten Zustand setzen.
> 

Bei mir kam der Fehler hauptsächlich zu Ladebeginn sobald die Batteriesperre aktiv wurde.

MikeTheTux, 2025-01-22:
aus https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2649.pdf

[image]

andig, 2025-01-22:
…es erklärt aber nicht, warum eine 0 dazu führen sollte, dass nicht aus anderen Quellen als dem Netz geladen werden kann?

MikeTheTux, 2025-01-22:
Hmm. Wo hängt der SmartMeter vom 1. WR mit der Batterie? Hat er eine Möglichkeit zu erkennen, dass der 2. WR gerade die 0,2 kw einspeist?

mweihberg, 2025-01-22:
Natürlich, denn er hängt am Einspeisepunkt. Daher ist der Gesamtsaldo des Haues bekannt. Es funktioniert ja auch einwandfrei ohne die Einmischung von evcc über das Modbus Register.

Wenn die beiden Eigenschaften ChaGriSet und die Einstellung aus der UI UND Verknüpft sind, dann ist es doch klar.
ChaGriSet = 0 UND UI = 1 => 0
ChaGriSet = 1 UND UI = 1 => 1
ChaGriSet = 1 UND UI = 0 => 0

Somit muss beides erlaubt sein, damit es überhaupt funktioniert.

Den letzten Fall könnte ich morgen mal testen. Ich erzwinge das Netzladen über evcc und stelle das Laden aus dem Netz in der UI aus. Ergebnis müsste somit keine Ladung aus dem Netz sein.

Ich habe es außerdem gerade geschafft, die Solution zu bauen und werde das Setzen von ChaGriSet im Modus normal einmal in den nächsten Tagen durchtesten.

TomF79, 2025-01-23:
> Natürlich, denn er hängt am Einspeisepunkt. Daher ist der Gesamtsaldo des Haues bekannt. Es funktioniert ja auch einwandfrei ohne die Einmischung von evcc über das Modbus Register.
> 
> Wenn die beiden Eigenschaften ChaGriSet und die Einstellung aus der UI UND Verknüpft sind, dann ist es doch klar. ChaGriSet = 0 UND UI = 1 => 0 ChaGriSet = 1 UND UI = 1 => 1 ChaGriSet = 1 UND UI = 0 => 0
> 
> Somit muss beides erlaubt sein, damit es überhaupt funktioniert.
> 
> Den letzten Fall könnte ich morgen mal testen. Ich erzwinge das Netzladen über evcc und stelle das Laden aus dem Netz in der UI aus. Ergebnis müsste somit keine Ladung aus dem Netz sein.
> 
> Ich habe es außerdem gerade geschafft, die Solution zu bauen und werde das Setzen von ChaGriSet = 1 im Modus normal einmal in den nächsten Tagen durchtesten.
> 
> Reicht es dafür einmal das Template zu ändern, dann zu kompilieren und es zu starten? Ich bin C# Dev - habe keine Ahnung von Go und Node. Außerdem bin ich nicht so tief in den Source Code eingestiegen.

Das erklärt dann natürlich auch was @mkrill und ich festgestellt haben, dass nach benutzen der Batteriesperre von evcc die Laderegeln für das Netzladen in der WebUi des Gen24 ohne Funktion sind und erst nach manuellen Reset der Modbus Einstellungen wieder funktioniert, da ChaGridSet Standard wieder auf 1 gesetzt wird.

Wisley7, 2025-01-24:
Bei der Steuerung meines Akkus über den Fronius Symo-Hybrid setze ich einfach den OutWrte auf folgende Werte:
Normal: 100
Hold: 0
Aus Netz laden : -100

StorCtl_Mod steht bei mir immer auf 3

Sonst wurde und wird nix verändert. 
Ich denke das wird beim Gen24 auch funktionieren

closed 2025-01-26 by premultiply

mweihberg, 2025-02-10:
Heute getestet mit Nightly: Funktioniert wunderbar!

Erst Netzladen erzwungen mit hohem Strompreis, dann ausgeschaltet und festgestellt, dass sämtlicher Überschuss (nicht nur der des WR mit der Batterie) in die Batterie geladen wird.

Batteriesperre während schnellem Laden funktioniert auch.