evcc-io/evcc #17585: Amperfied OCPP

maatinh, 2024-12-04:
### Describe the bug

When adding Amperfied connect.solar Wallbox using OCPP I get an error message in the log and the wallbox is not visible in evcc.

With latest firmware 5.1 (published Dec. 3rd) Amperfied supports OCPP 1.6 Smart Charging.

Additional info - if stationid is uncommented, evcc is not starting.

### Steps to reproduce

1. add WB using OCPP in yaml
2. configure evcc in WB
3. start evcc

### Configuration details

```yaml
evcc.yaml
[...]
  - name: Stellplatz-OCPP
    type: template
    template: ocpp
#    stationid: stellplatz
[...]

OCPP Parameters from WB:

OCPP-Parameter

AllowOfflineTxForUnknownId

AuthorizationCacheEnabled

AuthorizeRemoteTxRequests
10

BlinkRepeat
1

ChargeProfileMaxStackLevel
Current, Power

ChargingScheduleAllowedChargingRateUnit
1

ChargingScheduleMaxPeriods
900

ClockAlignedDataInterval
50

ConfigurationMaxKeys
60

ConnectionTimeOut
0.RST, 1.RST, 2.RTS

ConnectorPhaseRotation
5

ConnectorPhaseRotationMaxLength

ConnectorSwitch3to1PhaseSupported
60

HeartbeatInterval
50

LightIntensity

LocalAuthListEnabled
100

LocalAuthListMaxLength

LocalAuthorizeOffline

LocalPreAuthorize
FATAL

loglevel
1

MaxChargingProfilesInstalled
0

MaxEnergyOnInvalidId
Power.Active.Import,Energy.Active.Import.Register,Voltage.Outlet,Current.Import.Inlet.L1,Current.Import.Inlet.L2,Current.Import.Inlet.L3,Voltage.Inlet.L1,Voltage.Inlet.L2,Voltage.Inlet.L3,Power.Active.Import.L1-N,Power.Active.Import.L2-N,Power.Active.Import.L3-N

MeterValuesAlignedData
10

MeterValueSampleInterval
Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offered,SoC

MeterValuesSampledData
700

MeterValuesSampledDataMaxLength
1

MinimumStatusDuration
1

NumberOfConnectors

ReserveConnectorZeroSupported
3

ResetRetries
100

SendLocalListMaxLength

StopTransactionOnEVSideDisconnect

StopTransactionOnInvalidId
Energy.Active.Import.Register

StopTxnAlignedData
1

StopTxnAlignedDataMaxLength
Energy.Active.Import.Register

StopTxnSampledData
1

StopTxnSampledDataMaxLength
Core, FirmwareManagement, LocalAuthListManagement, Reservation, SmartCharging, RemoteTrigger

SupportedFeatureProfiles
6

SupportedFeatureProfilesMaxLength
FTP,FTPS,HTTP,HTTPS

supportedFileTransferProtocols
1

TransactionMessageAttempts
1

TransactionMessageRetryInterval

UnlockConnectorOnEVSideDisconnect
1

WebSocketPingInterval
ws://evcc.fritz.box:8887/stellplatz

WebSocketUrl
```

### Log details

```text
[ocpp ] INFO 2024/12/04 16:32:24 charge point connected, registering: stellplatz
[ocpp ] ERROR 2024/12/04 16:32:44 ocpp message (1967513926): NotSupported - Unsupported feature FirmwareStatusNotification ([2,"1967513926","FirmwareStatusNotification", { "status": "Installed" }])
[ocpp ] ERROR 2024/12/04 16:32:46 ocpp message (1653377373): OccurrenceConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue[0].Value required but not found for feature MeterValues ([2,"1653377373","MeterValues", { "meterValue": [ { "timestamp": "2024-12-04T15:32:46Z", "sampledValue": [ { "context": "Trigger", "measurand": "Power.Active.Import", "value": "" }, { "context": "Trigger", "measurand": "Energy.Active.Import.Register", "value": "" }, { "context": "Trigger", "measurand": "Current.Import", "value": "" }, { "context": "Trigger", "measurand": "Voltage", "value": "" }, { "context": "Trigger", "measurand": "Current.Offered", "value": "" }, { "context": "Trigger", "measurand": "Power.Offered", "value": "" }, { "context": "Trigger", "measurand": "SoC", "value": "" } ] } ], "connectorId": 0, "transactionId": 0 }])
[ocpp-1] WARN 2024/12/04 16:33:15 meter timeout
```

### What type of operating system are you running?

Docker container

### Nightly build

- [X] I have verified that the issue is reproducible with the latest nightly build

### Version

0.131.8

mfuchs1984, 2024-12-04:
Looks like it sends empty measurands 
`"Power.Active.Import", "value": ""`
This is why evcc runs into a timeout. Does it send valid values when a car is connected?

maatinh, 2024-12-04:
Meanwhile I see in the TRACE Log that the following entries are sent. However, the WB does not appear in evcc.

```

[ocpp ] TRACE 2024/12/04 18:29:42 send stellplatz: [2,"3668985748","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/12/04 18:29:42 recv stellplatz: [3,"3668985748", { "status": "Accepted" }]
[ocpp ] TRACE 2024/12/04 18:29:42 recv stellplatz: [2,"1694830070","MeterValues", { "meterValue": [ { "timestamp": "2024-12-04T17:29:42Z", "sampledValue": [ { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Power.Active.Import", "unit": "W", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "1024127" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1", "measurand": "Current.Import", "unit": "A", "value": "0.0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1-N", "measurand": "Voltage", "unit": "V", "value": "235.3" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Current.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Power.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "SoC", "unit": "V", "value": "0" } ] } ], "connectorId": 1, "transactionId": 0 }]
[ocpp ] TRACE 2024/12/04 18:29:42 send stellplatz: [3,"1694830070",{}]
``` 

I need to check when a car is connected.

maatinh, 2024-12-04:
Shame on me - I only added the charger to evcc.yaml but no loadpoint. When adding the loadpoint I still get log-entries but the OCPP Charger is appearing in the UI.

Log Entries:

```
[main  ] INFO 2024/12/04 18:48:55 evcc 0.131.8 (55d3c8eb)
[main  ] INFO 2024/12/04 18:48:55 using config file: /etc/evcc.yaml
[db    ] INFO 2024/12/04 18:48:55 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/12/04 18:48:55 listening at :7070
[ocpp  ] INFO 2024/12/04 18:48:59 charge point connected, registering: Stellplatz
[ocpp  ] ERROR 2024/12/04 18:49:03 ocpp message (982936784): OccurrenceConstraintViolation - Field Call.Payload.MeterValue[0].SampledValue[0].Value required but not found for feature MeterValues ([2,"982936784","MeterValues",
{
"meterValue": [
{
"timestamp": "2024-12-04T17:49:03Z",
"sampledValue": [
{
"context": "Trigger",
"measurand": "Power.Active.Import",
"value": ""
},

{
"context": "Trigger",
"measurand": "Energy.Active.Import.Register",
"value": ""
},

{
"context": "Trigger",
"measurand": "Current.Import",
"value": ""
},

{
"context": "Trigger",
"measurand": "Voltage",
"value": ""
},

{
"context": "Trigger",
"measurand": "Current.Offered",
"value": ""
},

{
"context": "Trigger",
"measurand": "Power.Offered",
"value": ""
},

{
"context": "Trigger",
"measurand": "SoC",
"value": ""
}
]
}
],
"connectorId": 0,
"transactionId": 0
}])
[ocpp-1] WARN 2024/12/04 18:49:33 meter timeout
[lp-1  ] WARN 2024/12/04 18:49:35 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-2  ] WARN 2024/12/04 18:49:35 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-3  ] WARN 2024/12/04 18:49:35 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] WARN 2024/12/04 18:49:36 `MaxGridSupplyWhileBatteryCharging` is deprecated- use `maxACPower` in pv configuration instead
[site  ] INFO 2024/12/04 18:49:36 site config:
[site  ] INFO 2024/12/04 18:49:36   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/12/04 18:49:36     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/12/04 18:49:36     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/12/04 18:49:36     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/12/04 18:49:36   vehicles:
[site  ] INFO 2024/12/04 18:49:36     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/12/04 18:49:36     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/12/04 18:49:36 loadpoint 1:
[lp-1  ] INFO 2024/12/04 18:49:36   mode:        pv
[lp-1  ] INFO 2024/12/04 18:49:36   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/12/04 18:49:36   meters:      charge ✓
[lp-1  ] INFO 2024/12/04 18:49:36     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/12/04 18:49:36 loadpoint 2:
[lp-2  ] INFO 2024/12/04 18:49:36   mode:        pv
[lp-2  ] INFO 2024/12/04 18:49:36   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
[lp-2  ] INFO 2024/12/04 18:49:36   meters:      charge ✓
[lp-2  ] INFO 2024/12/04 18:49:36     charge:    power ✓ energy ✓ currents ✓
[lp-3  ] INFO 2024/12/04 18:49:36 loadpoint 3:
[lp-3  ] INFO 2024/12/04 18:49:36   mode:        pv
[lp-3  ] INFO 2024/12/04 18:49:36   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
[lp-3  ] INFO 2024/12/04 18:49:36   meters:      charge ✓
[lp-3  ] INFO 2024/12/04 18:49:36     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2024/12/04 18:49:38 car disconnected
[lp-2  ] INFO 2024/12/04 18:50:07 car disconnected
[lp-3  ] INFO 2024/12/04 18:50:38 car disconnected

```

maatinh, 2024-12-04:
When connecting a Car evcc is taking a while until I see the Car being connected. Now when its showing up, its saying that its waiting for Autorization:

[image]

The WB is showing this:

[image]

Log:

```
[ocpp ] TRACE 2024/12/04 19:15:46 recv Garage: [2,"1329202900","StatusNotification", { "info": "", "vendorId": "", "vendorErrorCode": "", "connectorId": 1, "errorCode": "NoError", "status": "Preparing", "timestamp": "2024-12-04T18:15:46Z" }]
[ocpp-1] DEBUG 2024/12/04 19:15:46 waiting for local authentication
[ocpp ] TRACE 2024/12/04 19:15:46 send Garage: [3,"1329202900",{}]
[ocpp ] TRACE 2024/12/04 19:15:55 send Garage: [2,"2569752676","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/12/04 19:15:55 recv Garage: [2,"655495367","MeterValues", { "meterValue": [ { "timestamp": "2024-12-04T18:15:55Z", "sampledValue": [ { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Power.Active.Import", "unit": "W", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "89624" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1", "measurand": "Current.Import", "unit": "A", "value": "0.0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1-N", "measurand": "Voltage", "unit": "V", "value": "236.3" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Current.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Power.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "SoC", "unit": "V", "value": "0" } ] } ], "connectorId": 1, "transactionId": 0 }]
[ocpp ] TRACE 2024/12/04 19:15:55 send Garage: [3,"655495367",{}]
[ocpp ] TRACE 2024/12/04 19:15:55 recv Garage: [3,"2569752676", { "status": "Accepted" }]
[ocpp ] TRACE 2024/12/04 19:16:05 send Garage: [2,"2850175740","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/12/04 19:16:05 recv Garage: [2,"733053144","MeterValues", { "meterValue": [ { "timestamp": "2024-12-04T18:16:05Z", "sampledValue": [ { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Power.Active.Import", "unit": "W", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "89624" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1", "measurand": "Current.Import", "unit": "A", "value": "0.0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1-N", "measurand": "Voltage", "unit": "V", "value": "235.9" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Current.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Power.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "SoC", "unit": "V", "value": "0" } ] } ], "connectorId": 1, "transactionId": 0 }]
[ocpp ] TRACE 2024/12/04 19:16:05 send Garage: [3,"733053144",{}]
[ocpp ] TRACE 2024/12/04 19:16:05 recv Garage: [3,"2850175740", { "status": "Accepted" }]
[ocpp ] TRACE 2024/12/04 19:16:15 send Garage: [2,"1529028762","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/12/04 19:16:15 recv Garage: [3,"1529028762", { "status": "Accepted" }]
[ocpp ] TRACE 2024/12/04 19:16:15 recv Garage: [2,"1900553541","MeterValues", { "meterValue": [ { "timestamp": "2024-12-04T18:16:15Z", "sampledValue": [ { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Power.Active.Import", "unit": "W", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "89624" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1", "measurand": "Current.Import", "unit": "A", "value": "0.0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1-N", "measurand": "Voltage", "unit": "V", "value": "235.4" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Current.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "Power.Offered", "unit": "V", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Trigger", "measurand": "SoC", "unit": "V", "value": "0" } ] } ], "connectorId": 1, "transactionId": 0 }]
[ocpp ] TRACE 2024/12/04 19:16:15 send Garage: [3,"1900553541",{}]
[ocpp ] TRACE 2024/12/04 19:16:25 send Garage: [2,"266582363","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/12/04 19:16:25 recv Garage: [3,"266582363", { "status": "Accepted" }]
[ocpp ] TRACE 2024/12/04 19:16:25 recv Garage: [2,"1450956042","MeterValues", { "meterValue": [ { "timestamp": "2024-12-04T18:16:25Z", "sampledValue": [ { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Power.Active.Import", "unit": "W", "value": "0" }, { "format": "Raw", "location": "Outlet", "context": "Other", "measurand": "Energy.Active.Import.Register", "unit": "Wh", "value": "89624" }, { "format": "Raw", "location": "Outlet", "context": "Other", "phase": "L1", "measurand": "Current.Import", "unit": "A", "v [... cut ...]
[lp-1 ] WARN 2024/12/04 19:23:28 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/12/04 19:23:28 charger disable
[lp-1 ] DEBUG 2024/12/04 19:23:28 pv charge current: min 8A > 0A (538W @ 1p, battery: false)
[circuit-leitung_garage] TRACE 2024/12/04 19:23:28 validate current: 0A + (8A -> 8A) <= 32A ok
[circuit-leitung_garage] TRACE 2024/12/04 19:23:28 validate power: 0W + (0W -> 1840W) <= 20000W ok
[lp-1 ] DEBUG 2024/12/04 19:23:28 charger enable
[lp-1 ] DEBUG 2024/12/04 19:23:28 wake-up timer: start 
```

Does evcc need to unlock the charger somehow?

mfuchs1984, 2024-12-04:
You should setup the charger to free charging, open, ... If this is not possible, use `remotestart: true`

andig, 2024-12-04:
> OccurrenceConstraintViolation

Die WB verletzt die OCPP Spezifikation. Bitte Ticket beim Hersteller aufmachen.

closed 2024-12-04 by andig

maatinh, 2024-12-04:
> > OccurrenceConstraintViolation
> 
> Die WB verletzt die OCPP Spezifikation. Bitte Ticket beim Hersteller aufmachen.

Danke. Mal wieder ein typischer andig kommentar. Sehr hilfreich. Darf ich für das Ticket beim Hersteller bestimmt so zitieren, dann wissen die sicherlich gleich Bescheid. Auch wenn ich jetzt nicht weiß was genau das Problem ist.
Ich lerne daraus, mich hier künftig rauszuhalten.

mfuchs1984, 2024-12-04:
Die Verletzung habe ich hier beschrieben 
https://github.com/evcc-io/evcc/issues/17585#issuecomment-2518002031